kclyu/rpi-webrtc-streamer #1: RPI Audio Streaming?

frogg, 2016-12-08:
First of all, great project! I was wondering if it is also possible to transfer audio streams using WebRtc on the Raspberry PI using your framework. I couldn't find anything related to audio streaming in your documentation.

kclyu, 2016-12-08:
Thank you for your praise. Basically, the audio part is supposed to use the linux support part of the webrtc native code, and expect that the webrtc native linux code will work if there is no big problem. The audio part is going to be scheduled after some other high priority parts (TCP direct channel, websocket supports, video stream QoS) have progressed.

kclyu, 2017-05-24:
Please refer to the this [README_audio.md](README_audio.md)

closed 2017-05-24 by kclyu