kclyu/rpi-webrtc-streamer #3: Segfault while Chrome is connecting

franck-talbart, 2017-01-18:
When I push the connect button, the server crashes:

`
(basicportallocator.cc:1134): Jingle:Net[wlan0:192.168.1.0/24:Unknown]: Allocation Phase=Relay
(streamer.cc:297):  Received session description : "{"type":"answer","sdp":"v=0\r\no=- 6848690051349680870 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 0 8 105 13 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:YE6P\r\na=ice-pwd:V6QJbfO0JFFHXYsmPrDIURJN\r\na=fingerprint:sha-256 AE:DE:C6:92:D1:4C:AF:D0:B9:FC:8F:36:52:73:C8:F7:9C:0F:76:C9:4E:07:68:F7:F9:B1:DA:37:28:15:18:33\r\na=setup:active\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 100 96 98 127 104 97 99 101 125\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:YE6P\r\na=ice-pwd:V6QJbfO0JFFHXYsmPrDIURJN\r\na=fingerprint:sha-256 AE:DE:C6:92:D1:4C:AF:D0:B9:FC:8F:36:52:73:C8:F7:9C:0F:76:C9:4E:07:68:F7:F9:B1:DA:37:28:15:18:33\r\na=setup:active\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=recvonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtpmap:127 red/90000\r\na=rtpmap:104 ulpfec/90000\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:125 rtx/90000\r\na=fmtp:125 apt=127\r\n"}"
(stream_data_sockets.cc:111): GetSessionEntry
(stream_data_sockets.cc:116): Search Session 0xf9f110, id: 2, target: 2
(basicportallocator.cc:1134): Jingle:Net[wlan0:2a01:cb00:a1c:8c00::/64:Unknown]: Allocation Phase=Relay
(webrtcvideoengine2.cc:1325): Call stats: 1778819, {send_bw_bps: 300000, recv_bw_bps: 0, max_pad_bps: 0, pacer_delay_ms: 0, rtt_ms: -1}
(basicportallocator.cc:1134): Jingle:Net[wlan0:192.168.1.0/24:Unknown]: Allocation Phase=Relay
(webrtcsession.cc:1080): BUNDLE already enabled for audio on audio.

#
# Fatal error in ../webrtc/p2p/base/transportcontroller.cc, line 241
# last system error: 88
# Check failed: network_thread_->IsCurrent()
# 
#
(basicportallocator.cc:1134): Jingle:Net[wlan0:2a01:cb00:a1c:8c00::/64:Unknown]: Allocation Phase=Relay

==== C stack trace ===============================

./webrtc-streamer() [0x2825d4]
./webrtc-streamer() [0x59b738]
./webrtc-streamer() [0x24e5c4]
./webrtc-streamer() [0x24e87c]
./webrtc-streamer() [0x24d8fe]
./webrtc-streamer() [0x24cf58]
./webrtc-streamer() [0x1f31e2]
./webrtc-streamer() [0x1b9c6e]
./webrtc-streamer() [0x1b8706]
./webrtc-streamer() [0x1a55b6]
(basicportallocator.cc:1134): Jingle:Net[wlan0:192.168.1.0/24:Unknown]: Allocation Phase=Relay
./webrtc-streamer() [0x38b0f2]
./webrtc-streamer() [0x3ad35e]
./webrtc-streamer() [0x3acc9e]
./webrtc-streamer() [0x3acc7c]
Aborted
`

I am using the master branch here, since the branch56 is missing some header files.

Thanks!

kclyu, 2017-01-19:
Actually, the above message is not a crash, but a message created when invalid thead checking of the webrtc native stack. The situation seems to be in internal protection logic when IPv6 turn / stun gathering inside the native stack. At first, it seems that you should go to disable IPv6 and wait for update situation of webrtc.

One thing I'm wondering is whether this happens every time or what happens once.

As you told,  rws was working on Linux client, so it is better to go to disable IPv6 on client side.

- put ipv6 disable flag in native-to-browser-test.html
```
optional: [
      {DtlsSrtpKeyAgreement: true},
      **{'googIPv6': false}**
      ]
}
```
- temporarily disable IPv6 on Windows

The above issue does not occur in the network that I use,  so i would appreciate if you could test the two methods and let me know the result.

franck-talbart, 2017-01-19:
Hi Lyu,

To answer your question, I am hitting this on the linux client as well, and it's happening everytime.

So here is what I tried:
- I disabled ipv6 on the router and got this:

```
(streamer.cc:297):  Received session description : "{"type":"answer","sdp":"v=0\r\no=- 4232497907801999349 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 0 8 105 13 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:CUaD\r\na=ice-pwd:mnvxJKTMnI3aDbYG1aI/eAZc\r\na=fingerprint:sha-256 1B:56:42:3F:FD:44:61:0C:00:B2:31:94:FD:40:F7:D8:AE:15:50:46:F4:1E:F7:B3:41:C9:8C:DF:8E:03:8D:65\r\na=setup:active\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 100 96 98 127 104 97 99 101 125\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:CUaD\r\na=ice-pwd:mnvxJKTMnI3aDbYG1aI/eAZc\r\na=fingerprint:sha-256 1B:56:42:3F:FD:44:61:0C:00:B2:31:94:FD:40:F7:D8:AE:15:50:46:F4:1E:F7:B3:41:C9:8C:DF:8E:03:8D:65\r\na=setup:active\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=recvonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtpmap:127 red/90000\r\na=rtpmap:104 ulpfec/90000\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:125 rtx/90000\r\na=fmtp:125 apt=127\r\n"}"
(basicportallocator.cc:1134): Jingle:Net[wlan0:192.168.1.0/24:Unknown]: Allocation Phase=Relay
(basicportallocator.cc:1134): Jingle:Net[wlan0:192.168.1.0/24:Unknown]: Allocation Phase=Relay
(webrtcvideoengine2.cc:1325): Call stats: 54002321, {send_bw_bps: 300000, recv_bw_bps: 0, max_pad_bps: 0, pacer_delay_ms: 0, rtt_ms: -1}
(basicportallocator.cc:726): Jingle:Port[0x74cbbb18:video:2:0:local:Net[wlan0:192.168.1.0/24:Unknown]]: Gathered candidate: Cand[:1846443563:2:udp:1686052606:86.238.40.42:39154:stun:192.168.1.237:39154:4hVa:Xms3Y6to9ZST3QtR1EjV+hR7:3:50:0]
(basicportallocator.cc:835): Jingle:Port[0x74cbbb18:video:2:0:local:Net[wlan0:192.168.1.0/24:Unknown]]: Port completed gathering candidates.
(basicportallocator.cc:1134): Jingle:Net[wlan0:192.168.1.0/24:Unknown]: Allocation Phase=Relay
(webrtcsession.cc:1080): BUNDLE already enabled for audio on audio.

#
# Fatal error in ../webrtc/p2p/base/transportcontroller.cc, line 241
# last system error: 88
# Check failed: network_thread_->IsCurrent()
# 
#
(basicportallocator.cc:1134): Jingle:Net[wlan0:192.168.1.0/24:Unknown]: Allocation Phase=Relay

==== C stack trace ===============================

./webrtc-streamer() [0x2825d4]
./webrtc-streamer() [0x59b738]
./webrtc-streamer() [0x24e5c4]
./webrtc-streamer() [0x24e87c]
./webrtc-streamer() [0x24d8fe]
./webrtc-streamer() [0x24cf58]
./webrtc-streamer() [0x1f31e2]
./webrtc-streamer() [0x1b9c6e]
./webrtc-streamer() [0x1b8706]
./webrtc-streamer() [0x1a55b6]
./webrtc-streamer() [0x38b0f2]
./webrtc-streamer() [0x3ad35e]
./webrtc-streamer() [0x3acc9e]
./webrtc-streamer() [0x3acc7c]
Aborted
```

Then I have added the flag and got this:

```
(webrtcsdp.cc:2658): Ignored line: c=IN IP4 0.0.0.0
(thread.cc:279): Waiting for the thread to join, but blocking calls have been disallowed
(thread.cc:279): Waiting for the thread to join, but blocking calls have been disallowed
(thread.cc:279): Waiting for the thread to join, but blocking calls have been disallowed
(streamer.cc:297):  Received session description : "{"type":"answer","sdp":"v=0\r\no=- 7494635971191952462 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 0 8 105 13 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:Wtrk\r\na=ice-pwd:JlkJ/o1ELLLPHFHNt5w7SpUR\r\na=fingerprint:sha-256 06:63:CC:BA:7C:FD:24:09:22:C9:0E:43:69:5A:A0:E9:18:2D:15:96:04:BD:52:A4:6B:FC:58:F8:8C:A6:C5:5A\r\na=setup:active\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 100 96 98 127 104 97 99 101 125\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:Wtrk\r\na=ice-pwd:JlkJ/o1ELLLPHFHNt5w7SpUR\r\na=fingerpr [... cut ...]

==== C stack trace ===============================

./webrtc-streamer() [0x2825d4]
./webrtc-streamer() [0x59b738]
./webrtc-streamer() [0x24e5c4]
./webrtc-streamer() [0x24e87c]
./webrtc-streamer() [0x24d8fe]
./webrtc-streamer() [0x24cf58]
./webrtc-streamer() [0x1f31e2]
./webrtc-streamer() [0x1b9c6e]
./webrtc-streamer() [0x1b8706]
./webrtc-streamer() [0x1a55b6]
./webrtc-streamer() [0x38b0f2]
./webrtc-streamer() [0x3ad35e]
./webrtc-streamer() [0x3acc9e]
./webrtc-streamer() [0x3acc7c]
Aborted

```

kclyu, 2017-01-19:
Something's a lot weird. Attach the binaries that i have built. Please check how it works. Linux, window Both are verified binaries. 
[Uploading webrtc-streamer.gz…](webrtc-streamer.gz)

franck-talbart, 2017-01-19:
Your binary is working.

The most strange part is that it only takes 13Mb, whereas mine is 106Mb.

How big is your rpi-rootfs? Don't know if it's related but it would be interesting to try with yours.

kclyu, 2017-01-19:
The original file size is about 105M. It's a stripped file with debugging symbol removed for download. The following command removes the debugging symbol.

`` `
strip webrtc-streamer
`` `

So, the problem seems to be a webrtc building problem. It seems that webrtc and rpi-webrtc-streamer directories need to be deleted and rebuilt it from scratch.

franck-talbart, 2017-01-19:
Hmmm the current master branch of webrtc seems to be broken: I can't compile it anymore.
Branch 56 is still missing the header files.

Will investigate it as soon as I can.

kclyu, 2017-01-20:
First, I do not think that WebRTC native code package is not broken.  My guess is that the underlying problem is in the compiler toolchain.

I have some questions.
- what is cross cross compiler version
- what is the name of compiler distribution ( eg. linaro, raspi official... )

There are many kinds of compiler distribution, and depending on which distribution you use, sometimes the options do  work or do not, so it seems to make a problem sometimes.

I am using the custom build gcc version 4.9.4. and when I move the compiler binary to another directory location and build rws, the same problem appears as above.

In fact, there is no reason to use custom build gcc specifically. The official distribution gcc version of raspberry is old(4.7.1), but there was no problem using it. If there are reasons, i have to put the name of the executable in the environment variable and latest version(4.9.3) is not working correctly.

The current idea is embedding custom gcc in rpi_rootfs. I have forgotten the cross gcc building note made it before, so it will take some time to make it the same.

franck-talbart, 2017-01-21:
Hi Lyu,

I am using tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin taken from git://github.com/raspberrypi/tools.git

> arm-linux-gnueabihf-g++ -v
> Using built-in specs.
> COLLECT_GCC=arm-linux-gnueabihf-g++
> COLLECT_LTO_WRAPPER=/media/jeux_linux/Workspace/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/lto-wrapper
> Target: arm-linux-gnueabihf
> Configured with: /home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/.build/src/gcc-linaro-4.8-2014.03/configure --build=x86_64-build_unknown-linux-gnu --host=x86_64-build_unknown-linux-gnu --target=arm-linux-gnueabihf --prefix=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/install --with-sysroot=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/install/arm-linux-gnueabihf/libc --enable-languages=c,c++,fortran --disable-multilib --enable-multiarch --with-arch=armv6 --with-tune=arm1176jz-s --with-fpu=vfp --with-float=hard --with-pkgversion='crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03' --with-bugurl=https://bugs.launchpad.net/gcc-linaro --enable-__cxa_atexit --enable-libmudflap --enable-libgomp --enable-libssp --with-gmp=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/.build/arm-linux-gnueabihf/build/static --with-mpfr=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/.build/arm-linux-gnueabihf/build/static --with-mpc=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/.build/arm-linux-gnueabihf/build/static --with-isl=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/.build/arm-linux-gnueabihf/build/static --with-cloog=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/.build/arm-linux-gnueabihf/build/static --with-libelf=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/.build/arm-linux-gnueabihf/build/static --enable-threads=posix --disable-libstdcxx-pch --enable-linker-build-id --enable-plugin --enable-gold --with-local-prefix=/home/zhehe01/work/bzr/pi-build/builds/arm-linux-gnueabihf-raspbian-linux/install/arm-linux-gnueabihf/libc --enable-c99 --enable-long-long --with-float=hard
> Thread model: posix
> gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)

Current compilation is hitting a failing include:

> In file included from ../third_party/ffmpeg/libavutil/autorename_libavutil_cpu.c:2:0:
> ../third_party/ffmpeg/libavutil/cpu.c:24:23: fatal error: stdatomic.h: No such file or directory
>  #include <stdatomic.h>
>                        ^
> compilation terminated.
> [10/1318] CC obj/third_party/ffmpeg/ffmpeg_internal/h264pred.o
> ninja: build stopped: subcommand failed.

Apparently, this depends on the gcc version, as you said before.

Where do you get your gcc release?

Thanks!

kclyu, 2017-01-24:
I added gcc 4.9.4 to rpi_rootfs. Pull rpi_rootfs again and make the rpi_rootfs link under /opt and add '/opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin'  in your PATH environment variable.  Also RWS is now updated to WebRTC master so please pull it again.

franck-talbart, 2017-01-24:
Thank you very much for your patch!

Maybe I am missing something, as I am now hitting another compilation problem:

- It does not know what the armv7-a architecture is
- it fails to include stddef

```
Messages de l'assembleur:
Erreur fatale: option -march= invalide: « armv7-a »
In file included from ../third_party/ffmpeg/libavutil/avassert.h:30:0,
                 from ../third_party/ffmpeg/libavcodec/h264pred.c:29:
../../../../../../opt/rpi_rootfs/usr/include/stdlib.h:32:20: erreur fatale: stddef.h : Aucun fichier ou dossier de ce type
 #include <stddef.h>
                    ^
compilation terminée.

```

The compiler version looks fine:

```
arm-linux-gnueabihf-gcc -v
Utilisation des specs internes.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/media/jeux_linux/Workspace/rpi_rootfs/tools/arm-linux-gnueabihf/bin/../libexec/gcc/arm-linux-gnueabihf/4.9.4/lto-wrapper
Target: arm-linux-gnueabihf
Configuré avec: /home/kclyu/Workspace/cross/.build/src/gcc-linaro-4.9-2015.06/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=arm-linux-gnueabihf --prefix=/opt/rpi_rootfs/tools/arm-linux-gnueabihf --with-sysroot=/opt/rpi_rootfs/tools/arm-linux-gnueabihf/arm-linux-gnueabihf/sysroot --enable-languages=c,c++ --with-arch=armv6 --with-tune=arm1176jz-s --with-fpu=vfp --with-float=hard --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-248-gdf5a341 - Linaro GCC 2015.06' --with-bugurl=https://bugs.launchpad.net/gcc-linaro --disable-__cxa_atexit --disable-libmudflap --enable-libgomp --disable-libssp --enable-libquadmath --enable-libquadmath-support --disable-libsanitizer --with-gmp=/home/kclyu/Workspace/cross/.build/arm-linux-gnueabihf/buildtools --with-mpfr=/home/kclyu/Workspace/cross/.build/arm-linux-gnueabihf/buildtools --with-mpc=/home/kclyu/Workspace/cross/.build/arm-linux-gnueabihf/buildtools --with-isl=/home/kclyu/Workspace/cross/.build/arm-linux-gnueabihf/buildtools --with-cloog=/home/kclyu/Workspace/cross/.build/arm-linux-gnueabihf/buildtools --enable-lto --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --enable-threads=posix --disable-libstdcxx-pch --enable-linker-build-id --with-linker-hash-style=gnu --enable-plugin --enable-gold --disable-multilib --with-local-prefix=/opt/rpi_rootfs/tools/arm-linux-gnueabihf/arm-linux-gnueabihf/sysroot --enable-long-long --with-arch=armv6 --with-float=hard --with-fpu=vfp --enable-multiarch
Modèle de thread: posix
gcc version 4.9.4 20150629 (prerelease) (crosstool-NG crosstool-ng-1.22.0-248-gdf5a341 - Linaro GCC 2015.06) 
```

kclyu, 2017-01-25:
I think it's not appropriate to put the binaries on git. I reset rpi_rootfs and loaded tools_gcc_4.9.4.tar.gz. Try to unpack the tarball.

franck-talbart, 2017-01-25:
The tarball is corrupted when I clone the repo (I tried on several clients), but I could download it manually.

It's working now :) :) Thanks

closed 2017-01-25 by franck-talbart