kclyu/rpi-webrtc-streamer #4: RWS and Firefox

franck-talbart, 2017-01-19:
Did you have a chance to try on Firefox?
Even after applying the correct setup, I am not getting any video:

> Firefox: Activate H.264 for WebRTC
> 
> In URL address bar, type about:config
> In Search:, type h264
> The last line of "Preference Name" is:
> media.peerconnection.video.h264_enabled and it has the value false.
> Double-click on media.peerconnection.video.h264_enabled to make its
> value change to true. H.264 is now enabled to WebRTC.
> Firefox: Set H.264 as default video codec in place of VP8
> 
> Open about:config page
> Anywhere in the window, right-click to open contextual menu. Select
> NEW --> INTEGER
> A dialog window ask for preference name. Type:
> media.navigator.video.preferred_codec
> Next dialog window ask for the integer value that will be assigned to
> the name. Type: 126
> 126 is the code for H.264 mode 1 in Firefox.
> Start a new WebRTC communication. Your Firefox will now propose H.264
> mode 1 codec as first option in its offer.

Thanks a lot!

kclyu, 2017-01-19:
Firefox was successful in video streaming initially using ubuntu. But I did not try windows. I remember that I changed the configuration after installing the firefox plugin instead of changing the configuration just like chrome.

franck-talbart, 2017-01-19:
Ok, for some reasons the plugin can't be installed on my Debian version.
On Windows, it's installed with the correct parameters, but nothing is displayed :(

kclyu, 2017-05-02:
I have not tested it for a long time, but it works on the page that changed the javascript related to PeerConnection to adaptor.js. Please check if you can check.

The modified native-peerconnection html currently only works in the master repo.

closed 2017-05-24 by kclyu