kclyu/rpi-webrtc-streamer #5: Do you have any working configuration on Audio?

kclyu, 2017-05-22:
It is not easy to buy as many hardware as possible for the test, and there is not really much time to test it. If you have successfully set up your audio with your Raspberry PI and have verified audio transmission from RWS, please share it. I think it will help a lot of others.

Please add the comment to the issue as shown below.

- Device Type: (USB dongle | USB microphone | Raspberry PI audio hat | etc)
- Device Name: (Device Model)
- Command Output:
    - lsusb
    - cat / proc / asound / cards

kclyu, 2017-05-22:
My Working Audio configuration:
Device Type:  USB microphone
Device Name: Cosy USB microphone MK1343UB 
Command Output:
```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 0d8c:0061 C-Media Electronics, Inc. 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ cat /proc/asound/cards
 0 [Microphone     ]: USB-Audio - USB Microphone
                      Raysound USB Microphone at usb-3f980000.usb-1.4, full speed
pi@raspberrypi:~ $ cat /proc/asound/modules
 0 snd_usb_audio
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.27-v7+ #997 SMP Tue May 9 19:58:37 BST 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /proc/asound/modules
```

towolf, 2017-09-16:
I have a Zoom H1 that can server as a pretty good USB mic.

But alsamixer shows this on connection: `This sound device does not have any controls.`

And I think, this is why the WebRTC streaming does not work:

```
(audio_device_impl.cc:526): InitMicrophone
(webrtcvoiceengine.cc:672): webrtc: AudioMixerManagerLinuxALSA::OpenMicrophone(name=default)
(webrtcvoiceengine.cc:672): webrtc:      snd_mixer_attach(_inputMixerHandle, default)
(webrtcvoiceengine.cc:672): webrtc: Could not find capture volume on the mixer.
(webrtcvoiceengine.cc:672): webrtc: Init() failed to initialize the microphone (error=9004)
```
Full log:
```
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (websocket_server.cc:346): Found handler URI in config : /rws/ws
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (websocket_server_callback.cc:551): URI Path exist /rws/ws
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (websocket_server_callback.cc:453): LWS_CALLBACK_ESTABLISHED
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (app_channel.cc:177): New WebSocket connnection id : 12
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (app_channel.cc:192): OnMessage(12)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (app_channel.cc:209): JSON Parsing Success: {"cmd":"register","roomid":"987974376","clientid":"97868899"}
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (app_channel.cc:250): Room ID: 987974376, Client ID: 97868899
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (streamer_observer.cc:105): IsStreamSessionActive
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (streamer_observer.cc:66): ActivateStreamSession
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (streamer_observer.cc:147): ObtainStreamer
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (streamer.cc:236): OnPeerConnectedPeer 0,  connected, trying to initialize streamer instance
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (raspi_encoder.cc:454): Raspberry H.264 MMAL encoder factory.
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:238): WebRtcVoiceEngine::WebRtcVoiceEngine
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:245): Supported send codecs in order of preference:
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): ISAC/16000/1 (103)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): G722/8000/1 (9)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): ILBC/8000/1 (102)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): PCMU/8000/1 (0)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): PCMA/8000/1 (8)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): CN/16000/1 (105)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): CN/8000/1 (13)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): telephone-event/48000/1 (110)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): telephone-event/16000/1 (113)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:248): telephone-event/8000/1 (126)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:251): Supported recv codecs in order of preference:
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): ISAC/16000/1 (103)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): G722/8000/1 (9)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): ILBC/8000/1 (102)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): PCMU/8000/1 (0)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): PCMA/8000/1 (8)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): CN/16000/1 (105)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): CN/8000/1 (13)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): telephone-event/48000/1 (110)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): telephone-event/16000/1 (113)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:254): telephone-event/8000/1 (126)
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:262): VoiceEngine 4.1.0
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:84): Create
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_buffer.cc:62): AudioDeviceBuffer::ctor
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:128): AudioDeviceModuleImpl
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:136): CheckPlatform
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:150): current platform is Linux
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:177): CreatePlatformSpecificObjects
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:1844): PlatformAudioLayer
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:275): Linux PulseAudio is *not* supported => ALSA APIs will be utilized instead
Sep 16 17:27:42 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:344): AttachAudioBuffer
[... cut ...]
Sep 16 17:27:44 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:672): webrtc: Could not find output volume in the mixer.
Sep 16 17:27:44 zero-w webrtc-streamer[1760]: (webrtcvoiceengine.cc:672): webrtc:   InitSpeaker() failed
Sep 16 17:27:44 zero-w webrtc-streamer[1760]: (audio_device_buffer.cc:185): SetPlayoutSampleRate(48000)
Sep 16 17:27:44 zero-w webrtc-streamer[1760]: (audio_device_buffer.cc:209): SetPlayoutChannels(2)
Sep 16 17:27:44 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:1015): output: 1
Sep 16 17:27:44 zero-w webrtc-streamer[1760]: (audio_device_impl.cc:1024): SetStereoPlayout(1)
```

kclyu, 2017-09-18:
First of all, it is necessary to check whether the USB being used is normally operating ALSA.

```
pi@raspberrypi:~ $ dmesg
...
...
[3725352.416197] usb 1-1.4: new full-speed USB device number 4 using dwc_otg
[3725352.555438] usb 1-1.4: New USB device found, idVendor=0d8c, idProduct=0061
[3725352.555456] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[3725352.555466] usb 1-1.4: Product: USB Microphone
[3725352.555477] usb 1-1.4: Manufacturer: Raysound
[3725352.555487] usb 1-1.4: SerialNumber: 201409
[3725352.569117] input: Raysound USB Microphone as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.2/0003:0D8C:0061.0001/input/input0
[3725352.569532] hid-generic 0003:0D8C:0061.0001: input,hidraw0: USB HID v1.11 Device [Raysound USB Microphone] on usb-3f980000.usb-1.4/input2
[3725352.700615] usbcore: registered new interface driver snd-usb-audio
pi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 0d8c:0061 C-Media Electronics, Inc. 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ cat /proc/asound/cards
 0 [Microphone     ]: USB-Audio - USB Microphone
                      Raysound USB Microphone at usb-3f980000.usb-1.4, full speed
pi@raspberrypi:~ $ cat /proc/asound/modules
 0 snd_usb_audio
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.27-v7+ #997 SMP Tue May 9 19:58:37 BST 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /proc/asound/modules
 0 snd_usb_audio
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Microphone [USB Microphone], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

The command below saves the input from the actual mic as a wav file, and if it is confirmed that it is recorded normally on another PC, it will work normally. (Use ^ C to abort the command.)
```
pi@raspberrypi:~ $ arecord --format S16_LE --rate 44100 -c1 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Mono
^CAborted by signal Interrupt...
arecord: pcm_read:2031: read error: Interrupted system call
```

towolf, 2017-09-18:
Yes, all those work. But there is no volume control. It's not there in my laptop either, but it's works fine generally anyway.

Only the webrtc stuff insists on a mixer, which is not there.

kclyu, 2017-09-18:
I do not know if this is relevant, but try setting the volume with alasmixer and saving the changed settings as default values.
```
alsamixer    #  Adjust volume value
sudo alsactl store  # save the alsa config values to file
```
It seems that there is a problem with the setting of pi that you are using, but at this moment I do not have the same phenomenon and can not say anything.

Below is the log executed in my pi.
```
[019:540] [9391] (audio_device_impl.cc:467): Init
[019:546] [9391] (audio_device_impl.cc:1196): SetPlayoutDevice(0)
[019:652] [9391] (webrtcvoiceengine.cc:672): webrtc:   number of availiable audio output devices is 0
[019:652] [9391] (audio_device_impl.cc:516): InitSpeaker
[019:694] [9391] (webrtcvoiceengine.cc:672): webrtc: AudioMixerManagerLinuxALSA::OpenSpeaker(name=default)
[019:695] [9391] (webrtcvoiceengine.cc:672): webrtc:      snd_mixer_attach(_outputMixerHandle, default)
[019:696] [9391] (webrtcvoiceengine.cc:672): webrtc: snd_mixer_selem_get_name 0: Mic =750bf260
[019:696] [9391] (webrtcvoiceengine.cc:672): webrtc: Could not find output volume in the mixer.
[019:696] [9391] (webrtcvoiceengine.cc:672): webrtc: Init() failed to initialize the speaker (error=9005)
[019:696] [9391] (audio_device_impl.cc:1291): SetRecordingDevice(0)
[019:739] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 1 - default:CARD=Microphone
[019:739] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 2 - sysdefault:CARD=Microphone
[019:739] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 3 - front:CARD=Microphone,DEV=0
[019:739] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 4 - surround21:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 5 - surround40:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 6 - surround41:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 7 - surround50:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 8 - surround51:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 9 - surround71:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 10 - iec958:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 11 - dsnoop:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 12 - hw:CARD=Microphone,DEV=0
[019:740] [9391] (webrtcvoiceengine.cc:672): webrtc:     Enum device 13 - plughw:CARD=Microphone,DEV=0
[019:741] [9391] (webrtcvoiceengine.cc:672): webrtc:   number of availiable audio input devices is 14
[019:741] [9391] (audio_device_impl.cc:526): InitMicrophone
[019:784] [9391] (webrtcvoiceengine.cc:672): webrtc: AudioMixerManagerLinuxALSA::OpenMicrophone(name=default)
[019:784] [9391] (webrtcvoiceengine.cc:672): webrtc:      snd_mixer_attach(_inputMixerHandle, default)
[019:785] [9391] (webrtcvoiceengine.cc:672): webrtc:      Mic element found
[019:785] [9391] (webrtcvoiceengine.cc:672): webrtc:      Using Mic as capture volume.
[019:785] [9391] (webrtcvoiceengine.cc:672): webrtc:   the input mixer device is now open (0x750c4c40)
[019:785] [9391] (audio_device_impl.cc:1005): StereoPlayoutIsAvailable
[019:828] [9391] (webrtcvoiceengine.cc:672): webrtc: AudioMixerManagerLinuxALSA::OpenSpeaker(name=default)
[019:828] [9391] (webrtcvoiceengine.cc:672): webrtc: Closing playout mixer
[019:828] [9391] (webrtcvoiceengine.cc:672): webrtc:      snd_mixer_attach(_outputMixerHandle, default)
[019:829] [9391] (webrtcvoiceengine.cc:672): webrtc: snd_mixer_selem_get_name 0: Mic =750c8520
[019:829] [9391] (webrtcvoiceengine.cc:672): webrtc: Could not find output volume in the mixer.
[019:829] [9391] (webrtcvoiceengine.cc:672): webrtc:   InitSpeaker() failed
[019:871] [9391] (webrtcvoiceengine.cc:672): webrtc:   InitPlayout open (default)
[019:892] [9391] (webrtcvoiceengine.cc:672): webrtc:      unable to open playback device: No such file or directory (-2)
[019:892] [9391] (audio_device_impl.cc:1015): output: 0
[019:892] [9391] (audio_device_impl.cc:1024): SetStereoPlayout(0)
[019:892] [9391] (audio_device_buffer.cc:209): SetPlayoutChannels(1)
[019:892] [9391] (audio_device_impl.cc:891): StereoRecordingIsAvailable
[019:936] [9391] (webrtcvoiceengine.cc:672): webrtc: AudioMixerManagerLinuxALSA::OpenMicrophone(name=default)
[019:936] [9391] (webrtcvoiceengine.cc:672): webrtc: Closing record mixer
[019:936] [9391] (webrtcvoiceengine.cc:672): webrtc: Closing record mixer
[019:937] [9391] (webrtcvoiceengine.cc:672): webrtc: Closing record mixer
[019:937] [9391] (webrtcvoiceengine.cc:672): webrtc: Closing record mixer
[019:937] [9391] (webrtcvoiceengine.cc:672): webrtc:      snd_mixer_attach(_inputMixerHandle, default)
[019:937] [9391] (webrtcvoiceengine.cc:672): webrtc:      Mic element found
[019:937] [9391] (webrtcvoiceengine.cc:672): webrtc:      Using Mic as capture volume.
[019:938] [9391] (webrtcvoiceengine.cc:672): webrtc:   the input mixer device is now open (0x750cd958)
[019:980] [9391] (webrtcvoiceengine.cc:672): webrtc: InitRecording open (default)
[019:988] [9391] (webrtcvoiceengine.cc:672): webrtc:     capture snd_pcm_get_params buffer_size:2048 period_size:1024
[019:989] [9391] (audio_device_buffer.cc:178): SetRecordingSampleRate(48000)
[019:989] [9391] (audio_device_buffer.cc:202): SetRecordingChannels(2)
[... cut ...]
[020:361] [9391] (audio_processing_impl.cc:687): Highpass filter activated: 0
[020:361] [9391] (audio_device_impl.cc:1465): RegisterAudioCallback
[020:361] [9391] (audio_device_buffer.cc:77): RegisterAudioCallback
[020:361] [9391] (audio_device_impl.cc:1465): RegisterAudioCallback
[020:361] [9391] (audio_device_buffer.cc:77): RegisterAudioCallback
[020:361] [9391] (webrtcvideoengine2.cc:477): WebRtcVideoEngine2::WebRtcVideoEngine2()
[020:362] [9391] (webrtcvideoengine2.cc:485): WebRtcVideoEngine2::Init
```

towolf, 2017-09-19:
Hello, thanks for your suggestion, but that was my point, that the device does not have volume controls:

[image]

And I think it may not work because the code is assuming that all audio devices have these controls.

kclyu, 2017-09-20:
In my pi, alsamixer is displayed as below.
[image]
Press tab or press F4 to go to Mic item.

Below is what I set pi to. If there is a missing part in pi setting, it is better to try it.

- /boot/config.txt 
```
...
# Enable audio (loads snd_bcm2835)
#dtparam=audio=on ## comment out for USB audio
start_x=1
gpu_mem=128
...
```
- /lib/modprobe.d/aliases.conf
```
...
# prevent unusual drivers from appearing as the first sound device ###########
options snd-pcsp index=-2
#options snd-usb-audio index=-2	# comment out for USB audio priority
options cx88_alsa index=-2
options snd-atiixp-modem index=-2
...
```
When you changed something in your pi, you need to reboot pi to use new updated configuration. and use alsamixer and alsactl store.

towolf, 2017-09-20:
There are no audio volume controls whatsoever. This is basically a handheld
recorder which can work as a USB audio interface

https://goo.gl/images/VigWEG

I just don't understand why it is trying to control the volume, when I
disabled all volume control settings ...

kclyu, 2017-09-20:
There is no particular reason. The audio part of alsamixer should work normally so that the audio part will be used normally in WebRTC native stack as well. It is intended to check whether the related part of audio is operating normally in the ALSA tools.

kclyu, 2018-06-14:
The meaning of the log message seems to be conveying too many messages from the microphone device.   It seems to be a problem because it tried to use the speaker hardware set as a microphone.
```
# This thread is too busy: webrtc_audio_module_capture_thread 18ms sequence=1001 328847 vs 328829, 1 vs 2
```
In asound.conf, only one piece of hardware is defined. Refer to [Setting the default microphone / capture device](https://wiki.archlinux.org/index.php/Advanced_Linux_Sound_Architecture/Troubleshooting#Setting_the_default_microphone.2Fcapture_device) for the capture and playback device settings.

noliveleger, 2018-06-14:
@kclyu I've posted in this thread because of Audio issue. I just wanted to tell you about my /etc/asound.conf which implies the Speaker pHat, but if you want me to post in the other thread, let me know. 
So in `/etc/asound.conf`, only the speaker was configured, but I could record flawlessly with this command `arecord --device=plughw:1,0 --format S16_LE --rate 44100 -c1 default.wav`.

I did change the configuration for this 
````
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "softvol_and_pivumeter"
        }
        capture.pcm {
                type plug
                slave.pcm  "hw:1,0"
        }
}

ctl.!default {
        type hw
        card 0
}

pcm.pivumeter {
        type meter
        slave.pcm "hw:0,0"
        scopes.0 pivumeter
}

pcm.softvol_and_pivumeter {
        type softvol
        slave.pcm "pivumeter"
        control {
                name "PCM"
                card 0
        }
}

pcm_scope.pivumeter {
        type pivumeter
        decay_ms 500
        peak_ms 400
        brightness 128
        output_device speaker-phat
}

pcm_scope_type.pivumeter {
        lib /usr/local/lib/libpivumeter.so
}

pcm.dsp0 pivumeter
```
If I don't specify the `--device=plughw:1,0`, is

kclyu, 2018-06-14:
In my opinion, the following changes should be made in asound.conf.

```
pcm.usb
{
    type hw
    card 1
}

pcm.!default
{
    type asym
    playback.pcm
    {
        type plug
        slave.pcm "dmix"
    }
    capture.pcm
    {
        type plug
        slave.pcm "usb" 
    }
}
```

noliveleger, 2018-06-15:
Ok, I'm gonna do my homework to make my USB microphone as default device for recording and I let you know as soon as it works. 
Thank you for your help.

noliveleger, 2018-06-20:
@kclyu : FYI: The problem was the Speaker pHat. It uses PulseAudio. I could not find a way to make PulseAudio choose the microphone as the default recording device.
After removing PulseAudio and restoring `/etc/asound.conf` to 
```
pcm.!default {
   type asym
   playback.pcm "plug:hw:0"
   capture.pcm "plug:dsnoop:1"
}
```
**it works**.
Device Type: USB microphone & on-board output 3.5 mm jack 
Device Name: [Mini USB Microphone](https://www.adafruit.com/product/3367)
Command Output:
```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 005: ID 8086:0808 Intel Corp.
Bus 001 Device 007: ID 06cb:0009 Synaptics, Inc. Composite TouchPad and TrackPoint
Bus 001 Device 006: ID 04b3:3019 IBM Corp.
Bus 001 Device 004: ID 04b3:3016 IBM Corp. UltraNav Keyboard Hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
pi@raspberrypi:~ $ cat /proc/asound/cards
 0 [ALSA           ]: bcm2835_alsa - bcm2835 ALSA
                      bcm2835 ALSA
 1 [Device         ]: USB-Audio - USB PnP Sound Device
                      C-Media Electronics Inc. USB PnP Sound Device at usb-3f980000.usb-1.5, full spe
pi@raspberrypi:~ $ cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_usb_audio
```

```
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux
```

drewcovi, 2018-12-03:
@k-cactus any luck on your end? I have a GPIO mic as well. would love to crack this nut.

noliveleger, 2018-12-07:
@BloodyF , which configuration? SpeakerPHat or USB microphone?
I could not make it work with the SpeakerPHat because it uses PulseAudio. 
To make two-ways audio work, I used the frontend files @k-cactus made from this thread 
https://github.com/kclyu/rpi-webrtc-streamer/issues/45#issuecomment-399320898
[tway-audio.zip](https://github.com/kclyu/rpi-webrtc-streamer/files/2126340/tway-audio.zip)
I just replaced the files and it worked flawlessly.

BloodyF, 2018-12-08:
thank @noliveleger very much, you saved my life :)

k-cactus, 2019-01-22:
> @k-cactus any luck on your end? I have a GPIO mic as well. would love to crack this nut.

unfortunately,  that nut is hard for me... so still waiting when someone can crack it....

noliveleger, 2019-01-22:
@kclyu I have ordered one. I let you know if I have some success.
Do you think it's related to linux configuration or from your software?

kclyu, 2019-01-23:
It's hard to find out what point you're talking about, is it GPIO mic?
If you create a recording file through ALSA with GPIO mic and the file plays without problems, it seems to be OK to use it in RWS as well.

If you are considering the sound quality of the mic, it would be better to use hardware noise removal and echo cancellation from the mic if possible. Of course, there is noise and echo-related audio processing in the native stack of WebRTC, but the CPU load can be a problem for raspberry pi, which has low CPU performance.

noliveleger, 2019-01-23:
@kclyu : Sorry for the mistake. I wanted to mention @k-cactus. I misread the OP.
So my questions were for @k-cactus, not you. I'll edit my post.

k-cactus, 2019-01-23:
> @k-cactus: Are you able to make the mic work with `arecord`?

arecord works perfect... but WEBRTC log have something like 

```
.... Could not find capture volume on the mixer.
Sep 06 01:22:41 raspberrypi webrtc-streamer[1161]: [007:308] [1171] (audio_device_alsa_linux.cc:880): InitMicrophone() failed
Sep 06 01:22:41 raspberrypi webrtc-streamer[1161]: [007:326] [1171] (audio_device_alsa_linux.cc:919): unable to open record device: No such file or directory ....
```

leofds, 2019-02-25:
I use a usb adapter with input and output audio with p2 connectors. This adapter uses the CM108 Audio Controller IC.

```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 008: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The only configuration iI made was to edit the file /usr/share/alsa/alsa.conf

```
defaults.ctl.card 1
defaults.pcm.card 1
```

mintu19, 2019-05-04:
- Device Type: USB dongle (self fabricated)
- Card IC Name: PCM2912A
- Command Output:

```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 003: ID 0e6f:010c Logic3
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ cat /proc/asound/cards
 0 [Device         ]: USB-Audio - PDP Audio Device
                      Performance Designed Products PDP Audio Device at usb-3f980000.usb-1, full spee
pi@raspberrypi:~ $ cat /proc/asound/modules
 0 snd_usb_audio
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Device [PDP Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Device [PDP Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

blademckain, 2019-05-11:
Device Type: USB dongle https://www.adafruit.com/product/1475
Command Output:

`root@raspberrypi:~# lsusb
Bus 001 Device 002: ID 8086:0808 Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@raspberrypi:~# cat /proc/asound/modules
 0 snd_usb_audio

root@raspberrypi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

root@raspberrypi:~# arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`
tested with 
`speaker-test -c2`
works

and with
` arecord --device=hw:0,0 --format S16_LE --rate 44100 -c1 test.wav
aplay --device=plughw:0,0 test.wav
`
both commands works

But when I plug audio source to pi the client couldn't establish ws channel between peers ( without usb dongle stream from pi to Pc works)

logs here
[log_rpi-webrtc-streamer.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/3169398/log_rpi-webrtc-streamer.txt)

mcmara, 2019-06-23:
I'm using the following board:

- Device Type: 
  - Raspberry PI audio hat
- Device Name: 
  - [Respeaker 2 mic](https://www.seeedstudio.com/category/Development-Platforms-c-1002/single-board-computer-c-950/category/Speech-Recognition-c-44/ReSpeaker-2-Mics-Pi-HAT.html)
- Command Output:
```
pi@raspberrypi:~/Workspace/rws $ cat /proc/asound/cards 
 0 [seeed2micvoicec]: seeed-2mic-voic - seeed-2mic-voicecard
                      seeed-2mic-voicecard

pi@raspberrypi:~/Workspace/rws $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~/Workspace/rws $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

The only thing I had to do, besides installing the drivers, was disabling raspberry pi audio:
```
sudo emacs /boot/config.txt
#dtparam=audio=on ## comment out
```

Sensitivity is very good. The only issue I have is a small delay in audio (about 0.5s). This makes the audio stream not perfectly synced with the video. Not sure is this is an issue of this board or if the delay in the audio stream exists with every other mic.

tholmgren, 2020-12-27:
Has there been any update on this?  
I tested on Zero W and 3B, both with Buster OS.  

Video works great but there is no audio on either device.  

in the /opt/rws/etc/ 
I updated the webrtc_streamer.conf and media_config.conf to enable audio.

I did the same just incase in the template folder too.

I even tried to follow this audio readme:  
https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_audio.md

No luck... 

From my Zero W.

lsusb
Bus 001 Device 002: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

cat /proc/asound/cards
 0 [Device         ]: USB-Audio - USB Audio Device
                      C-Media Electronics Inc. USB Audio Device at usb-20980000.usb-1, full speed

cat /proc/asound/modules
 0 snd_usb_audio

I tried UV4L, the video and audio were fine.  
FFMPEG and VLC all work too for streaming audio.

Any ideas how to get the audio working with RWS?

tholmgren, 2020-12-28:
Incase anyone else runs into an audio issue, please check the media player control on the HTML, make sure it's not muted.   :-)

DocSneider, 2020-12-29:
https://github.com/kclyu/rpi-webrtc-streamer/issues/5#issuecomment-719679184

Same here :disappointed: 
- RPI Model:
  - Zero W V1.1

I'm using the following board:
- Device Type:
  - Raspberry PI audio hat ( I2C / I2S ) 

- Device Name:
  - [ReSpeaker 2-Mics Pi HAT](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/)

Video stream works very well, but when I set audio_enable=true, the cpu-usage goes up to 100%.
Even when I am only using audio but no video, the cpu-usage goes up to 70-90%.

Audio_processing_enable is set to false. The ReSpeaker 2-Mics Pi HAT contains an onboard stereo codec, so the CPU should normally don´t have that much to do. When I am executing the following command, it uses max 2% of the cpu:
`arecord -D hw:0,0 -V stereo -r 16000 -f S16_LE -c 2 /dev/null`

I am using "DietPii" without desktop enviroment, should be almost the same as Raspberry Pi OS lite.