kclyu/rpi-webrtc-streamer #6: Video is delay about 10s when view on chromium browser of webrtc

quancao, 2017-06-06:
Hi,
Thanks for the great project.
I could compile this and run on one raspberry pi 3 board.
Viewing on PC works fine but I got about 10s delay when viewing by chromium-browser on another raspberry.
Do you have any recommendations to improve that performance?
Many thanks,
Quan

kclyu, 2017-06-06:
In normal operating environment, it should be within 300 ~ 500 ms delay. It seems to be encoded as VP8 which is WebRTC native fallback encoding.
Make sure that WebRTC H.264 encoding is enabled in Chrome.
Please check the rws log message to see if there is something similar to the following.
```
grep log_file raspi_encoder
```
(Raspi_encoder.cc:454): Raspberry H.264 MMAL encoder factory.
(Raspi_encoder.cc:480): Create Raspberry PI HW MMAL video encoder for H264
(Raspi_encoder.cc:152): InitEncode request: 320 x 240
(Raspi_encoder.cc:167): MMAL encoder drain thread initialized.
(Raspi_encoder.cc:180): RaspiEncoder Release request:
(Raspi_encoder.cc:152): InitEncode request: 320 x 240
(Raspi_encoder.cc:167): MMAL encoder drain thread initialized.
(Raspi_encoder.cc:234): Resolution Changing by Bitrate Changing To: 400x300
(Raspi_encoder.cc:234): Resolution Changing by Bitrate Changing To: 640x480
(Raspi_encoder.cc:234): Resolution Changing by Bitrate Changing To: 1024x768
(Raspi_encoder.cc:234): Resolution Changing by Bitrate Changing To: 1152x864
(Raspi_encoder.cc:180): RaspiEncoder Release request:
(Raspi_encoder.cc:500): Destroy Raspberr PI HW MMAL video encoder.
(Raspi_encoder.cc:180): RaspiEncoder Release request:
(Raspi_encoder.cc:470): MMALVideoEncoderFactory destroy

quancao, 2017-06-06:
Yes, I got the H.264 encoder with this log
(raspi_encoder.cc:480): Create Raspberry PI HW MMAL video encoder for H264

And from other raspi board, chromium-browser also report H.264
[7:33:0606/075041.682140:INFO:webrtcvideoengine2.cc(820)] Using codec: VideoCodec[100:H264]
[7:33:0606/075041.682529:INFO:webrtcvideoengine2.cc(859)] SetFeedbackOptions on all the receive streams because the send codec or RTCP mode has changed.

kclyu, 2017-06-06:
Did you mean that there is a 10 second delay in the chromium of raspberry pi?

Chrome/chromium uses H.264 S/W decoding.  Since chrome/chromium does not use H/W decoding, raspberry pi does not have sufficient speed and performance by S/W decoding.

If you want to improve performance with chromium in raspberry pi, it seems that there is no way to modify chromium's webrtc decoding to use GPU  ;-)

quancao, 2017-06-06:
yes, it looks like the decoder of chromium on raspi have issue. It is slow down overtime.
In the first two mins, the performance is quite good. After that, it seems decoder is slow down and the delay is up to 10s.
I tried with UV4L-WebRTC, the issue did not happen. I think the different between uv4l-webrtc and rpi-webrtc-streamer is that UV4L has constant bit-rate 
<notice> [driver] H264 costant bitrate: 8000000

I tried to configure rpi-webrtc-streamer to constant bit-rate but I got crash.

kclyu, 2017-06-06:
please use initial_video_resolution=true option together when disable the use_dynamic_video_resolution

```
use_dynamic_video_resolution=false
use_initial_video_resolution=true
...
initial_video_resolution=640x480
```

quancao, 2017-06-06:
Great, the delay issue is solved with that configuration.
Thank for your support

kclyu, 2017-06-07:
It looks like you are testing with chromium M56 and It looks good to wait for the M57 or later version to come out and test again.  
Although it does not seem to be enough yet, there have been a lot of improvements related to BWE, and I'm not sure this issue(https://bugs.chromium.org/p/webrtc/issues/detail?id=6690) will fix  delay issue caused  by decoding delay. In the M57 and above have a related patch, so it may be better to try again later when you have latest chromium for raspberry pi.

kclyu, 2017-06-07:
In the middle of waiting, it may be better to set the fps from 30 to 15 and to reduce the resolution while adjusting the decoding delay at the raspberry pi.

quancao, 2017-06-07:
I will try to reduce fps to 15 and check while waiting for M57.
Many thanks for your recommendations.

kclyu, 2017-06-20:
Currently this issue seems to belong to Raspberry PI Chromium. First of all, this issue is close. If you have issues that require support from RWS later, please reopen.

closed 2017-06-20 by kclyu