kclyu/rpi-webrtc-streamer #7: gn gen arm_build    <- problem

adrian23, 2017-06-19:
Hi,
I'm trying to build my own packages I tried to do everything as You wrote ,but I got error while running this command You wrote notes:
>    note: You need to replace the whole absolute path in args.gn before doing 'gn gen arm_build'
>    note: webrtc_build_args.gn was renamed to webrtc_arm_build_args.gn
 unfortunetly I dont understand which paths You are talking about, alsoI dont understand note2.
ofcourse i've not changed  paths as u suggested My system is Ubuntu 17

ThankYou for this awesome project :) 

Regards,
Adrian

kclyu, 2017-06-20:
Initially, I used the home path as an absolute path, such as /home/????/ Workspace/webrtc in args.gn. it do not have it right now and  the document seems to have to be modified.

closed 2017-06-20 by kclyu

adrian23, 2017-06-20:
HI thanks,
I also reinstall ubuntu from 17 into 16. now everything works fine .
Thank You