kclyu/rpi-webrtc-streamer #8: build problem

adrian23, 2017-06-22:
HI,i try to build your project using instruction:
`https://github.com/kclyu/rpi-webrtc-streamer/blob/master/BUILD_ubuntu.md`
i changed the part `Building WebRTC with GN build` into :
```
cd ~/Workspace/
git clone https://github.com/kclyu/rpi-webrtc-streamer.git
cd ~/Workspace/webrtc/src
mkdir arm_build
cp ~/Workspace/rpi-webrtc-streamer/misc/webrtc_arm_build_args.gn arm_build/args.gn
cd arm_build
gn gen arm_build
mv -v ./arm_build/* .
```
without these (`cd arm_build` and `mv -v ./arm_build/* .`) I had errors.

and when I called make i got :
[afterMake.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/1096353/afterMake.txt)
as i understand WEBRTC_CFLAGS are wrong becasue it has this unrecognize parameters.

Can You Please tell me what i'm doing wrong ?
I even change my ubuntu username to `kclyu` becasue in build there was references to `/home/kclyu/Workspace/cross/.build/src/gcc-linaro-4.9-2015.06/configure` from rpi_rootfs. I also download `gcc-linaro-4.9-2015.06` into `home/kclyu/Workspace/cross/.build/src/` so i have the same configuration as You had.

I'm using newest webrtc sources
Thank You

PS at your tutorial u mention file : ` cross_arm_gn.mk` i think You meant `cross_arm_build.mk`

kclyu, 2017-06-23:
```
cp ~/Workspace/rpi-webrtc-streamer/misc/webrtc_arm_build_args.gn arm_build/args.gn
cd arm_build
gn gen arm_build
```
If you do 'gn gen arm_build' in the arm_build directory, you are making default x86_64 build  for ubuntu without args.gn. 

The first thing to check is the error message that occurred when executing 'gen gen arm_build' in the ~/Workspace/webrtc/src directories. Please let me know the error message.

First of all, it is highly probable that Raspberry PI's root fs (rpi_rootfs) does not have the package needed to build webrtc.  Make sure you have installed https://github.com/kclyu/rpi_rootfs#required-package-in-raspberry-pi in the rpi_rootfs repo in raspberry pi.

If the error message related to the gtk3 of the package, then try to install libgtk-3-dev which is added recently on webrtc build on raspiberry pi and recreate rpi_rootfs. 

p.s. If the problem is package dependency and you have solved it, please let us know the package list you installed.

adrian23, 2017-06-23:
Thank You very much :),
yes i didnt install requires packages at my rasbperry pi i've been doing this at my ubuntu. it solves build problem. So i have built but after put exe into rasbperry pi (i'm using zero W) i got :
`/webrtc-streamer: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ./webrtc-streamer)`
i tried to update `libstdc++6` but its telling me i already have nwest version 
```strings /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_DEBUG_MESSAGE_LENGTH
```
```
strings /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_DEBUG_MESSAGE_LENGTH
```
Do You know how to get GLIBCXX_3.4.21 ?

kclyu, 2017-06-24:
First of all, for Raspberry PI Zero W, if you have built RWS using webrtc_arm_build_args.gn, build it using **webrtc_armv6_build_args.gn**. And if you use ZeroW OS with **lite**, please change to **desktop** version.

adrian23, 2017-06-24:
HI i tried like You wrote to do I did everything the same until step :
```cp ~/Workspace/rpi-webrtc-streamer/misc/webrtc_arm_build_args.gn arm_build/args.gn```
I put 
```cp ~/Workspace/rpi-webrtc-streamer/misc/webrtc_armv6_build_args.gn arm_build/args.gn``` instead 

`gn gen arm_build`  run without problem but :
`ninja -C arm_build`
return errors : [ninja arm_build.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/1099805/ninja.arm_build.txt)
Can You please tell me if i did everything correct ?
Thank You

kclyu, 2017-06-24:
There seems to be no problem with the build process ....

```
kclyu@keunchang2:~/Workspace/webrtc/src$ 
kclyu@keunchang2:~/Workspace/webrtc/src$ 
kclyu@keunchang2:~/Workspace/webrtc/src$ mv arm_build arm_build.neon
kclyu@keunchang2:~/Workspace/webrtc/src$ mkdir arm_build
kclyu@keunchang2:~/Workspace/webrtc/src$ cp ~/Workspace/rpi-webrtc-streamer/misc/
AppRTCMobile.apk                       webrtc_arm_build_args.gn
WEBRTC_BRANCH.conf                     webrtc_armv6_build_args.gn
nginx-ssl-client-authenticate-example  webrtc_streamer_architecture.png
nginx-ssl-config-example               webrtc_x64_build_args.gn
webrtc_android_build_args.gn           
kclyu@keunchang2:~/Workspace/webrtc/src$ cp ~/Workspace/rpi-webrtc-streamer/misc/webrtc_armv6_build_args.gn arm_build/args.gn
kclyu@keunchang2:~/Workspace/webrtc/src$ ls
AUTHORS              OWNERS         arm_build        check_root_dir.py     out                tools
BUILD.gn             PATENTS        arm_build.neon   cleanup_links.py      pylintrc           tools_webrtc
CODE_OF_CONDUCT.md   PRESUBMIT.py   base             codereview.settings   resources          webrtc
DEPS                 README.md      build            data                  sync_chromium.pyc
LICENSE              WATCHLISTS     build_overrides  infra                 testing
LICENSE_THIRD_PARTY  android_build  buildtools       license_template.txt  third_party
kclyu@keunchang2:~/Workspace/webrtc/src$ gn gen arm_build
Done. Made 235 targets from 108 files in 2289ms
kclyu@keunchang2:~/Workspace/webrtc/src$ ninja -C arm_build
ninja: Entering directory `arm_build'
[571/1735] ACTION //third_party/opus:convert_rtcd_assembler(//build/toolchain/linux:arm)
perl /home/kclyu/Workspace/webrtc/src/third_party/opus/src/celt/arm/arm2gnu.pl /home/kclyu/Workspace/webrtc/src/third_party/opus/src/celt/arm/celt_pitch_xcorr_arm.s | sed "s/OPUS_ARM_MAY_HAVE_[A-Z]*/1/g" | sed "/.include/d" > gen/third_party/opus/celt_pitch_xcorr_arm_gnu.S
[858/1735] CXX obj/webrtc/base/rtc_task_queue/task_queue_libevent.o
../webrtc/base/task_queue_libevent.cc: In destructor ‘rtc::TaskQueue::PostAndReplyTask::~PostAndReplyTask()’:
../webrtc/base/task_queue_libevent.cc:182:50: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
     write(reply_pipe_, &message, sizeof(message));
                                                  ^
[1735/1735] STAMP obj/default.stamp
kclyu@keunchang2:~/Workspace/webrtc/src$ 
```
First of all,

1. build again with armv6 args.gn after completely deleting arm_build directory
2. Check that arm-linux-gnueabihf-gcc -version appears as shown below.
```
Kclyu @ keunchang2: ~ / Workspace / webrtc / src $ arm-linux-gnueabihf-gcc --version
Arm-linux-gnueabihf-gcc (crosstool-NG crosstool-ng-1.22.0-248-gdf5a341 - Linaro GCC 2015.06) 4.9.4 20150629 (prerelease)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; See the source for copying conditions. There is NO
Warranty; Not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
3. I'm wondering if you raspberry pi update after installing raspberry pi os.
```
sudo apt update
sudo apt full-upgrade
```
4, I do not think this problem will happen because of the latest os update, but I update the raspberry pi to the latest one so I can rebuild and report the result.

adrian23, 2017-06-24:
Hi, yes You had right i had wrong `arm-linux-gnueabihf-gcc` after I correct everything works perfect.
Thank You very much :)

closed 2017-06-25 by adrian23