kclyu/rpi-webrtc-streamer #10: rasbperry pi zero W ninja build problem

adrian23, 2017-08-05:
Hi,
`when i try to build webrtc for ~/Workspace/rpi-webrtc-streamer/misc/webrtc_armv6_build_args.gn`
after 
`ninja -C arm_build` i got error` :
`In file included from ../third_party/opus/src/celt/cpu_support.h:36:0,
                 from ../third_party/opus/src/celt/kiss_fft.h:35,
                 from ../third_party/opus/src/celt/mdct.h:46,
                 from ../third_party/opus/src/celt/modes.h:36,
                 from ../third_party/opus/src/celt/pitch.h:37,
                 from ../third_party/opus/src/celt/arm/arm_celt_map.c:32:
../third_party/opus/src/celt/arm/pitch_arm.h:93:54: error: ‘arch’ undeclared here (not in a function)
 #   define celt_pitch_xcorr (*CELT_PITCH_XCORR_IMPL[(arch)&OPUS_ARCHMASK])
                                                      ^
../third_party/opus/src/celt/arm/armcpu.h:38:32: note: in definition of macro ‘MAY_HAVE_MEDIA’
 #  define MAY_HAVE_MEDIA(name) name ## _media
                                ^
../third_party/opus/src/celt/arm/arm_celt_map.c:64:3: note: in expansion of macro ‘MAY_HAVE_NEON’
   MAY_HAVE_NEON(celt_pitch_xcorr)   /* NEON */
   ^
../third_party/opus/src/celt/arm/arm_celt_map.c:64:17: note: in expansion of macro ‘celt_pitch_xcorr’
   MAY_HAVE_NEON(celt_pitch_xcorr)   /* NEON */
                 ^
../third_party/opus/src/celt/arm/pitch_arm.h:93:74: error: pasting ")" and "_media" does not give a valid preprocessing token
 #   define celt_pitch_xcorr (*CELT_PITCH_XCORR_IMPL[(arch)&OPUS_ARCHMASK])
                                                                          ^
../third_party/opus/src/celt/arm/armcpu.h:38:32: note: in definition of macro ‘MAY_HAVE_MEDIA’
 #  define MAY_HAVE_MEDIA(name) name ## _media
                                ^
../third_party/opus/src/celt/arm/arm_celt_map.c:64:3: note: in expansion of macro ‘MAY_HAVE_NEON’
   MAY_HAVE_NEON(celt_pitch_xcorr)   /* NEON */
   ^
../third_party/opus/src/celt/arm/arm_celt_map.c:64:17: note: in expansion of macro ‘celt_pitch_xcorr’
   MAY_HAVE_NEON(celt_pitch_xcorr)   /* NEON */
                 ^
../third_party/opus/src/celt/arm/armcpu.h:38:40: error: expected ‘}’ before ‘_media’
 #  define MAY_HAVE_MEDIA(name) name ## _media
                                        ^
../third_party/opus/src/celt/arm/armcpu.h:46:31: note: in expansion of macro ‘MAY_HAVE_MEDIA’
 #  define MAY_HAVE_NEON(name) MAY_HAVE_MEDIA(name)
                               ^
../third_party/opus/src/celt/arm/arm_celt_map.c:64:3: note: in expansion of macro ‘MAY_HAVE_NEON’
   MAY_HAVE_NEON(celt_pitch_xcorr)   /* NEON */
   ^
cc1: warning: unrecognized command line option "-Wno-#pragma-messages"
[775/1802] CC obj/third_party/opus/opus/opus_encoder.o
ninja: build stopped: subcommand failed.
`
Can you check if this build conf is correct and work with newest webrts ?

Thanks

kclyu, 2017-08-06:
Hi, 
I get the same error. 
If you are not using OPUS,   Building without opus module seems to be the fastest way. 
Currently, OPUS for armv6 seems to have an issue with building. (ARM neon does not have the same issue).

To exclude the opus from building, add the following two lines to args.gn:
 
rtc_build_opus=false
rtc_include_opus=false

adrian23, 2017-08-06:
HI,
thanks yes it solve prlbem thankyou

closed 2017-08-06 by adrian23