kclyu/rpi-webrtc-streamer #11: Program crash after connect rasbperry pi zero

adrian23, 2017-08-06:
Hi,
I have newest program with newest webrtc and after click conenct at site . Application crashes.
I'm using rasbperry pi zero W

there are logs :
[logs.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/1203506/logs.txt)

application at sctoud prints :
`terminating
Aborted
`
and thats all. As You see at file logs webrtc cannot initialize microphone. I dont have and also didnt have microphone or speakers when using old verion of program (which works details below). I also at media_config.conf change to false all setitngs starts with audio but it ddint help.

with previous version of webrtc and your program (I had version from Jun 24.and this version was before they changed packages) application works perfectly

Thank You,

kclyu, 2017-08-07:
HI,

I do not get the same error. Compiling armv6 with the latest webrtc native package and no problem in running it on the RI Zero W with current master of RWS.  About audio-related error messages, it does no effect on rws running. If the device is not found or failed to to test whether device is working, the device will not be added to the  list of working devices in the native stack.

I need more detail information such like core  than log message to check what is the reason.

At first, you need to change ulimit to get core file like below.
```
pi@raspberryW:~ $ pwd
/home/pi
pi@raspberryW:~ $ ulimit -a
core file size          (blocks, -c) 0
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 2898
max locked memory       (kbytes, -l) unlimited
max memory size         (kbytes, -m) unlimited
open files                      (-n) 65536
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 95
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 2898
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited
pi@raspberryW:~ $ ulimit -c unlimited
pi@raspberryW:~ $ ulimit -a
core file size          (blocks, -c) unlimited
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 2898
max locked memory       (kbytes, -l) unlimited
max memory size         (kbytes, -m) unlimited
open files                      (-n) 65536
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 95
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 2898
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited
```

and disable and stop the rws process to run in terminal window.
```
pi@raspberryW:~ $ sudo systemctl disable rws
Removed symlink /etc/systemd/system/multi-user.target.wants/rws.service.
pi@raspberryW:~ $ sudo systemctl stop rws
pi@raspberryW:~ $ sudo systemctl status rws
● rws.service - Rpi WebRTC Streamer
   Loaded: loaded (/lib/systemd/system/rws.service; disabled)
   Active: inactive (dead)

Aug 07 12:13:48 raspberryW systemd[1]: Starting Rpi WebRTC Streamer...
Aug 07 12:13:48 raspberryW systemd[1]: Started Rpi WebRTC Streamer.
Aug 07 12:13:48 raspberryW webrtc-streamer[1236]: Using config file base path:/opt/rws/
Aug 07 12:14:23 raspberryW systemd[1]: Stopping Rpi WebRTC Streamer...
Aug 07 12:14:23 raspberryW systemd[1]: Stopped Rpi WebRTC Streamer.
Aug 07 12:38:59 raspberryW systemd[1]: Starting Rpi WebRTC Streamer...
Aug 07 12:38:59 raspberryW systemd[1]: Started Rpi WebRTC Streamer.
Aug 07 12:38:59 raspberryW webrtc-streamer[1445]: Using config file base path:/opt/rws/
Aug 07 12:39:14 raspberryW systemd[1]: Stopping Rpi WebRTC Streamer...
Aug 07 12:39:14 raspberryW systemd[1]: Stopped Rpi WebRTC Streamer.
```

and run the rws in terminal window
[... 89 lines cut ...]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `/opt/rws/webrtc-streamer --verbose'.
Program terminated with signal SIGQUIT, Quit.
#0  0xb6db4b80 in poll () at ../sysdeps/unix/syscall-template.S:81
81	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) where
#0  0xb6db4b80 in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x0022d8e8 in _lws_plat_service_tsi (context=0xf8cd90, timeout_ms=25, tsi=0)
    at /home/kclyu/Workspace/rpi-webrtc-streamer/lib/libwebsockets/lib/lws-plat-unix.c:147
#2  0x0022db08 in lws_plat_service (context=0xf8cd90, timeout_ms=25)
    at /home/kclyu/Workspace/rpi-webrtc-streamer/lib/libwebsockets/lib/lws-plat-unix.c:204
#3  0x0021e47c in lws_service (context=0xf8cd90, timeout_ms=25)
    at /home/kclyu/Workspace/rpi-webrtc-streamer/lib/libwebsockets/lib/service.c:1334
#4  0x000191c0 in LibWebSocketServer::RunLoop (this=<optimized out>, timeout=25, timeout@entry=0)
    at websocket_server.cc:223
#5  0x0002f34c in StreamingSocketServer::Wait (this=0xbea8eff4, cms=<optimized out>, process_io=<optimized out>)
    at main.cc:72
#6  0x000be108 in rtc::MessageQueue::Get (this=0xbea8f110, pmsg=0x56891c, cmsWait=3840304, process_io=<optimized out>)
    at ../webrtc/rtc_base/messagequeue.cc:353
#7  0x000c4a04 in rtc::Thread::ProcessMessages (this=this@entry=0xbea8f110, cmsLoop=cmsLoop@entry=-1)
    at ../webrtc/rtc_base/thread.cc:485
#8  0x000c4a84 in rtc::Thread::Run (this=this@entry=0xbea8f110) at ../webrtc/rtc_base/thread.cc:317
#9  0x0001806c in main (argc=16436588, argv=<optimized out>) at main.cc:202
(gdb) 
```

adrian23, 2017-08-07:
Hi so:
`pi@raspberrypi:~ $ ulimit -a
core file size          (blocks, -c) unlimited
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 2898
max locked memory       (kbytes, -l) unlimited
max memory size         (kbytes, -m) unlimited
open files                      (-n) 65536
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 95
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 2898
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited
pi@raspberrypi:~ $  sudo systemctl status rws
● rws.service
   Loaded: masked (/dev/null)
   Active: inactive (dead)

Warning: Unit file changed on disk, 'systemctl daemon-reload' recommended`

running and debugging :
[crash.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/1205941/crash.txt)

BUT when i run as sudo everything works great ...

Thanks,

kclyu, 2017-08-11:
Is this issue active yet? 
I do not know where the problem is with the gdb trace that you posted earlier. i think there is building(linking) or other environment problems, but i need gdb trace log  which have source file name and a line number like the gdb trace I posted above.

If you still have problems, try rebuilding the webrtc package using use_custom_libcxx=false in args.gn.

adrian23, 2017-08-11:
Hi,
as i wrote above when I run it as sudo everything works great. So i do not worry about this issue anymore. I will close ticket.
Thanks

closed 2017-08-11 by adrian23