kclyu/rpi-webrtc-streamer #12: Only one connected browser

adrian23, 2017-08-10:
What is the reason You made possibility to connect only one browser at the same time ?
I'm just curious was it only one of your assumption or there was some problems to keep streaming to two sessions at the same time.
Regards,
Adrian

kclyu, 2017-08-11:
First of all, the biggest reason is that Raspberry PI H/W currently supports one CSI Camera and the purpose of RWS program is Realtime Video Encoding.  
Currently, RWS encodes video by adjusting FPS, bitrate, screen resolution according to BWE (Bandwidth Estimation) of one connected RTC session, and directly configures setting of H/W component camera and MMAL encoder. 

Simultaneous sessions from one WebRTC Client have been recently added to the WebRTC native stack. However, in Raspberry PI H/W, it need to evaluate various methods (eg RWS Encoding Architecture change, hardware performance for encoding fps and bitrate per session) in order to check whether it is possible.

In a nutshell, many things should be considered and evaluated whether it is possible, and also it takes a lot of time and effort to develop. While concurrent multiple session features are still to be considered, they are not a high priority right now, so it is a feature that should be tried if higher priority ones are resolved.

Thanks, 
Lyu.

adrian23, 2017-08-11:
Thanks for  your answer :)
Cheers

closed 2017-08-11 by adrian23