kclyu/rpi-webrtc-streamer #14: How to make it working with appr.tc

Mirdox, 2017-08-31:
Hi Kclyu,

I like your project, it works perfectly.
I'm really glad, that there is WebRTC support on Raspberry Pi.

I'm wondering, if this your application can work directly with e.g. https://appr.tc/.
Is there some part, that needs to be changed, or it is configurable?
I guess the signaling should be somehow changed.

Can you give me some clues?

Thank you.

BR

kclyu, 2017-09-01:
Hi,

I am not sure whether the answer should be easy or should be a little complicated.

For simple answer, 
In the RWS, there is no feature that can be used  with https://appr.tc as client.  As you can see, https://appr.tc is for experimental use by google and Using https://appr.tc, which anyone can connect to, has privacy and security issues.

For complicated answer,
If you want to modify the AppRTC server to address security and privacy issues and use it with Google App Engine, you need to add the client functionality to RWS. So far, There is no need for this yet, so i am  not specially considering it.

Instead, RWS has a signaling server feature that connect directly to RWS through AppRTC Android App(It does not mean Direct Socket).
However, there are issues related to configuration and security for the most users(users who are not familiar with using internet domain name and installing certificate to use https, authenticating users something like client certificate for security) to use RWS as an AppRTC server.

If you have a domain name and you do not have any problems configuring your client certificate, you can install the client certificate in nginx and build the AppRTC andorid client by applying misc/AndroidApp_ClinetCerts.patch to use the client certificate of AppRTC android client.

BR,
Lyu.

closed 2017-09-10 by kclyu

exonreyes, 2018-07-05:
Como hacer que funciones https://appr.tc/ de manera local con app enginering?

kclyu, 2018-07-06:
AppRTC interface functions  are no longer supported. Support for older versions of RWS is possible, but this AppRTC interface function has been removed from the source.

Since the AppRTC interface function is judged to be unavailable to the most of users because it require setup domain name and  certificate/client certificate.
Even if you are an experienced user, using a certificate and building your own app is thought to be very difficult.

[rpi-webrtc-streamer-frontend](https://github.com/kclyu/rpi-webrtc-streamer-frontend) (WEB front-end) is not actively developed yet, but plan to use it for external WEB/App integration of RWS.