kclyu/rpi-webrtc-streamer #15: MMAL re-init sometimes fails

towolf, 2017-09-09:
Hi, I am mostly opening this issue to tell that this is a super-genius project. It is exactly what I thought is needed when I played with WebRTC Janus server before. Not a huge thing cobbled together out of lots of components, but a slim native implementation of WebRTC, in a small 2.4 MB package. The protocol is the way to go, because of native support in browsers. No other solution gives really live video on smartphone browsers. 

Now with my early tries sometimes the process was pegging CPU at 100% when the resolution was being scaled up, so the video started to lag behind. Not sure why.

So, I limited it to 640x480 and 1500000 bps max. But the the MMAL re-init would start to fail:

```
 (raspi_quality_config.cc:147): Bitrate changed from 814, to 818
 (raspi_quality_config.cc:147): Bitrate changed from 818, to 821
 (raspi_quality_config.cc:147): Bitrate changed from 821, to 826
 (raspi_quality_config.cc:147): Bitrate changed from 826, to 828
 (raspi_quality_config.cc:147): Bitrate changed from 828, to 829
 (raspi_quality_config.cc:147): Bitrate changed from 829, to 841
 (raspi_quality_config.cc:147): Bitrate changed from 841, to 873
 (raspi_quality_config.cc:147): Bitrate changed from 873, to 875
 (raspi_quality_config.cc:147): Bitrate changed from 875, to 877
 (raspi_quality_config.cc:147): Bitrate changed from 877, to 881
 (raspi_quality_config.cc:147): Bitrate changed from 881, to 884
 (raspi_quality_config.cc:147): Bitrate changed from 884, to 896
 (raspi_quality_config.cc:147): Bitrate changed from 896, to 909
 (raspi_quality_config.cc:147): Bitrate changed from 909, to 915
 (raspi_quality_config.cc:147): Bitrate changed from 915, to 969
 (raspi_quality_config.cc:147): Bitrate changed from 969, to 983
 (raspi_quality_config.cc:248): BestMatch Resolution (end of loop): 640x480
 (raspi_encoder.cc:234): Resolution Changing by Bitrate Changing To : 0x0
 (mmal_wrapper.cc:289): ReinitEncoder 0x0@30, 0 kbps
 (mmal_wrapper.cc:295): EncoderDelay Status changed from INIT_PASS to WAITING
 (mmal_wrapper.cc:503): Start reinitialize the MMAL encode wrapper.0x0@30, 0kbps
 (mmal_wrapper.cc:740): capture stopped.
 mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
 mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0xb4f227a0 (EINVAL)
 mmal: camera viewfinder format couldn't be set
 (mmal_wrapper.cc:542): Failed to create camera component
 (mmal_wrapper.cc:297): Failed to reinitialize MMAL encoder
 (raspi_encoder.cc:240): Failed to reinit MMAL encoder
 (raspi_quality_config.cc:147): Bitrate changed from 983, to 989
 (raspi_quality_config.cc:248): BestMatch Resolution (end of loop): 640x480
 (raspi_encoder.cc:234): Resolution Changing by Bitrate Changing To : 0x0
 (mmal_wrapper.cc:289): ReinitEncoder 0x0@30, 0 kbps
 (raspi_quality_config.cc:147): Bitrate changed from 989, to 1014
 (raspi_quality_config.cc:248): BestMatch Resolution (end of loop): 640x480
 (raspi_encoder.cc:234): Resolution Changing by Bitrate Changing To : 0x0
 (mmal_wrapper.cc:289): ReinitEncoder 0x0@30, 0 kbps
 (mmal_wrapper.cc:335): EncoderDelay Status changed from WAIT to PASS
 (video_send_stream.cc:960): SignalEncoderTimedOut, Encoder timed out.
 (bitrate_allocator.cc:191): UpdateAllocationLimits : total_requested_min_bitrate: 0bps, total_requested_padding_bitrate: 0bps
 (bitrate_allocator.cc:83): Current BWE 1014440
 (mmal_wrapper.cc:807): MMAL force key frame encoding
 (rtp_rtcp_impl.cc:175): Process: Timeout: No increase in RTCP RR extended highest sequence number.
```

So I disabled dynamic resolution for now, even though it is super-cool. It is more important to be reliable. 

Bitrate switches happen really frequently. Is this necessary? I think every few seconds could be sufficient?

Are you going to put out newer deb builds? The building procedure seems to be relatively complicated.

kclyu, 2017-09-10:
First of all, thank you for complimenting this repo. It's still a long way to go, but I'm glad it's useful to someone.

- About reinit failure
First, I can not tell you what the cause of the problem is because I have not yet experienced the same situation.   It seems that you have used 640x480 only in the Dynamic Resolution config list, but if you only set one video resolution, it's better to disable dynamic resolution and use static video resolution.
Anyway, I will fix reinit failure when i find the same problem.

- About frequent bitrate changing
Bitrate changing is an API call delivered from the WebRTC Native stack. Of course i can adjust the bitrate changing appropriately and  i know that the log message looks a bit strange, but the Google Bandwidth Estimation part is still evolving and even if changing bitrate frequently, there are no special issues currently in RWS.

- About latest DEB package
The DEB package will not be updated frequently because it will be updated if it stabilizes to some extent after adding major function. It seems that it takes more time because the function i am adding  is related to motion detection which is a complex feature of some degree.

Thanks, 
Lyu.

kclyu, 2017-09-10:
An issue has been closed with an abnormal mouse double-click. I will close the issue if reinit failure is resolved.

towolf, 2017-09-11:
Thanks for your explanations.

> About frequent bitrate changing

I was mostly thinking that the process is using quite a lot of CPU and maybe the load could be reduced by performing some functions less frequently ...

closed 2017-12-04 by kclyu