kclyu/rpi-webrtc-streamer #16: iOS 11 support

superddr, 2017-09-14:
Hey this is an excellent work! I can finally watch my own room from work now ! So cool and I will probably control a remote car to move arround and follow my little baby.
Recently iOS 11 is released ,with WebRTC support , does any body know how to make it work on iphone? I'm working on it , still have no luck yet. Hope we can watch live video on iphone soon!

superddr, 2017-09-25:
alright,i heard that chrome is using vp8 while apple is using h264, thats why the video wont show on iphone. is there any chance that we can use h264？

kclyu, 2017-09-25:
In practice RWS also uses only H.264 codec. Codec support might be better suited.

I've tried a simple test but I have confirmed that the video is not yet available. Right now I spend most of my time on the motion detection function, so I have not been able to test it yet deeply.

I will let you know when it is tested or if there are other issues.

superddr, 2017-09-26:
Hey good news! I finally figured it out. iOS just don't allow autoplay, so you have to manually start it by the controls. So the video tag has to be like this way: <video playsinline controls />.  Now you can click the start button to start it.

closed 2017-09-26 by superddr