kclyu/rpi-webrtc-streamer #18: Upside Down video

embadmath, 2017-10-16:
Hi,
is there an option to upside down the video streaming?

Can i set mjpeg as standard codec?

Thanks

kclyu, 2017-10-17:
video flip related media_config.conf option is as follows.
```
video_rotation=0
video_hflip=0
video_vflip=0
```
Default is 0 and after setting enable as 'video_rotaion=1`, you can set the desired horizontal or vertical option.
```
video_rotation=1
video_hflip=0
video_vflip=1
```

This section is missing from the media_config option document. We will update later.

rws does not support the mjpeg codec. but h.264 streaming is more efficient than mjpeg,  it does not need to support mjpeg codec.

Thanks.

kclyu, 2017-10-17:
First, I'm sorry for the confusion.
Currently, the options in the media_config.config file are not working on the deb package src. I will close the issue as the feature is updated.

My Old post contents was the below.

video flip related media_config.conf option is as follows.

video_rotation=0
video_hflip=0
video_vflip=0
Default is 0 and after setting enable as 'video_rotaion=1`, you can set the desired horizontal or vertical option.

video_rotation=1
video_hflip=0
video_vflip=1
This section is missing from the media_config option document. We will update later.

rws does not support the mjpeg codec. but h.264 streaming is more efficient than mjpeg, it does not need to support mjpeg codec.

Thanks.

embadmath, 2017-10-17:
Hi,
thanks for info. I prefer h264 codec but on android device using webview i cannot decode h264. I want to make a custom apk using android studio but i don't know where to start using webview and h264.

Thanks again

kclyu, 2017-10-17:
There is an example App for android  in the WebRTC source tree. Looks like a good start to the example of using the WebRTC API for Android and WebRTC android native.

https://webrtc.org/native-code/android/

VarolOkan, 2017-10-23:
Hey,
first of thanks for your program I am currently working on a small project where I need to flip the video horizontally. However I tried with different settings.

One thing I noted was that there is a difference in the source code with the binary from 
https://github.com/kclyu/rpi-webrtc-streamer-deb

Are you planing in updating these binarie soon ?
Note: I am using OpenSuSE so the cross compilation would not work for me. Also what are the chances of compiling directly on the Raspberry PI ?

kclyu, 2017-10-23:
Hi, 
When the major version stabilizes, I will update the deb version. But not soon.
I expect it to be about one month or two months later.

In raspberry pi, even a webrtc native source package was not downloaded a year ago, but I have not tested it recently.

If webrtc source package is downloaded and there is no problem with webrtc build, it will be possible to compile without any problem. Instead, only the rws make include files need to be modified appropriately.

VarolOkan, 2017-10-24:
hEY
JUST A QUICK QUESTION, WHY DON'T YOU START USING
https://github.com/aisouard/libwebrtc

It seems to be easier if you focus in rpi-webrtc-streamer instead of cross compiling webrtc.

I got libwebsockets to compile ( ../mk/config_libwebsockets.sh ) and am now trying to get rpi-webrtc-streamer to compil

Cheers

kclyu, 2017-10-24:
Hi,
libwebrtc does not yet provide arm builds. To use webrtc in raspberry pi, you still need to cross compile.

kclyu, 2017-12-04:
new 0.72 deb file newly updated.

closed 2017-12-04 by kclyu