kclyu/rpi-webrtc-streamer #19: DirectRTC connection through NAT

fraopess, 2017-10-29:
Hello Lyu, 
I successfully installed rws on a Rpi Zero W from your deb package. While the native peer connection works well on port 8889, I am trying to access rws via the android DirectRTCClient.
All works great if the android client is connected on the private LAN, however I can't connect if I try through the public IP (port 8888 being forwarded to the Pi by my router).
On the app I see the local media in fullscreen for a few seconds and then I get the message "ICE connection failed".
On the android logs I successfully see all the ICE candidates being published (host and srflx), but I can't see the logs on the rws. Is there any way to enable debug logs?

If I expose the Pi in DMZ mode the DirectRTCClient works well, so I think it is something related to ports access. Maybe stun is ineffective and I need a turn server?

Really appreciate your work and your help.
Francesco

kclyu, 2017-10-29:
Hi Francesco,

I have the same phenomenon, so I had been trying to figure out why for several days, but I thought it will be the same as you guessed. 

In order to understand the cause of the problem, I have to look at the WebRTC native code deeply.  but I did not because i thought that it was not a high priority issue.

Anyway, if there is an update on this issue, I will update it on this issue.

However, even though DirectRTCClient works well on private and public networks,  I do not recommend using DirectRTCClient on public networks because DirectRTCClient does not have security features.

Thanks,
Lyu.

kclyu, 2019-04-05:
Problem handling only one of multiple DirectSocket messages #76
DirectRTC TURN server #72

There have been several DirectRTC related issues, and now it seems to be solvable, so I will link them and close this issue.

closed 2019-04-05 by kclyu