kclyu/rpi-webrtc-streamer #21: stdio.h no such files

embadmath, 2017-11-18:
Hi,
i'm trying to build rpi-webrtc-streamer using cross compiling but i always get this error:

websocket_server.cc:30:19: fatal error: stdio.h: No such file or directory

could you help me?

Thanks

kclyu, 2017-11-19:
Hi,

There seems to be a problem with the environment for cross compiling.
You will need to make sure that you have created the cross compile environment in the order in which it appears in the BUILD_ubuntu.md document.

Thanks

embadmath, 2017-11-21:
Hi,
i try to solve but now i get this:

`arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-fcolor-diagnostics’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wtautological-constant-out-of-range-compare’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wheader-hygiene’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wstring-conversion’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wtautological-overlap-compare’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wc++11-narrowing’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wimplicit-fallthrough’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wthread-safety’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Winconsistent-missing-override’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wunused-lambda-capture’
Makefile:67: recipe for target 'websocket_server.o' failed
make: *** [websocket_server.o] Error 1
`
what could be?

kclyu, 2017-11-22:
It seems that you tried to build the webrtc native stack using the wrong args.gn file.
Please refer to the BUILD_ubuntu.md file again and try to build RWS after completing the webrtc stack build.

closed 2017-12-29 by kclyu