kclyu/rpi-webrtc-streamer #22: Video resolution stuck at 640x480

nikivanov, 2017-12-28:
Works great on my Rpi Zero W, but I can't get the resolution to be anything but 640x480. My config file is below media_config.conf is below. Any ideas?

`max_bitrate=3000000
use_4_3_video_resolution=false
video_rotation=0
video_hflip=true
video_vflip=true
use_dynamic_video_resolution=false
fixed_video_resolution=1280x720
video_resolution_list_4_3=320x240,400x300,512x384,640x480,1024x768,1152x864,1296x972,1640x1232
video_resolution_list_16_9=384x216,512x288,640x360,768x432,896x504,1024x576,1152x648,1280x720,1408x864,1920x1080
audio_processing_enable=false
audio_echo_cancellation=true
audio_gain_control=true
audio_high_passfilter=true
audio_noise_suppression=true
audio_level_control_enable=true
`

kclyu, 2017-12-28:
Currently, there seems to be a bug related to config below. I will post this part again after the source modification is completed.

use_dynamic_video_resolution = false
fixed_video_resolution = 1280x720

kclyu, 2017-12-29:
Sorry for the confusion. It was a mistake made by omitting the deleted config item.

`use_4_3_video_resolution` is no longer used.

Please remove the `use_4_3_video_resolution=false` config item,
and use 'resolution_4_3_enable' config .

resolution_4_3_enable = false -> using 16: 9 video resolution
resolution_4_3_enable = true -> using 4: 3 video resolution

closed 2018-01-15 by kclyu