kclyu/rpi-webrtc-streamer #24: Adjusting brightness

rskwan, 2018-01-03:
What would be the recommended way for one to adjust the brightness of the camera? I'm currently using the `.deb` package, and adjusting the brightness would be helpful to improve the image we're getting.

kclyu, 2018-01-04:
Brightness and other settings are not yet supported. I will notify you after adding the feature.

alin256, 2018-02-12:
Is it possible to have an API for an image filter? This should solve this and many other similar problems...
Or would it sufficiently slow down the streaming?

kclyu, 2018-02-12:
@alin256  
Something seems to be wrong with point. First, rws uses the MMAL library because it uses Raspberry PI's GPU. Therefore, there is no part that can be modified for the video frame. The brightness adjustment is already pushed to the master and will be included in the next deb.

closed 2018-02-19 by kclyu