kclyu/rpi-webrtc-streamer #25: Safari / WKWebView support

TheJKM, 2018-01-23:
Hi, is it possible to add support for Safari / WKWebView on iOS? Safari supports WebRTC since iOS 11 / macOS High Sierra, but this example (I tried /native-peerconnection) does not work. Support for Safari would be very nice, because there's no alternative on iOS and I'd like to use this piece of software because the low latency is very nice, especially for a Raspberry Pi.

kclyu, 2018-01-24:
I will make a plan to find a workaround to make it work on iOS/MacOS Chrome or Safari. 
But, I have spent a lot of time in other functions, so I do not have time to get into the cause of the problem yet, so I have been delaying the problem for a while. 

Advice and support are welcome if you have advice or have some troubleshooting.

If there is any progress in troubleshooting, I'll post it here.

kclyu, 2018-01-31:
I have committed the js/html files for this issue but it seems to work normally using the latest WebRTC native stack for normal operation. I will close the current issue after deb uploading.

closed 2018-02-01 by kclyu