kclyu/rpi-webrtc-streamer #27: Why does PeerConnection add a fake video track.

sliver-chen, 2018-01-29:
Hi
I'm confused about peerconnection add a fake video track,this generated sdp does not include
video unit?
Or say that,i'm trying to create peerconnection in my demo and i add local video track just like webrtc offical demo do that.but when i send generated sdp to remote,remote set sdp failed and report error just like "failed to set video send codecs".So i try to find a answer.
Could tell me about it?
best regards.

sliver-chen, 2018-01-29:
```
rtc::scoped_refptr<webrtc::VideoTrackInterface> video_track(
        peer_connection_factory_->CreateVideoTrack(
            kVideoLabel,
            peer_connection_factory_->CreateVideoSource(new cricket::FakeVideoCapturer(false),
                    nullptr)));
```
rpi-webrtc-streamer create video track like above.
but i do this ny using webrtc::VideoCaptureFactory::CreateDeviceInfo to enum device info,
then  capturer = factory.Create(cricket::Device(name, 0));

best regards.

kclyu, 2018-01-29:
```
    rtc::scoped_refptr<webrtc::VideoTrackInterface> video_track(
        peer_connection_factory_->CreateVideoTrack(
            kVideoLabel,
            peer_connection_factory_->CreateVideoSource(new cricket::FakeVideoCapturer(false),
                    nullptr)));
```
The reason I just used FakeVideoCaptuer is that the built-in Video Capturer of the WebRTC native stack will not be used in RWS.  When the V4l2 module runs on Raspberry PI, WebRTC native statck uses this as the VideoCapturer and opens /dev/video first. So, This is used to prevent this case because the camera can not be used in MMAL when /dev/video is already opened.

BR,

sliver-chen, 2018-01-29:
thanks to your reply.
Any advice to my error code about "failed to set send video codecs?",i don't know how to 
deal with it.
local peerconnection run in linux and remote peerconnection run in android.
remote peerconnection report above error when set remote sdp.
Anyway,thank for your reply.

kclyu, 2018-01-29:
Have you tried the Testing URL http: //x.x.x.x:8889/ native-peerconnection / ?
The source is in web-root/native-peerconnction/. Please check the Testing URL at first.

sliver-chen, 2018-01-29:
let me see see.
thanks.

closed 2018-02-01 by kclyu