kclyu/rpi-webrtc-streamer #28: Integration with Kerberos.io

cedricve, 2018-02-07:
Hey @kclyu,

Amazing work you've did. I would like to have this project integrated in the kerberos.io project (https://github.com/kerberos-io). Are you interested to talk about this?

Thanks
Cédric

kclyu, 2018-02-08:
Thank you for your interest.
As you can see, RWS is an open source, so you can use RWS as you like. Most of the included licenses are BSD style licenses, so you only need to comply with the license of the packages.

I do not know if you want the packaging (or customization) needed for service on the site (https://kerberos.io/). If you give me a little more detail, it is appropriate to talk according to the matter.

Thanks,
Lyu.

closed 2018-02-12 by kclyu

cedricve, 2018-02-12:
hey @kclyu, sorry I missed your response. Can we discuss over mail? 
cedric@kerberos.io

Thanks Lyu

kclyu, 2018-02-13:
Final-Recipient: rfc822; cedric@kerberos.io
Action: failed
Status: 5.7.1
Remote-MTA: dns; mxb.mailgun.org. (54.210.206.63, the server for the domain kerberos.io.)
Diagnostic-Code: smtp; 550 5.7.1 Relaying denied
Last-Attempt-Date: Tue, 13 Feb 2018 00:17:49 -0800 (PST)

kclyu, 2018-02-13:
My e-mail address is lyu.keun.chang@gmail.com