kclyu/rpi-webrtc-streamer #29: Where is the configuration files if installed with deb?

rexlow, 2018-02-08:
I got everything working by installing with deb file. But there are a few tweaks I need to make so that the video quality could be better. Went through `/etc` but I could not find anything related there.

kclyu, 2018-02-08:
The rws config files are located in /opt/rws/ etc. The current config settings related to video quality are not available except for the realtime related settings. I'm curious to know what kind of setup you are trying to do.

rexlow, 2018-02-08:
Cool! Was able to locate the files with this command.
```
dpkg --search rws | grep conf
```

I am planning to run a face recognition model on top of the real-time video feed actually. Will need to access a fair quality of frames before downsampling them. 

Can you tell me how to adjust the configuration in order to get the best quality possible? I am using Raspberry Pi 3 Model B with Raspberry Pi 8MP Camera Board V2.

I looked through the config files and adjusted the bitrates, but there isn't much to do there. Please correct me if I'm wrong.

Thanks, 
Rex.

kclyu, 2018-02-09:
In media_config.conf, set `use_dynamic_video_resolution=false` and manually enter the resolution in `fixed_video_resolution` to find the desired video quality. The resolution set in `fixed_video_resolution` must have the same resolution in video_resolution_list_x_x. 

If the resolution does not improve the image quality, the maximum bandwidth at high resolution in `chrome://webrtc-logs/` will be the maximum bandwidth between browser and raspberry pi.

Thanks, 
Lyu.

rexlow, 2018-02-09:
Thanks for the suggestion. Reconfigure but did not notice any significant changes, though its running on low memory which is good.

rexlow, 2018-03-29:
Was busy for the past 2 months. Revisited my work and found that the reason why I was not able to set the resolution. Because `1920x1080` is `16x9` instead of `4x3`, which is why RWS will ignore even if I edited other flags.

Set

```
resolution_4_3_enable=false
```

to enable any resolution inside the `video_resolution_list_16_9` list.

closed 2018-03-29 by rexlow