kclyu/rpi-webrtc-streamer #32: issue about peer between linux and android

sliver-chen, 2018-02-26:
hi kclyu.

last time i meet a problem in my application that android failed set sdp from remote linux ,
and log prompt me that "failed to set video send codecs".
now i have solved this problem by remove codec number 97 98 from sdp created by linux pc.

i found that is very easy leads to sdp negotiation failure between cross-platform peer connection,
and i want to know that if you meet this problem and how does you solve it.

Best regards.

kclyu, 2018-02-26:
Hi,

My answer does not seem to help much. I have not seen many errors with 'failed to set video send codecs'.  If you want to get answers to other solutions or other ways to avoid it, you may need to add a description of the environment you are running (os, tool, webrtc stack version, error logging message) and how to test or run it.
If you are testing with rws or using a similar program on linux, there are html and js files under `web-root/native-peerconnection', so you can refer to how rws works.

BR,

sliver-chen, 2018-02-27:
hi kclyu
I am not using rws,i am trying to let a linux device connect to android client in my project.
i have deal above problem about sdp by process sdp before call setLocalDescription and connect them.

what i want to know by this issue is that if you meet the sdp negotiate conflicts,
because it is easy to apper that different platforms support different codecs and just support 
section of them.

zero863, 2018-04-02:
Hi, I'm studying WebRTC recently, I wonder if I can make a Raspberry Pi connect to PC(Windows system) using WebRTC. Does it have to involve modifying the underlying code？Is it very difficult? Can you give me an answer?
Best regards.

kclyu, 2018-04-02:
RWS is a unidirectional program to send raspberry pi's video feed to another client. There is currently no function related to video decoding. You have to do a lot of coding to create it.

In addition, any issues not related to this issue should be made by creating other issues.

closed 2018-05-01 by kclyu