kclyu/rpi-webrtc-streamer #34: Attempting to run on docker

lucabelluccini, 2018-03-03:
Hello,
I've started playing with your project yesterday on my RPi 3.

My objective is to create an electric RC car built out of lego completely controlled via WebRTC.

With the aim of making things modular, I would like to contribute to the project adding 2 Dockerfiles:
1. a docker image to build the deb package or the binaries
2. a docker image to run rws in a container

I'm starting with a Docker image to run rws in a container.

This is my first attempt:
```
FROM resin/rpi-raspbian

RUN apt-get update -y 

ADD https://github.com/kclyu/rpi-webrtc-streamer-deb/raw/master/rws_0.73.1_armhf.deb /tmp/

RUN dpkg -i /tmp/rws_0.73.1_armhf.deb

RUN apt-get install libasound2

ENV LD_LIBRARY_PATH /opt/vc/lib

RUN sed -i 's/wss:/ws:/g' /opt/rws/web-root/native-peerconnection/js/main_rws.js

EXPOSE 8888
EXPOSE 8889

CMD /opt/rws/webrtc-streamer --verbose
```

You can try to build & run it using the following:

```
docker build -t rpi-webrtc-streamer:1.0 .
docker run --rm -p 8889:8889 -p 8888:8888 --device /dev/vchiq -v /opt/vc/lib:/opt/vc/lib rpi-webrtc-streamer:1.0
```

My current status:

- The audio is not yet supported, and it will require to expose the audio device to the docker container plus add the configuration files for alsa
- The configuration files for rws are the default ones, and I'll try to make them exposed via a volume (you can already do that using the docker run params)
- I have to patch the secure web socket in the Javascript file as I get an error when connecting to WSS endpoint (I tried both from localhost, 127.0.0.1 or the local ip in my network)
- When I try to connect from the RPi locally, I manage to see the video but the STUN pings seem to not work correctly: ```
(port.cc:1624): Jingle:Conn[0x74e182d0:audio:OzvntcD9:1:0:local:udp:172.17.0.x:53571->r35AAXMd:1:2113937151:local:udp:192.168.1.x:33842|C--I|0|0|9079290933588934143|-]: Sent STUN ping, id=694f797068617331625a744b, use_candidate=1, nomination=0
(network.cc:838): Connect failed with 99
(port.cc:1624): Jingle:Conn[0x74e182d0:audio:OzvntcD9:1:0:local:udp:172.17.0.x:53571->r35AAXMd:1:2113937151:local:udp:192.168.1.x:33842|C--I|0|0|9079290933588934143|-]: Sent STUN ping, id=4c322f6f4971413775653632, use_candidate=1, nomination=0
(port.cc:1624): Jingle:Conn[0x74e182d0:audio:OzvntcD9:1:0:local:udp:172.17.0.x:53571->r35AAXMd:1:2113937151:local:udp:192.168.1.x:33842|C--I|0|0|9079290933588934143|-]: Sent STUN ping, id=6734376233364e2b4e4b734c, use_candidate=1, nomination=0
(bitrate_allocator.cc:88): Current BWE 2042252
(port.cc:1624): Jingle:Conn[0x74e182d0:audio:OzvntcD9:1:0:local:udp:172.17.0.x:53571->r35AAXMd:1:2113937151:local:udp:192.168.1.x:33842|C--I|0|0|9079290933588934143|-]: Sent STUN ping, id=7032524451384a465541704b, use_candidate=1, nomination=0
(network.cc:838): Connect failed with 99

...

(port.cc:1369): Jingle:Conn[0x74e182d0:audio:OzvntcD9:1:0:local:udp:172.17.0.x:53571->r35AAXMd:1:2113937151:local:udp:192.168.1.x:33842|C--I|0|0|9079290933588934143|-]: Timed out after 15218 ms without a response, rtt=6000
(p2ptransportchannel.cc:1597): Jingle:Channel[audio|1|RW]: Transport channel state changed from 1 to 2
(transportcontroller.cc:1051): audio TransportChannel 1 state changed. Check if state is complete.
(port.cc:1671): Connection deleted with number of pings sent: 25
(p2ptransportchannel.cc:2015): Jingle:Channel[audio|1|RW]: Removed connection 0x74e182d0 (1 remaining)
(peerconnection.cc:4871): Changing to ICE completed state because all transports are complete.
(peerconnection.cc:3064): Changing IceConnectionState 2 => 3
(network.cc:838): Connect failed with 99
(bitrate_allocator.cc:88): Current BWE 2080665
(network.cc:838): Connect failed with 99
(network.cc:838): Connect failed with 99
(bitrate_allocator.cc:88): Current BWE 2565925
(network.cc:838): Connect failed with 99
(network.cc:838): Connect failed with 99
(bitrate_allocator.cc:88): Current BWE 2855812
(network.cc:838): Connect failed with 99
(network.cc:838): Connect failed with 99
(network.cc:838): Connect failed with 99
(bitrate_allocator.cc:88): Current BWE 3020180
(network.cc:838): Connect failed with 99
(network.cc:838): Connect failed with 99

```
- When I try to access from another client (a Windows 10 machine running Chrome 64 or Firefox 58), I get no video, but no error on the console. The logs of the process can be checked at this [link](https://gist.github.com/lucabelluccini/3c4bfab3d1adf946ac29c6752d121e8c)

kclyu, 2018-03-03:
Hi, 
First of all thank you for trying docker related work. I do not know if it will help because I have not worked on the docker yet. But, there are some questions that I have answered.

- public ip address requires https to be used.
  Since certificate related development and testing in rws has not been done, rws now use reverse proxy to use nginx for https as document ([running-rws-with-nginx](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_rws_setup.md#running-rws-with-nginx)) . WebRTC must use https if the signaling server(rws) ip address is not a private ip address.  
If you configure RWS to run in a private network, you can use http/ws, To configure it for use with public ip address, you need to configure https/wss related certificate and dns. It's getting a bit more complicated.
To make the configuration easier on the docker, it would be better to use it within the private ip address.

- WebRTC udp port range
8888 and 8889 are tcp ports for webrtc signaling of rws. In order to transmit RTP media, WebRTC uses UDP port, usually 1025 ~ 65535 (that is, available random port number). I do not know if there is any problem in using the port range mentioned above in the docker.

- About rws log
First, the operation may not be normal even if there is no error. Most of the log messages are log messages coming from the WebRTC  native stack. We can not control them directly from RWS. So, first of all, you need to create a log that runs normally and compare it when it does not work.

BR,
Lyu.

lucabelluccini, 2018-03-03:
Hello @kclyu, thank you for your quick feedback.

I am not using any public ip address. I tried local ips, but it seems the js code was not detecting the local ip properly (?).
Still, I could setup nginx as a separate container and connect them.

Regarding the ports, if the connection is initiated from inside docker, it is not a problem. The exposed ports are ports where an external client could bind to.

I'll try to install it on the RPi as native daemon and compare the logs.

Thank you for the help, I'll keep you posted.

Best regards,
Luca

lucabelluccini, 2018-03-14:
Hello, I managed to make it work. It was linked to the network interface of Docker.

I've refined the Dockerfile:
```
FROM resin/rpi-raspbian

ARG RWS_DEB_URL=https://github.com/kclyu/rpi-webrtc-streamer-deb/raw/master/rws_0.73.1_RaspiZeroW_armhf.deb

RUN apt-get update -y

ADD $RWS_DEB_URL /tmp/

RUN dpkg -i /tmp/*.deb

RUN apt-get install libasound2

ENV LD_LIBRARY_PATH /opt/vc/lib

EXPOSE 8888
EXPOSE 8889

CMD /opt/rws/webrtc-streamer --verbose
```

It is possible to build the image for the RPi Zero or others passing the deb URL or local path
```
docker build --build-arg RWS_DEB_URL=http://... -t rpi-webrtc-streamer-zero:1.0 .
```

Running with the following command makes it work properly:
```
docker run --rm -it --network host -p 8889:8889 -p 8888:8888 --device /dev/vchiq -v /opt/vc/lib:/opt/vc/lib rpi-webrtc-streamer-zero:1.0
```

I'll try to submit a PR which would allow to pass configuration files.

closed 2018-07-04 by kclyu