kclyu/rpi-webrtc-streamer #35: Ability to set action if motion is detected

rexlow, 2018-03-28:
Is it possible to define our own action? 

For instance, I would like to be able to execute a python script if a motion is detected.

kclyu, 2018-03-29:
There is currently no ability to call python scripts directly from RWS or to pass events to other processes. The ability to deliver events  from rws will be planned after further enhancements to motion detection.

Currently, it is only possible to continuously monitor the directory where the motion file is stored and to deliver the event when a new video file is created. You can use [TelegramBot](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_TelegramBot.md) and you can refer to it as an example.

rexlow, 2018-03-29:
Thanks for the clarification. My application intends to capture and process feed if a motion is detected.  I guess I will have to create a python script to capture the feed through a socket connection instead, then only process the feed.

closed 2018-03-29 by rexlow