kclyu/rpi-webrtc-streamer #36: RWS exited if port 8888 is accessed

rexlow, 2018-03-29:
I was trying to obtain the stream through direct socket via port 8888. As soon as I visit port 8888, RWS shuts down. 

It will run again after some time, but may I know the reason behind this?

````
● rws.service - Rpi WebRTC Streamer
   Loaded: loaded (/lib/systemd/system/rws.service; enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2018-03-29 16:35:04 +08; 507ms ago
  Process: 20017 ExecStart=/opt/rws/webrtc-streamer --log /opt/rws/log (code=exited, status=255)
 Main PID: 20017 (code=exited, status=255)

Mar 29 16:35:04 raspberrypi systemd[1]: Unit rws.service entered failed state.
````

kclyu, 2018-03-29:
First of all, I think you need to send me a log of what the problem is.

the following commands will disable rws in systemd. and you can run the webrtc-streamer manually.
Please don't forgot to enable rws it again, when you finished the log capture.
```
sudo systemctl disable rws
sudo systemctl stop rws
cd /opt/rws
./webrtc-streamer --verbose
```

By the way, I was able to run without problems with the android webrtc app. You're not trying to get a video feed directly from the 8888 port, right?

The 8888 port is a signaling channel for TCP.

rexlow, 2018-03-29:
The web example is working correctly, but when I try to access the stream with an android webrtc app called [steamer](https://play.google.com/store/apps/details?id=com.flashphoner.wcsexample.streamer) with the following address

```
ws://192.168.0.146:8889/rws/ws
```

Nothing was returned.

Below is part of the log messages I found on the `log` directory, not sure if its helpful.

```
[4938:698] [21203] (send_side_congestion_controller.cc:440): SignalNetworkState Down
[4938:698] [21207] (paced_sender.cc:108): PacedSender paused.
[4938:698] [21207] (delay_based_bwe.cc:316): BWE Setting start bitrate to: 300000
[4938:698] [21203] (paced_sender.cc:306): ProcessThreadAttached 0x0x74f12778
[4938:700] [21202] (opensslidentity.cc:44): Making key pair
[4938:700] [2804] (streamer.cc:448): Audio options: AudioOptions {level_control: true, }
[4938:703] [21202] (opensslidentity.cc:92): Returning key pair
[4938:703] [21202] (opensslidentity.cc:99): Making certificate for WebRTC
[4938:708] [2804] (app_ws_client.cc:136): New WebSocket Name: 59033522
[4938:709] [21202] (opensslidentity.cc:146): Returning certificate
[4938:710] [21204] (mediasession.cc:353): Duplicate id found. Reassigning from 102 to 127
[4938:712] [21202] (p2ptransportchannel.cc:478): Set ping most likely connection to 0
[4938:712] [21202] (p2ptransportchannel.cc:498): Set presume writable when fully relayed to 0
[4938:712] [21203] (webrtcvoiceengine.cc:1458): Setting voice channel options: AudioOptions {audio_jitter_buffer_max_packets: 50, audio_jitter_buffer_fast_accelerate: false, }
[4938:712] [21203] (webrtcvoiceengine.cc:323): WebRtcVoiceEngine::ApplyOptions: AudioOptions {audio_jitter_buffer_max_packets: 50, audio_jitter_buffer_fast_accelerate: false, }
[4938:712] [21203] (webrtcvoiceengine.cc:511): NetEq capacity is 50
[4938:712] [21203] (webrtcvoiceengine.cc:517): NetEq fast mode? 0
[4938:712] [21203] (webrtcvoiceengine.cc:535): Delay agnostic aec is enabled? 0
[4938:712] [21203] (webrtcvoiceengine.cc:545): Extended filter aec is enabled? 0
[4938:712] [21203] (webrtcvoiceengine.cc:555): Experimental ns is enabled? 0
[4938:712] [21203] (webrtcvoiceengine.cc:561): Intelligibility Enhancer is enabled? 0
[4938:712] [21203] (webrtcvoiceengine.cc:573): Level control: 0
[4938:712] [21203] (audio_processing_impl.cc:728): Level controller activated: 0
[4938:712] [21203] (audio_processing_impl.cc:735): Highpass filter activated: 1
[4938:712] [21203] (audio_processing_impl.cc:749): Gain Controller 2 activated: 0
[4938:712] [21203] (webrtcvoiceengine.cc:1476): Set voice channel options. Current options: AudioOptions {audio_jitter_buffer_max_packets: 50, audio_jitter_buffer_fast_accelerate: false, }
[4938:712] [21203] (channel.cc:116): Created channel for audio
[4938:713] [21202] (channel.cc:296): Setting RTP Transport for audio on audio transport 0x75901ab8
[4938:713] [21203] (call.cc:1115): Transport audio is disconnected
[4938:713] [21202] (p2ptransportchannel.cc:478): Set ping most likely connection to 0
[4938:713] [21202] (p2ptransportchannel.cc:498): Set presume writable when fully relayed to 0
[4938:713] [21203] (webrtcvideoengine.cc:550): CreateChannel. Options: VideoOptions {}
[4938:714] [21203] (channel.cc:116): Created channel for video
[4938:714] [21202] (channel.cc:296): Setting RTP Transport for video on video transport 0x7598a5c8
[4938:715] [21203] (call.cc:1115): Transport video is disconnected
[4938:715] [21202] (transportcontroller.cc:829): Set local transport description on audio
[4938:715] [21202] (p2ptransportchannel.cc:399): Set ICE ufrag: FJbc pwd: PEmOZs+cdDXiSjLgbTPaE7oW on transport audio
[4938:715] [21202] (transportcontroller.cc:829): Set local transport description on video
[4938:715] [21202] (p2ptransportchannel.cc:399): Set ICE ufrag: FJbc pwd: PEmOZs+cdDXiSjLgbTPaE7oW on transport video
[4938:715] [21204] (peerconnection.cc:3107): Session: 8374093060510362566 Old state: kStable New state: kHaveLocalOffer
[4938:715] [21203] (channel.cc:1225): Setting local voice description
[4938:715] [21203] (webrtcvoiceengine.cc:1314): WebRtcVoiceMediaChannel::SetRecvParameters: {codecs: [AudioCodec[111:opus:48000:0:2], AudioCodec[103:ISAC:16000:0:1], AudioCodec[9:G722:8000:0:1], AudioCodec[102:ILBC:8000:0:1], AudioCodec[0:PCMU:8000:0:1], AudioCodec[8:PCMA:8000:0:1], AudioCodec[105:CN:16000:0:1], AudioCodec[13:CN:8000:0:1], AudioCodec[110:telephone-event:48000:0:1], AudioCodec[113:telephone-event:16000:0:1], AudioCodec[126:telephone-event:8000:0:1]], extensions: [{uri: urn:ietf:params:rtp-hdrext:ssrc-audio-level, id: 1}]}
[4938:716] [21203] (webrtcvoiceengine.cc:1486): Setting receive voice codecs.
[4938:716] [21203] (webrtcvoiceengine.cc:1756): AddSendStream: {id:audio_label;ssrcs:[1040962283];ssrc_groups:;cname:VeKMNk3iGE39+ADD;sync_label:stream_label}
[4938:716] [21203] (neteq_impl.cc:108): NetEq config: sample_rate_hz=16000, enable_post_decode_vad=true, max_packets_in_buffer=0, background_noise_mode=2, playout_mode=0, enable_fast_accelerate=false, enable_muted_state= true
[4938:716] [21203] (audio_coding_module.cc:462): Created
[4938:716] [21203] (audio_send_stream.cc:135): AudioSendStream: 1040962283
[4938:717] [21203] (audio_send_stream.cc:201): AudioSendStream::ConfigureStream: {rtp: {ssrc: 1040962283, extensions: [], nack: {rtp_history_ms: 0}, c_name: VeKMNk3iGE39+ADD}, send_transport: (Transport), min_bitrate_bps: -1, max_bitrate_bps: -1, send_codec_spec: <unset>}
[... cut ...]
[5611:714] [23685] (probe_bitrate_estimator.cc:149): Probing successful [cluster id: 3] [send: 65064 bytes / 11 ms = 5914.91 kb/s] [receive: 65184 bytes / 11 ms = 5925.82 kb/s]
[5611:714] [23685] (probe_bitrate_estimator.cc:149): Probing successful [cluster id: 3] [send: 74416 bytes / 15 ms = 4961.07 kb/s] [receive: 74536 bytes / 15 ms = 4969.07 kb/s]
[5611:714] [23685] (probe_bitrate_estimator.cc:149): Probing successful [cluster id: 3] [send: 83768 bytes / 16 ms = 5235.5 kb/s] [receive: 83888 bytes / 15 ms = 5592.53 kb/s]
[5612:935] [2%
```

kclyu, 2018-03-29:
Not all webrtc-enabled apps have direct access to rws. Because signaling has different specifications, 
If you want to create an app, you need to be compatible with signaling supported by rws.

rexlow, 2018-03-29:
I see. Sorry, I am relatively new to streaming. I am planning to build a python script that captures the frame, but unfortunately, I'm unable to access the stream just yet, what will you suggest to go about?

Here's a script I made to load a `rtsp` stream. When I try to access the stream from `rws`, it could not read the video stream. Tried with different protocols to no avail.

```
import cv2
import imutils
from imutils.video import VideoStream

stream = "rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov"
stream_rws = "rstp://192.168.0.146:8889/native-peerconnection/"

vs = VideoStream(src=stream).start()

while True:
    frame = vs.read()

    cv2.imshow('frame', frame)

    if cv2.waitKey(1) != -1:
        break
vs.release()
cv2.destroyAllWindows()
```

kclyu, 2018-03-29:
As mentioned, the 8888 port is the tcp signaling channel of the WebRTC android sample app. That is, it is a channel that negates SDP with json.

When the signaling is completed with 8888 port, streaming starts with Secured RTP through negotiated udp port. SDP negotiation and SRTP is WebRTC specifiation. That is, it uses WebRTC protocol instead of rtsp protocol. it menas RWS does not supports rtsp protocol.

Let me know what you are going to do with rws and I will let you know if there is a way I can.

rexlow, 2018-03-29:
Thanks for your speedy feedback, really appreciate it. 

My project intends to detect the presence of a human by analyzing video feed from the Raspberry Pi. The human detection part is already done and now I need to have access to the video frame from a local computer.

Essentially, the rPi will act as a video streamer and the computer will do the processing part.

kclyu, 2018-03-29:
It would be better to start from below. I did not know whether it should be correct or not.

https://sourcey.com/recording-native-webrtc-streams-with-libsourcey-and-ffmpeg/
https://github.com/sourcey/libsourcey/
https://webrtchacks.com/video_replay/
https://deepstreamhub.com/tutorials/protocols/webrtc-video-manipulation/

and, I think you can find what you need with googling. 'video frame capture from webrtc stream'
I wish you good luck.

kclyu, 2018-03-30:
I will close this issue. Please reopen if necessary.

closed 2018-03-30 by kclyu