kclyu/rpi-webrtc-streamer #37: Segfault at FullHD resolution

antimof, 2018-04-05:
Hello.
Every time when camera reaches FullHD resolution (1920x1080) with bitrate above 8 mbit/s i get segfault from vc.ril.video_encode thread (gdb). I turned on debug messages in mmal_wrapper.cc and get the following output:
------------------------------
(mmal_wrapper.cc:148): frame_buf_pos : 131120, buffer length: 5843

#
# Fatal error in mmal_wrapper.cc, line 151
# last system error: 0
# Check failed: (int)(frame_buf_pos_ + buffer->length) < size_
# 
#
Aborted
-----------------------------------------
What can a reason for such bahavior?
Thank you in advance....

kclyu, 2018-04-05:
This message occurs when the size of the actual frame is larger than the buffer size already defined.
The size of the buffer is larger than the expected frame size, but I'm a little unexpected.

The attached webrtc-streamer binary [webrtc-streamer.gz](https://github.com/kclyu/rpi-webrtc-streamer/files/1880049/webrtc-streamer.gz) is an executable that changes the buffer size by 1.5 times and prints out the frame size for each frame.
Please run ./webrtc-streamer --verbose from the console and attach the log contents.

In addition, I would like to ask you to let me know the version of raspberry pi and camera.

Thanks,
Lyu.

kclyu, 2018-04-05:
The message below is a message from running the attached binary at 1920x1080 @ 30.
( raspberry pi 2b , camera version is 1.3 )
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 14119
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 20441
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13790
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13454
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13280
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13349
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 12951
(mmal_wrapper.cc:152): frame_buf_pos : 65584, buffer length: 36044
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 22954
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 10877
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 6459
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 4142
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 6219
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 6117
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 6466
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 11478
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 10986
(network.cc:838): Connect failed with 99
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 10717
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 10615
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 10537
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 10330
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 10158
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 9741
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 9937
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13488
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13764
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13529
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13273
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13463
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13228
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13304
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13089
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13274
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13178
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13328
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13666
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13543
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13394
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13466
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 18852
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13192
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 12953
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13027
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13117
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13012
(mmal_wrapper.cc:152): frame_buf_pos : 0, buffer length: 13399

closed 2018-04-24 by kclyu