kclyu/rpi-webrtc-streamer #38: auto reconnect

allen2000github, 2018-05-05:
Sometimes the video is freezed, "Connect" button is still disabled and "Disconnect" button is enabled.
I had to click "Disconnect" then "Connect" to recover the video.
Is there any chance it could recover the connection automatically?

kclyu, 2018-05-05:
Can you upload javascript console message from web browser when you see the same symptom?

It is expected that 'ICE connection state' will be 'closed' in log, and if it is correct, it will be possible to add function using javascript.
If you can make your own edits, it would be helpful to refer to https://webrtc.github.io/samples/src/content/peerconnection/states/.

  RWS expects 'PeerConnectionState' to be implemented in Native source package. When the implementation is completed by Google, WebSocket connction will be closed directly from RWS according to the status change, and the browser will be changed to Disconnect status.
Thanks, 
Lyu.

allen2000github, 2018-05-06:
Thanks for you reply. Finally I add some code in file "peerconnection_client.js" which might be helpful.

PeerConnectionClient.prototype.onIceConnectionStateChanged_ = function() {
    ...
    if ((this.peerConnection_.iceConnectionState === 'failed' || this.peerConnection_.iceConnectionState === 'closed') && websocketSignalingChannel.disconnectButton_.disabled === false) {
        websocketSignalingChannel.doSignalingDisconnnect();
		setTimeout(function(){websocketSignalingChannel.doSignalingConnect();},1000);
    }
};

closed 2018-05-12 by kclyu