kclyu/rpi-webrtc-streamer #43: Connect to Janus Gateway

BarryReid, 2018-05-30:
Is there a way to get the Pi to stream to a Janus WebRTC Gateway?

kclyu, 2018-05-30:
RWS currently does not support interworking with janus. So far, I have been postponing interworking with janus because I am concentrating on other tasks of RWS.

If you can develop it yourself, there are two ways to do it. One is to add the app_channel directly in RWS and the other is to use [janus-client](https://github.com/sipwise/janus-client), which uses janus's websocket interface. Using janus-client, it seems to be possible to exchange signaling messages to websocket between rws (message exchange example of rws, see [rws proxy](https://github.com/kclyu/rpi-webrtc-streamer-frontend/tree/master/proxy)) and janus.

vishrutiPawar, 2020-11-06:
Does it support internetworking with something else if not Janus ?

kclyu, 2020-11-06:
Not yet supported.
 
As a manager function of rws (a function to manage multiple rws together with nMDS), it is currently under development as another project and plans to create another repo once a certain level of alpha level function is developed.
Among the features under development, the interworking with Janus VideoRoom will be included.

To talk about Janus-related development in a little detail, the Janus integration part currently under development is considering two methods. One method is to allocate one videoroom per rws. To use this function, you can directly link with VideoRoom of Janus gateway through Janus Web API. The other method is to use rws websocket signaling to connect with janus video room. It will be added to the rws websocket api so that it can be interworked with janus videoroom.  The first one of the features mentioned above is currently working at the alpha version.