kclyu/rpi-webrtc-streamer #44: Trouble compiling for pizero

clockwise9, 2018-06-03:
I'm running into the following compile error when running make on `rpi-web-streamer`:
```
arm-linux-gnueabihf-g++: error: unrecognized argument in option ‘-march=x86-64’
```
Full output
```
justin@justin-VirtualBox:~/Workspace/rpi-webrtc-streamer/src$ make
arm-linux-gnueabihf-g++ --sysroot=/opt/rpi_rootfs    -I. -DINSTALL_DIR=\"/opt/rws\" -Wno-deprecated-declarations -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -funwind-tables -fPIC -pipe -B../third_party/binutils/Linux_x64/Release/bin -pthread -fcolor-diagnostics -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -no-canonical-prefixes -m64 -march=x86-64 -Wall -Werror -Wextra -Wimplicit-fallthrough -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-covered-switch-default -Wno-unneeded-internal-declaration -Wno-inconsistent-missing-override -Wno-undefined-var-template -Wno-nonportable-include-path -Wno-address-of-packed-member -Wno-unused-lambda-capture -Wno-user-defined-warnings -Wno-enum-compare-switch -Wno-null-pointer-arithmetic -Wno-ignored-pragma-optimize -O0 -fno-omit-frame-pointer -g2 -gsplit-dwarf -ggnu-pubnames -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wc++11-narrowing -Wimplicit-fallthrough -Wthread-safety -Winconsistent-missing-override -Wundef -Wunused-lambda-capture -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/gtk-3.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/at-spi2-atk/2.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/at-spi-2.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/dbus-1.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/lib/x86_64-linux-gnu/dbus-1.0/include -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/gtk-3.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/gio-unix-2.0/ -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/cairo -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/pango-1.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/harfbuzz -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/pango-1.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/atk-1.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/cairo -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/pixman-1 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/freetype2 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/libpng16 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/gdk-pixbuf-2.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/libpng16 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/glib-2.0 -isystem/home/justin/Workspace/webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/lib/x86_64-linux-gnu/glib-2.0/include -Wno-multichar -Wno-unused-but-set-variable -fPIC -O3 -DDEBUG -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__ -D__WEBRTC_DEFAULT__ -DINSTALL_DIR=\"/opt/rws\" -Wall -fvisibility=hidden -pthread -Wno-undefined-bool-conversion -Wno-tautological-undefined-compare -std=gnu++11 -fno-exceptions -fno-rtti -nostdinc++ -isystem../../webrtc/src/buildtools/third_party/libc++/trunk/include -isystem../../webrtc/src/buildtools/third_party/libc++abi/trunk/include --sysroot=../build/linux/debian_sid_amd64-sysroot -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -Wno-multichar -Wno-unused-but-set-variable -fPIC -O3 -DDEBUG -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__ -D__WEBRTC_DEFAULT__ -D__RWS_VERSION__=\"v0.73-11-g0a82b8cb13409a10bc24a8a22f8f3393\" -D__WEBRTC_VERSION__="\"23ffbe78f3b386769b2a2e3fc301df31,Cr-Commit-Position:refs/heads/master@{#22695}\"" -I/home/justin/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DCHROMIUM_BUILD -DFIELDTRIAL_TESTING_ENABLED -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DCR_CLANG_REVISION=\"328575-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_DEBUG -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DWTF_USE_DYNAMIC_ANNOTATIONS=1 -D_GLIBCXX_DEBUG=1 -DWEBRTC_ENABLE_PROTOBUF=1 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DUSE_BUILTIN_SW_CODECS -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=1 -DWEBRTC_POSIX -DWEBRTC_LINUX -DHAVE_WEBRTC_VIDEO -DHAVE_WEBRTC_VOICE -I/home/justin/Workspace/webrtc/src  -I../../webrtc/src -I../../webrtc/src/arm_build/gen -I../../webrtc/src/third_party/libyuv/include -I../../webrtc/src/third_party/jsoncpp/overrides/include -I../../webrtc/src/third_party/jsoncpp/source/include  -I/opt/rpi_rootfs/opt/vc/include/interface/vcos/pthreads -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/linux -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/khronos -I/opt/rpi_rootfs/opt/vc/include/interface/khronos/include -I/opt/rpi_rootfs/opt/vc/include/interface/vchiq_arm -I/opt/rpi_rootfs/opt/vc/include -I/opt/rpi_rootfs/opt/vc/include/interface/mmal   -c websocket_server.cc -o websocket_server.o
arm-linux-gnueabihf-g++: error: unrecognized argument in option ‘-march=x86-64’
arm-linux-gnueabihf-g++: note: valid arguments to ‘-march=’ are: armv2 armv2a armv3 armv3m armv4 armv4t armv5 armv5e armv5t armv5te armv6 armv6-m armv6j armv6k armv6s-m armv6t2 armv6z armv6zk armv7 armv7-a armv7-m armv7-r armv7e-m armv7ve armv8-a armv8-a+crc iwmmxt iwmmxt2 native
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-fcolor-diagnostics’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Xclang’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-mllvm’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Xclang’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-instcombine-lower-dbg-declare=0’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-m64’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wimplicit-fallthrough’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wthread-safety’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wheader-hygiene’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wstring-conversion’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wtautological-overlap-compare’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wc++11-narrowing’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wimplicit-fallthrough’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wthread-safety’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Winconsistent-missing-override’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wunused-lambda-capture’
Makefile:74: recipe for target 'websocket_server.o' failed
make: *** [websocket_server.o] Error 1
```

Any ideas?  Likely unrelated, but the only difference I did than your doc was needing to add the following directory to get websockets to build:

```
sh ../mk/config_libwebsockets.sh
...
start building libwebsockets library
[... cut ...]
    /home/justin/tools/rpi_tools/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
```

I added the following link to get this bit working, just stuck on the last build command:
```
cd ~/tools
ln -s ~/Workspace/rpi_rootfs/tools/ rpi_tools
```

Thanks!
Justin

clockwise9, 2018-06-03:
I followed 4.2 and built successfully. I used webrtc_armv6_build_args.gn for the pi zero. 

Any idea what would the make command be getting x86_64 instead of armv6?

kclyu, 2018-06-03:
The makefile in rws use the  build options of webrtc native code package. Since webrtc native build is for x86_64, rws is supposed to use that option in the makefile.

clockwise9, 2018-06-03:
Okay I’ll try rebuilding again.  Where can I find the build options for webrtc?  I assumed they were in webrtc_armv6_build_args.gn.

Is there anyway I can check the architecture post build?

kclyu, 2018-06-03:
The args.gn file is used as the build configuration file for the webrtc native code package when it builds. There is no way to check the build target after the build of webrtc  but we can only verify that the compiler options during the  the webrtc native code package building.

clockwise9, 2018-06-03:
Okay thanks. I’ll repost the output with flags when I re-build (in a couple hours).

clockwise9, 2018-06-03:
Looks like you need to copy `webrtc_armv6_build_args.gn` to `args.gn`?

I copied the README as is, ie
```
cp ~/Workspace/rpi-webrtc-stremer/misc/webrtc_arm_build_args.gn arm_build
```
Should be?
```
cp ~/Workspace/rpi-webrtc-stremer/misc/webrtc_arm_build_args.gn arm_build/args.gn
```

Building now...

clockwise9, 2018-06-03:
Almost there, running into a different build issue now.  I had started this process ages ago, is it possible my webrtc pull is older than your latest master branch?  Again, thanks for the help!

```
justin@justin-VirtualBox:~/Workspace/rpi-webrtc-streamer/src$ make
arm-linux-gnueabihf-g++ --sysroot=/opt/rpi_rootfs    -I. -DINSTALL_DIR=\"/opt/rws\" -Wno-deprecated-declarations -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv6 -mfloat-abi=hard -mfpu=vfp -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -gdwarf-3 -g2 -fvisibility=hidden -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -isystem/opt/rpi_rootfs/usr/include/gtk-3.0 -isystem/opt/rpi_rootfs/usr/include/at-spi2-atk/2.0 -isystem/opt/rpi_rootfs/usr/include/at-spi-2.0 -isystem/opt/rpi_rootfs/usr/include/dbus-1.0 -isystem/opt/rpi_rootfs/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -isystem/opt/rpi_rootfs/usr/include/gtk-3.0 -isystem/opt/rpi_rootfs/usr/include/gio-unix-2.0/ -isystem/opt/rpi_rootfs/usr/include/cairo -isystem/opt/rpi_rootfs/usr/include/pango-1.0 -isystem/opt/rpi_rootfs/usr/include/harfbuzz -isystem/opt/rpi_rootfs/usr/include/pango-1.0 -isystem/opt/rpi_rootfs/usr/include/atk-1.0 -isystem/opt/rpi_rootfs/usr/include/cairo -isystem/opt/rpi_rootfs/usr/include/pixman-1 -isystem/opt/rpi_rootfs/usr/include/freetype2 -isystem/opt/rpi_rootfs/usr/include/libpng12 -isystem/opt/rpi_rootfs/usr/include/gdk-pixbuf-2.0 -isystem/opt/rpi_rootfs/usr/include/libpng12 -isystem/opt/rpi_rootfs/usr/include/glib-2.0 -isystem/opt/rpi_rootfs/usr/lib/arm-linux-gnueabihf/glib-2.0/include -DINSTALL_DIR=\"/opt/rws\" -Wall -fvisibility=hidden -pthread -std=gnu++11 -Wno-narrowing -fno-exceptions -fno-rtti --sysroot=../../../../../../opt/rpi_rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -D__RWS_VERSION__=\"v0.73-18-g70d233c9c4d9dd98d8575132b986893d\" -D__WEBRTC_VERSION__="\"23ffbe78f3b386769b2a2e3fc301df31,Cr-Commit-Position:refs/heads/master@{#22695}\"" -Wno-multichar -Wno-unused-but-set-variable -fPIC -O3 -DDEBUG -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__ -D__WEBRTC_DEFAULT__ -I/home/justin/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DUSE_BUILTIN_SW_CODECS -DWEBRTC_ARCH_ARM -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DHAVE_WEBRTC_VIDEO -DHAVE_WEBRTC_VOICE -I/home/justin/Workspace/webrtc/src  -I../../webrtc/src -I../../webrtc/src/arm_build/gen -I../../webrtc/src/third_party/libyuv/include -I../../webrtc/src/third_party/jsoncpp/overrides/include -I../../webrtc/src/third_party/jsoncpp/source/include  -I/opt/rpi_rootfs/opt/vc/include/interface/vcos/pthreads -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/linux -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/khronos -I/opt/rpi_rootfs/opt/vc/include/interface/khronos/include -I/opt/rpi_rootfs/opt/vc/include/interface/vchiq_arm -I/opt/rpi_rootfs/opt/vc/include -I/opt/rpi_rootfs/opt/vc/include/interface/mmal   -c fakevideocapturer_notaskqueue.cc -o fakevideocapturer_notaskqueue.o
fakevideocapturer_notaskqueue.cc: In member function ‘virtual bool cricket::FakeVideoCapturer::CaptureCustomFrame(int, int)’:
fakevideocapturer_notaskqueue.cc:57:76: error: no matching function for call to ‘cricket::FakeFrameSource::GetFrame(int&, int&, webrtc::VideoRotation&, long long int)’
                                               rtc::kNumMicrosecsPerSec / 30));
                                                                            ^
fakevideocapturer_notaskqueue.cc:57:76: note: candidates are:
In file included from /home/justin/Workspace/webrtc/src/media/base/fakevideocapturer.h:21:0,
                 from fakevideocapturer_notaskqueue.cc:11:
/home/justin/Workspace/webrtc/src/media/base/fakeframesource.h:27:22: note: webrtc::VideoFrame cricket::FakeFrameSource::GetFrame()
   webrtc::VideoFrame GetFrame();
                      ^
/home/justin/Workspace/webrtc/src/media/base/fakeframesource.h:27:22: note:   candidate expects 0 arguments, 4 provided
/home/justin/Workspace/webrtc/src/media/base/fakeframesource.h:29:22: note: webrtc::VideoFrame cricket::FakeFrameSource::GetFrame(int, int, int)
   webrtc::VideoFrame GetFrame(int width, int height, int interval_us);
                      ^
/home/justin/Workspace/webrtc/src/media/base/fakeframesource.h:29:22: note:   candidate expects 3 arguments, 4 provided
fakevideocapturer_notaskqueue.cc:58:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:83: recipe for target 'fakevideocapturer_notaskqueue.o' failed
make: *** [fakevideocapturer_notaskqueue.o] Error 1
```

kclyu, 2018-06-04:
- arm_build/args.gn file name issue
The arm_build/args.gn file name is correct. The README_build document has been modified.

-  WebRTC native code package
Currently RWS master is # 23500 of WebRTC native code Cr-Commit-Position. To match the current WebRTC native code with the latest master, try the following commands.
```
cd ~/Workspace/webrtc/src
git pull
gclient sync
```
If you get errors from git pull or gclient, please download WebRTC native code from scratch.

clockwise9, 2018-06-09:
Sorry to bug you again.  I seem to be getting this error with the latest pull
```
justin@justin-VirtualBox:~/Workspace/webrtc/src$ ninja -C arm_build
ninja: Entering directory `arm_build'
[1/1965] CXX obj/api/audio_options_api/audio_options.o
FAILED: obj/api/audio_options_api/audio_options.o 
arm-linux-gnueabihf-g++ -MMD -MF obj/api/audio_options_api/audio_options.o.d -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=b2165ab47ab9c4cc780db53ace1b08607d68ae31 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DUSE_BUILTIN_SW_CODECS -DWEBRTC_ARCH_ARM -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv6 -mfloat-abi=hard -mfpu=vfp -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -gdwarf-3 -g2 -fvisibility=hidden -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -isystem ../third_party/abseil-cpp -std=gnu++11 -Wno-narrowing -fno-exceptions -fno-rtti --sysroot=../../../../../../opt/rpi_rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../api/audio_options.cc -o obj/api/audio_options_api/audio_options.o
In file included from ../api/optional.h:25:0,
                 from ../api/audio_options.h:16,
                 from ../api/audio_options.cc:11:
../rtc_base/sanitizer.h: In function ‘constexpr bool rtc::sanitizer_impl::IsTriviallyCopyable()’:
../rtc_base/sanitizer.h:101:28: error: ‘is_trivially_copy_constructible’ is not a member of ‘std’
   return static_cast<bool>(std::is_trivially_copy_constructible<T>::value &&
                            ^
../rtc_base/sanitizer.h:101:66: error: expected primary-expression before ‘>’ token
   return static_cast<bool>(std::is_trivially_copy_constructible<T>::value &&
                                                                  ^
../rtc_base/sanitizer.h:101:67: error: ‘::value’ has not been declared
   return static_cast<bool>(std::is_trivially_copy_constructible<T>::value &&
                                                                   ^
../rtc_base/sanitizer.h:102:29: error: ‘is_trivially_copy_assignable’ is not a member of ‘std’
                            (std::is_trivially_copy_assignable<T>::value ||
                             ^
../rtc_base/sanitizer.h:102:64: error: expected primary-expression before ‘>’ token
                            (std::is_trivially_copy_assignable<T>::value ||
                                                                ^
../rtc_base/sanitizer.h:102:65: error: ‘::value’ has not been declared
                            (std::is_trivially_copy_assignable<T>::value ||
                                                                 ^
[2/1965] CXX obj/webrtc_common/common_types.o
FAILED: obj/webrtc_common/common_types.o 
arm-linux-gnueabihf-g++ -MMD -MF obj/webrtc_common/common_types.o.d -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=b2165ab47ab9c4cc780db53ace1b08607d68ae31 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DUSE_BUILTIN_SW_CODECS -DWEBRTC_ARCH_ARM -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv6 -mfloat-abi=hard -mfpu=vfp -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -gdwarf-3 -g2 -fvisibility=hidden -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -isystem ../third_party/abseil-cpp -std=gnu++11 -Wno-narrowing -fno-exceptions -fno-rtti --sysroot=../../../../../../opt/rpi_rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../common_types.cc -o obj/webrtc_common/common_types.o
In file included from ../api/optional.h:25:0,
                 from ../common_types.h:20,
                 from ../common_types.cc:11:
../rtc_base/sanitizer.h: In function ‘constexpr bool rtc::sanitizer_impl::IsTriviallyCopyable()’:
../rtc_base/sanitizer.h:101:28: error: ‘is_trivially_copy_constructible’ is not a member of ‘std’
   return static_cast<bool>(std::is_trivially_copy_constructible<T>::value &&
                            ^
[... cut ...]
                             ^
../rtc_base/sanitizer.h:102:64: error: expected primary-expression before ‘>’ token
                            (std::is_trivially_copy_assignable<T>::value ||
                                                                ^
../rtc_base/sanitizer.h:102:65: error: ‘::value’ has not been declared
                            (std::is_trivially_copy_assignable<T>::value ||
                                                                 ^
ninja: build stopped: subcommand failed.
```

clockwise9, 2018-06-09:
looks like a bug showed up in their CI
https://github.com/vsimon/webrtcbuilds/issues/83

And they updated their build env
https://github.com/vsimon/webrtcbuilds/pull/84/commits/ca4eb9c7b4002d60c3d108b39d0f1a26d979417d

clockwise9, 2018-06-09:
Unsure i this is related either
```
justin@justin-VirtualBox:~/Workspace/webrtc/src$ gn gen arm_build
WARNING at build arg file (use "gn args <out_dir>" to edit):30:20: Build argument has no effect.
rtc_use_openmax_dl=false
                   ^----
The variable "rtc_use_openmax_dl" was set as a build argument
but never appeared in a declare_args() block in any buildfile.

To view all possible args, run "gn args --list <out_dir>"

The build continued as if that argument was unspecified.

Done. Made 471 targets from 157 files in 10539ms
```

kclyu, 2018-06-09:
```
   return static_cast<bool>(std::is_trivially_copy_constructible<T>::value &&
```
The above `is_trivially_copy_constructible` error message is an error message that comes from WebRTC native code using cpp syntax supported by GCC 5.x.

Try using GCC 6.4.1 in [Custom Compiled GCC for Raspberry PI](https://github.com/kclyu/rpi_rootfs#custom-compiled-gcc-for-raspberry-pi).

clockwise9, 2018-06-11:
I got everything built, thanks for the help!  Took me a while, I was on Jessie and had to upgrade to Stretch...

Running the testing I get this error:
```
pi@raspberrypi:~/workspace/rws $ ./webrtc-streamer --verbose
Using config file base path:CWD
(main.cc:171): Using Media Config file: etc/media_config.conf
(main.cc:178): Using Motion Config file: etc/motion_config.conf
(main.cc:199): Direct socket port num: 8888
(direct_socket.cc:77): Failed to listen 0.0.0.0:8888.
(physicalsocketserver.cc:1286): PhysicalSocketServer asked to remove a unknown dispatcher, potentially from a duplicate call to Add.
```
Any ideas

kclyu, 2018-06-11:
you need to make sure that the webrtc-streamer process is already running. or check to see if you are using port 8888.

```
netstat -na | grep 8888
tcp        0      0 0.0.0.0:8888            0.0.0.0:*               LISTEN  
```

clockwise9, 2018-06-11:
I had the precomplied webrtc-streamer running that kicks in on boot.

Now I'm getting an Illegal instruction.  I'll have figure out what's going on here.  I'm on a raspberry pi zero

```
pi@raspberrypi:~/workspace/rws $ ./webrtc-streamer --verbose
Using config file base path:CWD
(main.cc:171): Using Media Config file: etc/media_config.conf
(main.cc:178): Using Motion Config file: etc/motion_config.conf
(main.cc:199): Direct socket port num: 8888
(direct_socket.cc:74): Start listening 0.0.0.0:8888.
(app_channel.cc:64): Using http file mapping : /home/pi/workspace/rws
(app_channel.cc:65): Using motion video mapping : /opt/rws/motion_captured
(app_channel.cc:70): WebSocket port num : 8889
Illegal instruction
```

kclyu, 2018-06-12:
The result of the `Illegal instruction` comes from using an instruction that is not supported by the CPU. I'm wondering if you used 'Custom Compiled GCC for Raspberry PI' in rpi_rootfs. If so, please completely erase libwebsockets and recompile.

```
cd ~/Workspace/src/rpi-webrtc-streamer/lib
rm -fr libwebsockets
sh ../mk/config_libwebsockets.sh 
cd ../src
make clean;make
```
If you are using Ubuntu's arm-linux-gnueabihf cross-compile package, do not use it. It compiles for armv7, but does not compile for armv6 (Raspiberry PI Zero).

Please check the compiler spec as below.
```
arm-linux-gnueabihf-gcc -v
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/home/kclyu/Workspace/rpi_rootfs/tools/gcc-linaro-6.4.1-2017.01-x86_64_arm-linux-gnueabihf/bin/../libexec/gcc/arm-linux-gnueabihf/6.4.1/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: '/home/kclyu/Workspace/_build2/snapshots/gcc.git~linaro~gcc-6-branch/configure' SHELL=/bin/bash --with-mpc=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu --with-mpfr=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu --with-gmp=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu --with-gnu-as --with-gnu-ld --disable-libmudflap --enable-lto --without-included-gettext --enable-nls --enable-libgomp --enable-plugin --enable-gold --disable-sjlj-exceptions --enable-gnu-unique-object --enable-linker-build-id --disable-libstdcxx-pch --enable-c99 --enable-clocale=gnu --enable-libstdcxx-debug --enable-long-long --with-cloog=no --with-ppl=no --with-isl=no --disable-multilib --with-float=hard --with-fpu=vfp --with-mode=arm --with-tune=arm1176jz-s --with-arch=armv6 --enable-threads=posix --enable-multiarch --enable-libstdcxx-time=yes --enable-gnu-indirect-function --with-build-sysroot=/home/kclyu/Workspace/_build2/sysroots/arm-linux-gnueabihf --with-sysroot=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu/arm-linux-gnueabihf/libc --enable-checking=release --disable-bootstrap --enable-languages=c,c++,fortran,lto --build=x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --target=arm-linux-gnueabihf --prefix=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu
```

clockwise9, 2018-06-12:
I downloaded gcc-linaro-6.4.1-2017.01-x86_64_arm-linux-gnueabihf.tar.xz from the Custom Compiled GCC for Raspberry PI section.

I had rebuilt libwebsockets and did a clean rebuild.  Just tried again with the same error :/

Here's my arm-linux-gnueabihf-gcc version info
```
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/home/justin/Workspace/rpi_rootfs/tools/gcc-linaro-6.4.1-2017.01-x86_64_arm-linux-gnueabihf/bin/../libexec/gcc/arm-linux-gnueabihf/6.4.1/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: '/home/kclyu/Workspace/_build2/snapshots/gcc.git~linaro~gcc-6-branch/configure' SHELL=/bin/bash --with-mpc=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu --with-mpfr=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu --with-gmp=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu --with-gnu-as --with-gnu-ld --disable-libmudflap --enable-lto --without-included-gettext --enable-nls --enable-libgomp --enable-plugin --enable-gold --disable-sjlj-exceptions --enable-gnu-unique-object --enable-linker-build-id --disable-libstdcxx-pch --enable-c99 --enable-clocale=gnu --enable-libstdcxx-debug --enable-long-long --with-cloog=no --with-ppl=no --with-isl=no --disable-multilib --with-float=hard --with-fpu=vfp --with-mode=arm --with-tune=arm1176jz-s --with-arch=armv6 --enable-threads=posix --enable-multiarch --enable-libstdcxx-time=yes --enable-gnu-indirect-function --with-build-sysroot=/home/kclyu/Workspace/_build2/sysroots/arm-linux-gnueabihf --with-sysroot=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu/arm-linux-gnueabihf/libc --enable-checking=release --disable-bootstrap --enable-languages=c,c++,fortran,lto --build=x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --target=arm-linux-gnueabihf --prefix=/home/kclyu/Workspace/_build2/builds/destdir/x86_64-unknown-linux-gnu
Thread model: posix
gcc version 6.4.1 20180307 (Linaro GCC 6.4-2018.02~dev)
```

I'm on the a raspberry pi zero, not raspbery pi zeroW if that makes a difference.

kclyu, 2018-06-12:
First of all, there is a problem with the PI.cmake file of rpi_rootfs. When upgrading to libwebsocket 3.0, the option has been enabled and the armv7 option has been added on libwebsockets

Commit out add_compile_option from the rpi_rootfs/PI.cmake file as shown below, and please recompile libwebsockets, and rebuild rws .
```
add_definitions("-DBOOST_COROUTINES_NO_DEPRECATION_WARNING")
#add_compile_options(-mfpu=neon-vfpv4 -mfloat-abi=hard -funsafe-math-optimizations)
```

clockwise9, 2018-06-12:
awesome, got it all working! Thanks for all your help and patience. 

I had looked for -mfpu flags, but only on the rpi-webrtc-streamer build.

Thanks again!

closed 2018-06-12 by clockwise9