kclyu/rpi-webrtc-streamer #49: decrease video latency

adrian23, 2018-07-22:
Hello,
with your program i was able to send vide ( local network ) with 200ms of video delay, Can You please tell me is it possibel to decrease this latency, where to look what to change. I'm using webrtc to control robot thats why its important for me to have as low latency as possible. ThankYou

kclyu, 2018-07-22:
I do not think there is a way to recommend it at this time. RWS can be considered to be optimized for WebRTC h.264 transmission part using WebRTC native code.

I think there are some areas where there is room for improvement in the environmental part, but I have not considered this part yet. 

Some of the things to try is to fix video resolution to low resolution (640x480 or less), to use RTP without using SRTP if you use only in local network, or to use wired LAN instead of wireless network if you can use wired LAN There may be ways.

adrian23, 2018-07-22:
Thank you very much, no wired LAN is not an option since its just driving model car ;) . I will try then with decrease resolution and have it fixed as You recomended. If You have any other ideas or clues  just let me know
ThankYou for your response

closed 2018-09-22 by kclyu

tartavull, 2019-03-23:
@adrian23 I'm just curious, at what resolution and fps did you experience the 200ms latency?

adrian23, 2019-03-24:
https://www.youtube.com/watch?v=iy9wFZaTxFA&feature=youtu.be
finally i have 140 in local network and 170 outside
I think resulution was 320x240 and 512x384 - since its dynamic