kclyu/rpi-webrtc-streamer #52: Internal compiler error

pbafff, 2018-09-14:
Tried to build WebRTC library with your custom compiled GCC but I ran into this error:

`[515/2069] CXX obj/modules/audio_processing/aec3/aec3/reverb_decay_estimator.o
../modules/audio_processing/aec3/reverb_decay_estimator.cc: In member function ‘void webrtc::ReverbDecayEstimator::LateReverbLinearRegressor::Reset(int)’:
../modules/audio_processing/aec3/reverb_decay_estimator.cc:288:10: warning: operation on ‘((webrtc::ReverbDecayEstimator::LateReverbLinearRegressor*)this)->webrtc::ReverbDecayEstimator::LateReverbLinearRegressor::count_’ may be undefined [-Wsequence-point]
   count_ = N > 0 ? count_ = -N * 0.5f + 0.5f : 0.f;
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[697/2069] CXX obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o
FAILED: obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o 
arm-linux-gnueabihf-g++ -MMD -MF obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o.d -DBWE_TEST_LOGGING_COMPILE_TIME_ENABLE=0 -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=b732cc2c870314130b767557b34764f8dd0a3df8 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DUSE_BUILTIN_SW_CODECS -DWEBRTC_ARCH_ARM -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I.. -Igen -I../third_party/abseil-cpp -I../third_party/libyuv/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv6 -mfloat-abi=hard -mfpu=vfp -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -gdwarf-3 -g2 -fvisibility=hidden -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -std=gnu++11 -Wno-narrowing -fno-exceptions -fno-rtti --sysroot=../../../../../../opt/rpi_rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../modules/rtp_rtcp/source/rtp_receiver_video.cc -o obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o
In file included from ../modules/rtp_rtcp/include/rtp_rtcp_defines.h:18:0,
                 from ../modules/rtp_rtcp/source/rtp_receiver_video.h:14,
                 from ../modules/rtp_rtcp/source/rtp_receiver_video.cc:11:
../third_party/abseil-cpp/absl/types/variant.h: In substitution of ‘template<unsigned int I, class T> using variant_alternative_t = typename absl::variant_alternative::type [with unsigned int I = I; T = absl::variant<absl::monostate, webrtc::RTPVideoHeaderVP8, webrtc::RTPVideoHeaderVP9, webrtc::RTPVideoHeaderH264>]’:
../third_party/abseil-cpp/absl/types/variant.h:580:7:   required by substitution of ‘template<class T, unsigned int I, class Tj, typename std::enable_if<(std::is_assignable<Tj&, T>::value && std::is_constructible<Tj, T>::value), void>::type* <anonymous> > absl::variant<T0, Ts ...>& absl::variant<T0, Ts ...>::operator=(T&&) [with T = const absl::variant<absl::monostate, webrtc::RTPVideoHeaderVP8, webrtc::RTPVideoHeaderVP9, webrtc::RTPVideoHeaderH264>&; unsigned int I = <missing>; Tj = <missing>; typename std::enable_if<(std::is_assignable<Tj&, T>::value && std::is_constructible<Tj, T>::value), void>::type* <anonymous> = <missing>]’
../modules/rtp_rtcp/source/rtp_receiver_video.cc:73:60:   required from here
../third_party/abseil-cpp/absl/types/variant.h:580:7: internal compiler error: unexpected expression ‘I’ of kind template_parm_index
       class Tj = absl::variant_alternative_t<I, variant>,
       ^~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
[702/2069] CXX obj/modules/rtp_rtcp/rtp_rtcp/rtp_sender.o
ninja: build stopped: subcommand failed.`

kclyu, 2018-09-14:
What is the current webrtc Cr-Commit Position? (Cr-Commit-position is seen in git log)

Often, an error occurs in compiling the library according to the webrtc commit position. In the WebRTC team, it takes a day or two to fix an error. I think it would be nice to wait for now.

Currently, the proxy_interface branch of RWS uses Cr-Commit-Position: 23500. If you want to use Cr-Commit-Position 23500, please sync and try 23500 commit. (Cr-Commit-Position: 23500 -> a687812c7087ade6b9c92f1fc4d68eafe9180f02)

```
cd ~ / Workspace / webrtc / src
gclient sync -n -D -r a687812c7087ade6b9c92f1fc4d68eafe9180f02
```

pbafff, 2018-09-15:
I tried this and still no luck. I used the method described here: [`https://github.com/ipop-project/ipop-project.github.io/wiki/Build-WebRTC-Libraries-for-Raspberry-Pi-Zero-(Cross-compile-on-Ubuntu)`](url) to compile the WebRTC library but when I try to compile this error happens :
`arm-linux-gnueabihf-g++ --sysroot=/opt/rpi_rootfs    -I. -DINSTALL_DIR=\"/opt/rws\" -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -fcolor-diagnostics -fdebug-prefix-map=/home/andre/workspace/webrtc-checkout/src=. --target=arm-linux-gnueabihf -march=armv6 -mfloat-abi=hard -pthread -mfpu=vfp -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-covered-switch-default -Wno-deprecated-register -Wno-unneeded-internal-declaration -Wno-inconsistent-missing-override -Wno-shift-negative-value -Wno-undefined-var-template -Wno-nonportable-include-path -Wno-address-of-packed-member -O0 -fno-omit-frame-pointer -g2 --sysroot=../../build/linux/debian_wheezy_arm-sysroot -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-strict-overflow -Wimplicit-fallthrough -Wthread-safety -Winconsistent-missing-override -Wundef -Wno-inconsistent-missing-override -DINSTALL_DIR=\"/opt/rws\" -Wall -fvisibility=hidden -pthread -fno-threadsafe-statics -fvisibility-inlines-hidden -Wno-undefined-bool-conversion -Wno-tautological-undefined-compare -std=gnu++11 -fno-rtti -fno-exceptions -Wnon-virtual-dtor -Woverloaded-virtual -D__RWS_VERSION__=\"v0.73-24-g7178a21493b4f64795c6924738f3b714-dirty\" -D__WEBRTC_VERSION__="\",\"" -Wno-multichar -Wno-unused-but-set-variable -fPIC -O3 -DDEBUG -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__ -D__WEBRTC_DEFAULT__ -I/home/andre/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUI_COMPOSITOR_IMAGE_TRANSPORT -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DUSE_PROPRIETARY_CODECS -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DGOOGLE_CHROME_BUILD -DENABLE_MEDIA_ROUTER=1 -DCR_CLANG_REVISION=289944-2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_DEBUG -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DWTF_USE_DYNAMIC_ANNOTATIONS=1 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DEXPAT_RELATIVE_PATH -DHAVE_SCTP -DWEBRTC_ARCH_ARM -DWEBRTC_POSIX -DWEBRTC_LINUX -DWEBRTC_BUILD_LIBEVENT -I/home/andre/Workspace/webrtc/src  -I../../webrtc -I../../webrtc/src/arm_build/gen -I../../webrtc/third_party/libyuv/include -I../../webrtc/build/linux/debian_wheezy_arm-sysroot/usr/include/gio-unix-2.0 -I../../webrtc/build/linux/debian_wheezy_arm-sysroot/usr/include/glib-2.0 -I../../webrtc/build/linux/debian_wheezy_arm-sysroot/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I../../webrtc/third_party/jsoncpp/overrides/include -I../../webrtc/third_party/jsoncpp/source/include  -I/opt/rpi_rootfs/opt/vc/include/interface/vcos/pthreads -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/linux -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/khronos -I/opt/rpi_rootfs/opt/vc/include/interface/khronos/include -I/opt/rpi_rootfs/opt/vc/include/interface/vchiq_arm -I/opt/rpi_rootfs/opt/vc/include -I/opt/rpi_rootfs/opt/vc/include/interface/mmal  -I/home/andre/Workspace/webrtc/src/third_party/abseil-cpp/  -c websocket_server.cc -o websocket_server.o
`
`arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-fcolor-diagnostics’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wheader-hygiene’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wstring-conversion’; did you mean ‘-Wint-conversion’?
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wtautological-overlap-compare’; did you mean ‘-Wtautological-compare’?
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wimplicit-fallthrough’; did you mean ‘-Wno-fallthrough’?
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wthread-safety’; did you mean ‘-fthread-jumps’?
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Winconsistent-missing-override’; did you mean ‘-Wno-suggest-override’?
Makefile:88: recipe for target 'websocket_server.o' failed
make: *** [websocket_server.o] Error 1`

kclyu, 2018-09-15:
When building RWS, it use the cflags and compile options used in WebRTC library build. The above messages appear as if the clang options were used when compiling RWS with the WebRTC library built with clang compiler.

First, you need to check the Cr-Commit-Position of the WebRTC library and use the Cr-Commit-Position or higher WebRTC library used in RWS. Currently, the proxy-interface branch uses the WebRTC library Cr-Commit-Position 23500.

kclyu, 2018-09-15:
The wrong close issue button is pressed, so reopened.

pbafff, 2018-09-16:
I ran `gclient sync -n -D -r a687812c7087ade6b9c92f1fc4d68eafe9180f02` and `git show` tells me that the Cr-Commit-Position is `refs/heads/master@{#24470}`. I was able to get that branch to compile using the RPI tools cross-compiler instead of your custom GCC. But I still can't get past the same error when trying to compile RWS.

```arm-linux-gnueabihf-g++ --sysroot=/home/andre/workspace/webrtc-checkout/rootfs    -I. -DINSTALL_DIR=\"/opt/rws\" -Wno-deprecated-declarations -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -funwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -no-canonical-prefixes --target=arm-linux-gnueabihf -march=armv6 -mfloat-abi=hard -mfpu=vfp -Wall -Wextra -Wimplicit-fallthrough -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-covered-switch-default -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-nonportable-include-path -Wno-user-defined-warnings -Wno-unused-lambda-capture -Wno-null-pointer-arithmetic -Wno-enum-compare-switch -Wno-ignored-pragma-optimize -O0 -fno-omit-frame-pointer -g2 -gsplit-dwarf -ggnu-pubnames -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wc++11-narrowing -Wimplicit-fallthrough -Wthread-safety -Winconsistent-missing-override -Wundef -Wunused-lambda-capture -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/gtk-3.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/at-spi2-atk/2.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/at-spi-2.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/dbus-1.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/gtk-3.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/gio-unix-2.0/ -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/cairo -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/pango-1.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/harfbuzz -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/pango-1.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/fribidi -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/atk-1.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/cairo -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/pixman-1 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/uuid -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/freetype2 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/libpng16 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/gdk-pixbuf-2.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/libpng16 -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/glib-2.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/lib/arm-linux-gnueabihf/glib-2.0/include -DINSTALL_DIR=\"/opt/rws\" -Wall -fvisibility=hidden -pthread -Wno-undefined-bool-conversion -Wno-tautological-undefined-compare -std=c++11 -fno-exceptions -fno-rtti --sysroot=../../build/linux/debian_sid_arm-sysroot -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -D__RWS_VERSION__=\"v0.73-20-ga177b49d027fca4468be7f1190979502-dirty\" -D__WEBRTC_VERSION__="\"a687812c7087ade6b9c92f1fc4d68eaf,Cr-Commit-Position:refs/heads/master@{#24470}\"" -Wno-multichar -Wno-unused-but-set-variable -fPIC -O3 -DDEBUG -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__ -D__WEBRTC_DEFAULT__ -I/home/andre/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"338452-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -DCR_SYSROOT_HASH=b732cc2c870314130b767557b34764f8dd0a3df8 -D_DEBUG -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DWTF_USE_DYNAMIC_ANNOTATIONS=1 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DUSE_BUILTIN_SW_CODECS -DWEBRTC_ARCH_ARM -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -DHAVE_WEBRTC_VIDEO -DHAVE_WEBRTC_VOICE -I/home/andre/workspace/webrtc-checkout/src  -I../../../workspace/webrtc-checkout -I../../../workspace/webrtc-checkout/src/out/release/gen -I../../../workspace/webrtc-checkout/third_party/abseil-cpp -I../../../workspace/webrtc-checkout/third_party/libyuv/include -I../../../workspace/webrtc-checkout/third_party/jsoncpp/overrides/include -I../../../workspace/webrtc-checkout/third_party/jsoncpp/source/include  -I/home/andre/workspace/webrtc-checkout/rootfs/opt/vc/include/interface/vcos/pthreads -I/home/andre/workspace/webrtc-checkout/rootfs/opt/vc/include/interface/vmcs_host/linux -I/home/andre/workspace/webrtc-checkout/rootfs/opt/vc/include/interface/vmcs_host -I/home/andre/workspace/webrtc-checkout/rootfs/opt/vc/include/interface/vmcs_host/khronos -I/home/andre/workspace/webrtc-c [... cut ...]
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wc++11-narrowing’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wimplicit-fallthrough’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wthread-safety’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Winconsistent-missing-override’
arm-linux-gnueabihf-g++: error: unrecognized command line option ‘-Wunused-lambda-capture’
Makefile:83: recipe for target 'websocket_server.o' failed
make: *** [websocket_server.o] Error 1```

kclyu, 2018-09-16:
Coincidentally, I use the local development of the webrtc Cr-Commit-Position # 24470. First of all, # 24470 is no problem to build using the Custom Compile GCC I use.

As mentioned before, the above error message appears to be that the WebRTC native code library has been built using the clang compiler, not the compile option used by gcc.

First of all, please check [the section 4.2 of the README_build.md](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md#42-building-webrtc-native-code-library) file.

pbafff, 2018-09-17:
I finally got it! I ran `git log` again to just scrolled down until I saw the commit with `refs/heads/master@{#23500}`. Then `gclient sync -n -D -r 0c12d8d8c614bcb6fb44c4e08d5d0f7f127a664c` and checked out RWS from proxy-interface to master. 
Thanks for your help and the fast replies. By the way, what is the proxy-interface branch for?

kclyu, 2018-09-17:
It is a function to manage multiple rws through one proxy server. rws exchanges messages with one proxy server and tries to use streaming, motion noti, and cloud interworking functions with multiple rws through one proxy server.

In rws, an additional part for interworking with the proxy server is added to the proxy interface branch.

closed 2018-09-22 by kclyu