kclyu/rpi-webrtc-streamer #54: Streaming 720p not possible on Pi Zero W

otherippo, 2018-09-28:
I like RWC and it's working fine with 480p. But streaming 720p on a Pi Zero W is impossible due to 100% CPU usage. Why does it use that much compared to ~40% at 480p even using a text overlay? The video stream coming from the camera module is h264, so no conversion needed right? Are there some settings that I can improve?

kclyu, 2018-09-29:
As you said, rws does not use CPU resources for video encoding because it receives frames with h264 encoding.  I have not checked every CPU resource in detail, but at the moment I think you are using it as a limitation of the PI ZeorW CPU.

720p is about twice as big as 480p by resolution/bitrate used. However, WebRTC uses SRTP. As you know, 'S' means Secure, and as the size grows, the size that you have to encrypt becomes bigger.

Another factor is that if the size of the data to be transmitted by wireless increases, there will be a delay in wireless and I do not think that Pi ZeroW has good wireless performance. If there is a delay, it seems to be better to see the CPU usage increase.

If you really want to use 720p, you might be able to set the fps to a low level at a satisfactory screen quality level.

otherippo, 2018-09-29:
Thanks for your answer. I didn't think of the encryption part; I'll test different values for bitrate and fps till I get a good/reasonable result.

closed 2018-09-29 by otherippo