kclyu/rpi-webrtc-streamer #55: Compiling errors

tovil, 2018-09-30:
Hi,
I'm trying to compile the master branch of webrtc and get and error:
```
ninja: Entering directory `arm_build'
[350/2482] CXX obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o
FAILED: obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o
arm-linux-gnueabihf-g++ -MMD -MF obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o.d -DBWE_TEST_LOGGING_COMPILE_TIME_ENABLE=0 -DV8_DEPRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=b732cc2c870314130b767557b34764f8dd0a3df8 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DRTC_DISABLE_VP9 -DHAVE_SCTP -DUSE_BUILTIN_SW_CODECS -DWEBRTC_ARCH_ARM -DWEBRTC_ARCH_ARM_V7 -DWEBRTC_HAS_NEON -DWEBRTC_LIBRARY_IMPL -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I.. -Igen -I../third_party/abseil-cpp -I../third_party/libyuv/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv7-a -mfloat-abi=hard -mtune=cortex-a7 -mfpu=neon -mthumb -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -gdwarf-3 -g2 -fvisibility=hidden -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -std=gnu++11 -Wno-narrowing -fno-exceptions -fno-rtti --sysroot=../../../../../../opt/rpi_rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../modules/rtp_rtcp/source/rtp_receiver_video.cc -o obj/modules/rtp_rtcp/rtp_rtcp/rtp_receiver_video.o
In file included from ../modules/rtp_rtcp/include/rtp_rtcp_defines.h:18:0,
                 from ../modules/rtp_rtcp/source/rtp_receiver_video.h:14,
                 from ../modules/rtp_rtcp/source/rtp_receiver_video.cc:11:
../third_party/abseil-cpp/absl/types/variant.h: In substitution of גtemplate<unsigned int I, class T> using variant_alternative_t = typename absl::variant_alternative::type [with unsigned int I = I; T = absl::variant<absl::monostate, webrtc::RTPVideoHeaderVP8, webrtc::RTPVideoHeaderVP9, webrtc::RTPVideoHeaderH264>]ג:
../third_party/abseil-cpp/absl/types/variant.h:580:7:   required by substitution of גtemplate<class T, unsigned int I, class Tj, typename std::enable_if<(std::is_assignable<Tj&, T>::value && std::is_constructible<Tj, T>::value), void>::type* <anonymous> > absl::variant<T0, Ts ...>& absl::variant<T0, Ts ...>::operator=(T&&) [with T = const absl::variant<absl::monostate, webrtc::RTPVideoHeaderVP8, webrtc::RTPVideoHeaderVP9, webrtc::RTPVideoHeaderH264>&; unsigned int I = <missing>; Tj = <missing>; typename std::enable_if<(std::is_assignable<Tj&, T>::value && std::is_constructible<Tj, T>::value), void>::type* <anonymous> = <missing>]ג
../modules/rtp_rtcp/source/rtp_receiver_video.cc:73:60:   required from here
../third_party/abseil-cpp/absl/types/variant.h:580:7: internal compiler error: unexpected expression גIג of kind template_parm_index
       class Tj = absl::variant_alternative_t<I, variant>,
       ^~~~~
```
What webrtc branch are you using?
I've tried branch-heads/69 (without compilation errors) but is not compatible with the current RWS code without changing it.

kclyu, 2018-10-01:
The latest webrtc branch is being used by the proxy_interface branch (WebRTC Cr-Commit Position 24781). The master branch will be updated with a merge of proxy_interface bracch.

The commit log for RWS specifies the WebRTC Cr-Commit-Position you are using on a branch-by-branch basis.

Sepsu, 2018-10-19:
@tovil I had the same issue when compiling WebRTC Cr-Commit-Position 25138 which is the latest matching position for the proxy_interface branch. This is a compiler bug that has been fixed in newer gcc compilers. You will have to update your cross-compiler to new version, for me gcc-linaro-7.3.1-2018.05-x86_64_arm-linux-gnueabihf worked. You can download it from: https://releases.linaro.org/components/toolchain/binaries/latest-7/arm-linux-gnueabihf/ 

After this you will have to rerun some steps in the rws build documentation. Basically remove the old arm-linux-gnueabihf linked folder in rpi_rootfs/tools and then relink according 1.3 step with ln -sf to point to the new compiler. Also make sure that you build webrtc and rws from clean that all objects are build using the same same compiler, otherwise you will get linker errors.

@kclyu I have seen few people struggling with this same compiler error, maybe it's best to provide link to the new linaro toolchain in the docs as this is a compiler bug that is visible when building WebRTC (not related to the used RWS version)

kclyu, 2018-10-19:
I think there is another cause for something different  depending on the environment you use.
Currently I do not have any compiling problem at all with Cr-Commit-Position 25138 and below.

Of course, using the latest version of linaro 7.3.1 will not be a problem, but linaro version does not use hardfp and  has problems with armv6 compilation because it used armv7 and above only. That's why Custom Build GCC download link.

If you are having trouble using Custom Build GCC at this time, you can try it out as mentioned above. Instead, the armv6 target is expected to suffer other compilation problems.

@Sepsu  Anyway, thanks for letting us know about other solutions to the problem. By the way, this problem seems not to be easily solved. Currently I do not see it as a compiler bug because I use it without any problems.

kclyu, 2018-10-21:
First of all, there seems to be a problem with the Gcc binary in the download link. So far I have thought that the difference between the GCC binaries I use is the same because the building time difference is one to two weeks apart.

@Sepsu Thank you for letting  me know that there is a problem with GCC.

Download the new GCC compiled GCC below and try it out.

https://drive.google.com/file/d/1vBFMTHGdATggRWAK0CFYQXNbchmjGQCj/view?usp=sharing

If there is no problem, I will fix Download link.

closed 2018-11-07 by kclyu