kclyu/rpi-webrtc-streamer #56: Issue

Darkcast, 2018-10-14:
After I install it via the Deb file I open chromium and I go to the url page and I get the player but when I hit connect nothing happens. Another thing is that sometime it will work on my phone but I have to have the rpi cam web interface install but stopped

Are there any requirements for this? 

I am running the latest version of raspian stretch fully updated

kclyu, 2018-10-14:
In the /opt/rws/log directory, find the error message and attach it or find the error and let me know. It is difficult to understand what you are talking about right now.

My guess is that rws failed to connect camera because camera is not available.

Darkcast, 2018-10-15:
Will do.
BTW I want to see the video locally on the pi.

Darkcast, 2018-10-16:
Let me explain a little bit better,
I am running a rasp 3 with stretch 
I have chromium and Vivaldi installed 
what I am trying to view the video of the camera directly from the raspberry browser
after I install it I can see the video using my phone but no directly from the raspberry pi and I was wonder if its a limitation or do I need to modify something or install.

Thanks for your help 

[log.tar.gz](https://github.com/kclyu/rpi-webrtc-streamer/files/2481126/log.tar.gz)

kclyu, 2018-10-18:
I have not confirmed that it works properly with raspberry pi chromium.

In the past, another person tried to connect with rws installed in another raspberry pi in raspberry pi chromium, but the frame rate was too slow and I concluded that there would be a performance problem.

First of all, it is possible to use WebRTC as chromium in raspberry pi basically, but since chromium does not yet support raspberry pi hardware decoding, it seems that there will be a lot of performance issues even if it works.

In addition, raspberry pi on X desktop and rws on chromium together will surely cause performance problems.

If you want to use raspberry pi's chromium, you might want to try another raspberry pi hardware if possible. If you are successful in other raspberry pi, please let me know how much video frame rate you have.

Sorry for the delay. I checked it too late.

Darkcast, 2018-10-18:
what I am trying to do is to watch it the raspberry pi camera from the raspberry pi itself I have used tried it chromium, vivaldi and firefox and it will never display the video. interestingly enought trying to get it to work I installed RPI web cam gui and if I stopped the it the webrtc streamer would work but I had to flash my sdcard and I haven't been able to get it to work again. :(

kclyu, 2018-10-19:
To view a video from a single raspberry pi in chromium, you can simply use the WebRTC script.

https://webrtc.github.io/samples/src/content/getusermedia/resolution/

Darkcast, 2018-10-20:
Will check it out thanks a lot

Darkcast, 2018-10-23:
This is exactly what I was looking for my project thanks. 
please close the ticket.

closed 2018-10-23 by Darkcast