kclyu/rpi-webrtc-streamer #58: Stream shows only in the same WiFi

kentloog, 2018-11-24:
Hello Lyu,

I have successfully installed the rpi-webrtc streamer on my Raspberry Pi 3 b+ and i have also implemented your frontend application running on firebase. Problem is that i can only see the video stream from rpi when my laptop is connected to the same WiFi as my rpi is connected to. Is it supposed to only work in the same network or i have done something wrong? How can i get it to work over the internet without having to be in the same WiFI?

Really appreciate your work and your help.
Kent

kclyu, 2018-11-25:
Of course, it is not intended to work on the same network. The reason for using firebase is that it works on networks that use https and connect to firebaseapp. Test was also works on my smart phone with  Wifi disabled.
For most networks, WebRTC can connect using ICE. However, there are some cases where the router or network environment you are using does not work with ICE only.

There seems to be an ICE connection error in the browser java console or RWS log file.

kclyu, 2018-11-27:
The absence of the 'ICE connection state changed to: connected' message means that the ICE connection failed. In case of ICE connection failure, I will try to find  an additional method to add ice failure log message.

I have the same results in the tests I did, but I would like to ask you to test the attached recently built [webrtc-streamer.gz binary](https://github.com/kclyu/rpi-webrtc-streamer/files/2619007/webrtc-streamer.gz) rws binary. In the meantime, many of the fixes in the WebRTC native code package have been updated.

kentloog, 2018-12-02:
I unzipped webrtc-streamer file and copied it to rpi-webrtc-streamer folder. Still not working outside wifi. I also pulled the latest commits from github.

kclyu, 2018-12-03:
In the WebRTC native code package, a tool called stun_prober has been added before, and the arm build has a problem in outputting the log, so I attach [stun_prober.gz](https://github.com/kclyu/rpi-webrtc-streamer/files/2640148/stun_prober.gz) as  ubuntu x64 build. I would appreciate your feedback.

Below is the result of running on my ubuntu, and my public ip has been changed to xxx. If you look at the results below, 'Success Percent' is 100%.
```
$ ./stun_prober 
(main.cc:95): Result: 0
(main.cc:74): Shared Socket Mode: 0
(main.cc:75): Requests sent: 30
(main.cc:76): Responses received: 30
(main.cc:77): Target interval (ns): 10000
(main.cc:79): Actual interval (ns): 10448
(main.cc:81): NAT Type: Unknown NAT type
(main.cc:82): Host IP: 
(main.cc:83): Server-reflexive ips: 
(main.cc:85): 	xxx.xxx.xxx.25:33306
(main.cc:85): 	xxx.xxx.xxx.25:33814
(main.cc:85): 	xxx.xxx.xxx.25:33836
(main.cc:85): 	xxx.xxx.xxx.25:34673
(main.cc:85): 	xxx.xxx.xxx.25:35693
(main.cc:85): 	xxx.xxx.xxx.25:38107
(main.cc:85): 	xxx.xxx.xxx.25:38397
(main.cc:85): 	xxx.xxx.xxx.25:39792
(main.cc:85): 	xxx.xxx.xxx.25:40614
(main.cc:85): 	xxx.xxx.xxx.25:41049
(main.cc:85): 	xxx.xxx.xxx.25:41227
(main.cc:85): 	xxx.xxx.xxx.25:42712
(main.cc:85): 	xxx.xxx.xxx.25:43654
(main.cc:85): 	xxx.xxx.xxx.25:43870
(main.cc:85): 	xxx.xxx.xxx.25:44460
(main.cc:85): 	xxx.xxx.xxx.25:44482
(main.cc:85): 	xxx.xxx.xxx.25:44937
(main.cc:85): 	xxx.xxx.xxx.25:45518
(main.cc:85): 	xxx.xxx.xxx.25:45848
(main.cc:85): 	xxx.xxx.xxx.25:46232
(main.cc:85): 	xxx.xxx.xxx.25:47799
(main.cc:85): 	xxx.xxx.xxx.25:50425
(main.cc:85): 	xxx.xxx.xxx.25:52893
(main.cc:85): 	xxx.xxx.xxx.25:53330
(main.cc:85): 	xxx.xxx.xxx.25:54513
(main.cc:85): 	xxx.xxx.xxx.25:55556
(main.cc:85): 	xxx.xxx.xxx.25:57063
(main.cc:85): 	xxx.xxx.xxx.25:58185
(main.cc:85): 	xxx.xxx.xxx.25:58596
(main.cc:85): 	xxx.xxx.xxx.25:60063
(main.cc:88): Success Precent: 100
(main.cc:89): Response Latency
```