kclyu/rpi-webrtc-streamer #59: Unable to stream video on Pi Zero W with attached Speaker/Mic

drewcovi, 2018-12-03:
Love the promise of this package. Installed Pi Zero W using `gdebi` and have the [RaspiAudio](https://www.raspiaudio.com/) speaker/mic combo attached.

After connecting, CPU soars, and hangs.

Any way to fix audio input/output or disable it for the time being (short of removing the hat)?

aplay -l
```
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

cat /proc/asound/modules 
```
 0 snd_soc_googlevoicehat_soundcard
```

```
[1627:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1628:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1629:008] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1630:013] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1631:013] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1632:006] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1633:007] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1634:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1635:007] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1636:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1637:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1638:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1639:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1640:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1641:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1642:008] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1643:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1644:007] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
[1645:004] [951] (audio_device_alsa_linux.cc:1561): capture snd_pcm_readi error: Broken pipe
```

kclyu, 2018-12-03:
Since Raspberry PI ZeroW does not use mostly audio, But depending on the user setting, it may be correct to add the configuration flag to disable audio. i will add it at the next patch.
Because audio encoding is CPU intensive, it is not suitable for covering with Zero W CPU performance.

drewcovi, 2018-12-03:
@kclyu definitely true under most contexts. The hat in the link above is pretty incredible, however and does both recording and playback with amplifier. If there's an option to pass audio through, ideally there's a less CPU intensive process.

Thanks in advance for the patch!

drewcovi, 2018-12-03:
@kclyu I've attached my latest log to help sort out any issues. 

If there's even a chance we can capture mic input that would be *huge*

If I can pause the WebRTC video stream to support audio out through a separate channel, that's even better (perhaps audio only).

I believe I've heard this is possible in another logged issue, where two-way audio was sorted out.

[rws_log_0.log](https://github.com/kclyu/rpi-webrtc-streamer/files/2640115/rws_log_0.log)

drewcovi, 2019-01-03:
I see we have a patch in place! I'll have to wait for the debian package or find some extra time to get this running from the source. but thanks a million!!

closed 2019-01-03 by drewcovi