kclyu/rpi-webrtc-streamer #64: cross compiling fail

leofds, 2019-02-19:
Hello
I followed the procedure described in "Moving to specific commit position of WebRTC native code package".

    Cr-Commit-Position: refs/heads/master@{#26361}
    WebRTC commit position: 41dd0bc4e5af6c25a222f9e47d61eac3b825edbe

But when I run the make command in "rpi-webrtc-streamer/src" I get the following error:

~/Workspace/rpi-webrtc-streamer/src$ make
/bin/sh: 1: gawk: not found
arm-linux-gnueabihf-g++ --sysroot=/opt/rpi_rootfs    -I. -DINSTALL_DIR=\"/opt/rws\" -Wno-deprecated-declarations -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pipe -pthread -march=armv7-a -mfloat-abi=hard -mtune=cortex-a7 -mfpu=neon -mthumb -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-comments -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -gdwarf-3 -g2 -fvisibility=hidden -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -isystem../../../../../../opt/rpi_rootfs/usr/include/gtk-3.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/at-spi2-atk/2.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/at-spi-2.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/dbus-1.0 -isystem../../../../../../opt/rpi_rootfs/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -isystem../../../../../../opt/rpi_rootfs/usr/include/gtk-3.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/gio-unix-2.0/ -isystem../../../../../../opt/rpi_rootfs/usr/include/cairo -isystem../../../../../../opt/rpi_rootfs/usr/include/pango-1.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/harfbuzz -isystem../../../../../../opt/rpi_rootfs/usr/include/pango-1.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/atk-1.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/cairo -isystem../../../../../../opt/rpi_rootfs/usr/include/pixman-1 -isystem../../../../../../opt/rpi_rootfs/usr/include/freetype2 -isystem../../../../../../opt/rpi_rootfs/usr/include/libpng16 -isystem../../../../../../opt/rpi_rootfs/usr/include/gdk-pixbuf-2.0 -isystem../../../../../../opt/rpi_rootfs/usr/include/libpng16 -isystem../../../../../../opt/rpi_rootfs/usr/include/glib-2.0 -isystem../../../../../../opt/rpi_rootfs/usr/lib/arm-linux-gnueabihf/glib-2.0/include -DINSTALL_DIR=\"/opt/rws\" -Wall -fvisibility=hidden -pthread -std=gnu++11 -Wno-narrowing -fno-exceptions -fno-rtti --sysroot=../../../../../../opt/rpi_rootfs -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -D__RWS_VERSION__=\"v0.73-53-g6588491778e9af6137742480e1b2e851\" -D__WEBRTC_VERSION__="\"41dd0bc4e5af6c25a222f9e47d61eac3,\"" -Wno-multichar -Wno-unused-but-set-variable -fPIC -O3 -DDEBUG -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__ -D__WEBRTC_DEFAULT__ -I/home/boss/Workspace/rpi-webrtc-streamer/lib/libwebsockets/arm_build -I/home/boss/Workspace/rpi-webrtc-streamer/lib/libwebsockets -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=ef5c4f84bcafb7a3796d36bb1db7826317dde51c -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DHAVE_SCTP -DWEBRTC_ARCH_ARM -DWEBRTC_ARCH_ARM_V7 -DWEBRTC_HAS_NEON -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -DHAVE_WEBRTC_VIDEO -DHAVE_WEBRTC_VOICE -I/home/boss/Workspace/webrtc/src  -I../../webrtc/src -I../../webrtc/src/arm_build/gen -I../../webrtc/src/third_party/abseil-cpp -I../../webrtc/src/third_party/libyuv/include -I../../webrtc/src/third_party/jsoncpp/overrides/include -I../../webrtc/src/third_party/jsoncpp/source/include  -I/opt/rpi_rootfs/opt/vc/include/interface/vcos/pthreads -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/linux -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host -I/opt/rpi_rootfs/opt/vc/include/interface/vmcs_host/khronos -I/opt/rpi_rootfs/opt/vc/include/interface/khronos/include -I/opt/rpi_rootfs/opt/vc/include/interface/vchiq_arm -I/opt/rpi_rootfs/opt/vc/include -I/opt/rpi_rootfs/opt/vc/include/interface/mmal  -I/home/boss/Workspace/webrtc/src/third_party/abseil-cpp/  -c websocket_server.cc -o websocket_server.o
In file included from websocket_server.h:41:0,
                 from websocket_server.cc:36:
websocket_server_internal.h:37:27: erro fatal: libwebsockets.h: Arquivo ou diretório inexistente
 #include "libwebsockets.h"
                           ^
compilação terminada.
Makefile:90: recipe for target 'websocket_server.o' failed
make: *** [websocket_server.o] Error 1

kclyu, 2019-02-20:
Added include directory to mk/libwebsocket_arm_build.mk. It should have been added at libwebsockets upgrade, but I did commit this patch at first because of other modifications.

Please, Pull the rws repository and try to recompile.

leofds, 2019-02-20:
Went ahead but stopped again:

/home/boss/Workspace/rpi_rootfs/tools/gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld.gold: erro: não pôde encontrar -lvcfiled_check
collect2: erro: ld returned 1 exit status
Makefile:94: recipe for target '../webrtc-streamer' failed
make: *** [../webrtc-streamer] Error 1

kclyu, 2019-02-20:
I think you did not create rootfs properly with rpi_rootfs.

Please refer to [Making Raspberry PI Root FS](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md#41-making-raspberry-pi-root-fs)  to create rpi_rootfs.

leofds, 2019-02-20:
I still have the output with error:

```
/home/boss/Workspace/rpi_rootfs/tools/gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld.gold: file_posix.o: previous definition here
collect2: error: ld returned 1 exit status
Makefile:92: recipe for target '../webrtc-streamer' failed
make: *** [../webrtc-streamer] Error 1
```
What am I doing wrong ?

My steps:

I recorded a new MicroSD card with 2018-11-13-raspbian-stretch.img, enabled SSH, set up Audio and Video (V4L2).
I deleted ~/Workspace/rpi_rootfs
I deleted ~/Workspace/rpi-webrtc-streamer
I followed according to items:

**3.1. Installing Raspberry PI Prerequisite**

```
ssh pi@your-raspberry-pi-ipaddress

sudo apt-get update && sudo apt-get upgrade
sudo apt-get install rsync
sudo apt-get install libasound2-dev libcairo2-dev  libffi-dev libglib2.0-dev  \
libgtk2.0-dev libpam0g-dev libpulse-dev  libudev-dev  libxtst-dev  \
ttf-dejavu-core libatk1.0-0 libc6 libasound2  libcairo2 libcap2 libcups2  \
libexpat1 libffi6 libfontconfig1 libfreetype6  libglib2.0-0 libgnome-keyring0  \
libgtk2.0-0 libpam0g libpango1.0-0  libpcre3 libpixman-1-0 libpng12-0 libstdc++6  \
libx11-6 libx11-xcb1 libxau6 libxcb1 libxcomposite1 libxcursor1 libxdamage1   \
libxdmcp6 libxext6 libxfixes3 libxi6 libxinerama1 libxrandr2 libxrender1  \
libxtst6 zlib1g gtk+-3.0
```

**1.3. Custom Compiled GCC for Raspberry PI**

```
git clone https://github.com/kclyu/rpi_rootfs
cd rpi_rootfs
mkdir tools
cd tools
# (Download Custom Compiled GCC) # note1
xz -dc ~/Downloads/gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf.tar.xz  | tar xvf -
ln -sf gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf  arm-linux-gnueabihf
cd /opt
sudo ln -sf ~/Workspace/rpi_rootfs
export PATH=/opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin:$PATH
```
[terminal output](https://drive.google.com/file/d/1WjdTWUgXBEy9btTr8w6cuYhmxHs5lTR9/view?usp=sharing)

**4.1. Making Raspberry PI Root FS**

```
cd ~/Workspace/rpi_rootfs
./rpi_rootfs.py pi@your-raspberry-pi-ipaddress ./
```
[rpi_rootfs.py terminal output](https://drive.google.com/file/d/1KgXsuaHUrEtnPqdEd46JTeBKqiM2ipw9/view?usp=sharing)

**2.3. Rpi-WebRTC-Streamer**

```
cd ~/Workspace
git clone https://github.com/kclyu/rpi-webrtc-streamer
```

**4.3. Building Rpi-WebRTC-Streamer**

```
cd ~/Workspace/rpi-webrtc-streamer/src
sh ../mk/config_libwebsockets.sh
make # note 4
```
[config_libwebsockets.sh terminal output](https://drive.google.com/file/d/1NigozXCU8tw613IbGWH7sVSi1_UoKBRa/view?usp=sharing)
[make terminal output](https://drive.google.com/file/d/1h9tVl4nl9n1UAWLicETP7TilMsj2A-UV/view?usp=sharing)

kclyu, 2019-02-21:
It looks like you have not seen the latest RWS commit log.
The latest RWS commit is compiled using the latest WebRTC native code. commit  description also mentioned WebRTC commit position.

Please update and rebuild WebRTC and RWS  in the following way.
```
cd ~/Workspace/webrtc/src
git pull
gclient sync 
gclient sync # double checking
gn clearn arm_build
gn gen arm_build
ninja -C arm_build
cd ~/Workspace/rpi-webrtc-streamer
make clean
make
```

But what is the 'gawk not found' message? gawk does not seem to be installed on the system, but I think you should check this a bit. 
I do not know if you did [3.2. Installing prerequisite s/w package in Ubuntu Linux](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md#32-installing-prerequisite-sw-package-in-ubuntu-linux) procedure. Please check.

leofds, 2019-02-22:
I already deleted everything and followed the compilation tutorial several times.
I'm sure I followed the tutorial to compile correctly but I still get error:

```
/home/boss/Workspace/rpi_rootfs/tools/gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld.gold: error: cannot find -lvcfiled_check
collect2: error: ld returned 1 exit status
Makefile:92: recipe for target '../webrtc-streamer' failed
make: *** [../webrtc-streamer] Error 1
```

I moved the commit position to;

```
Cr-Commit-Position: refs/heads/master@{#26750}
WebRTC commit position: d8d3248d95437863f3448affad7a1bc2238a1959
```
I get warning on:
```
boss@ubuntu:~/Workspace/webrtc/src$ gn gen arm_build
WARNING at build arg file (use "gn args <out_dir>" to edit):44:23: Build argument has no effect.
rtc_initialize_ffmpeg=false
                      ^----
The variable "rtc_initialize_ffmpeg" was set as a build argument
but never appeared in a declare_args() block in any buildfile.

To view all possible args, run "gn args --list <out_dir>"

The build continued as if that argument was unspecified.

Done. Made 623 targets from 181 files in 552ms
```
I'm using Ubuntu 18.04 64 bit, be it the problem, or the gcc version ?

[rpi_rootfs.py terminal output](https://drive.google.com/file/d/1gUHi-NRChpIAsmI-VolVc7k-yyAix7LF/view?usp=sharing)
[gn gen arm_build terminal output](https://drive.google.com/file/d/1ZeHwprEJRHwvUiCCLgu-Tt_9TJNTzBRX/view?usp=sharing)
[config_libwebsockets.sh terminal output](https://drive.google.com/file/d/1nYBF0RYEWPlQcc7YDGypc-fJ6h-jwZsf/view?usp=sharing)
[make terminal output](https://drive.google.com/file/d/1DGcjtJbKjA02QRfIR8aeD1_8DNhHnOxB/view?usp=sharing)

kclyu, 2019-02-22:
What kind of Raspbian OS do you use? If you use 'Stretch Lite', please change it to 'Raspbian Stretch with desktop' or 'Raspbian Stretch with desktop and recommended software'.

Below is the contents of opt/vc/lib.
```
kclyu@keunchang2:~/Workspace/rpi_rootfs/opt/vc$ find . -name \*vcfiled_check\*
./include/interface/vmcs_host/linux/vcfiled/vcfiled_check.h
./lib/libvcfiled_check.a
kclyu@keunchang2:~/Workspace/rpi_rootfs/opt/vc$ ls -lR | wc -l
659
```
As you can see from the above, 'opt/vc/lib/libvcfiled_check.a' exists in my rpi_rootfs. It is also the cause of the building error.

leofds, 2019-02-22:
I use the last image of 'Raspbian Stretch with desktop' (2018-11-13)

Below is the contents of opt/vc/lib.

boss@ubuntu:~/Workspace/rpi_rootfs/opt/vc$ find . -name \*vcfiled_check\*
./include/interface/vmcs_host/linux/vcfiled/vcfiled_check.h

boss@ubuntu:~/Workspace/rpi_rootfs/opt/vc$ ls -lR | wc -l
620

kclyu, 2019-02-22:
Can you run the following command from raspberry pi and tell me the results of the find and grep commands in /opt/vc of raspberry pi?
```
sudo apt update
sudo apt full-upgrade
```

leofds, 2019-02-22:
After the update and upgrade...

```
pi@raspberrypi:/opt/vc $ find . -name \*vcfiled_check\*
./include/interface/vmcs_host/linux/vcfiled/vcfiled_check.h

pi@raspberrypi:/opt/vc $ grep -r -i -name \*vcfiled_check\*
grep: invalid max count

```

kclyu, 2019-02-22:
The following command is a procedure for directly viewing the Raspberry PI OS img file. Please run the following commands in the folder where 2018-11-13-raspbian-stretch.zip is located and Please, let me know the result of the last `ls -ㅣ` command.
```
sudo apt install p7zip-full
unzip 2018-11-13-raspbian-stretch.zip
7z e 2018-11-13-raspbian-stretch.img -o.
sudo mount -t ext4 1.img /mnt
ls -l /mnt/opt/vc/lib
```

leofds, 2019-02-22:
```
boss@ubuntu:~/Downloads$ sudo mount -t ext4 1.img /mnt
boss@ubuntu:~/Downloads$ ls -l /mnt/opt/vc/lib
total 2908
-rw-r--r-- 1 root root  99516 Nov 12 09:25 libbcm_host.so
-rw-r--r-- 1 root root 202072 Nov 12 09:25 libbrcmEGL.so
-rw-r--r-- 1 root root 105768 Nov 12 09:25 libbrcmGLESv2.so
-rw-r--r-- 1 root root  99208 Feb  9  2018 libbrcmOpenVG.so
-rw-r--r-- 1 root root  78552 Nov 12 09:25 libbrcmWFC.so
-rw-r--r-- 1 root root 364540 Aug 17  2018 libcontainers.so
-rw-r--r-- 1 root root  15932 Aug 17  2018 libdebug_sym.so
-rw-r--r-- 1 root root  13254 Nov 12 09:26 libdebug_sym_static.a
-rw-r--r-- 1 root root  53088 Sep 11 02:35 libdtovl.so
-rw-r--r-- 1 root root 245356 Nov 12 09:26 libEGL_static.a
-rw-r--r-- 1 root root 344732 Feb  9  2018 libelftoolchain.so
-rw-r--r-- 1 root root 100076 Nov 12 09:26 libGLESv2_static.a
-rw-r--r-- 1 root root  13548 Nov 12 09:26 libkhrn_client.a
-rw-r--r-- 1 root root 450544 Nov 12 09:26 libkhrn_static.a
-rw-r--r-- 1 root root  68884 Nov 12 09:25 libmmal_components.so
-rw-r--r-- 1 root root  82168 Nov 12 09:25 libmmal_core.so
-rw-r--r-- 1 root root  15368 Nov 12 09:25 libmmal.so
-rw-r--r-- 1 root root  91024 Nov 12 09:25 libmmal_util.so
-rw-r--r-- 1 root root  44424 Nov 12 09:25 libmmal_vc_client.so
-rw-r--r-- 1 root root  33208 Aug 17  2018 libopenmaxil.so
-rw-r--r-- 1 root root   3264 Nov 12 09:26 libvcfiled_check.a
-rw-r--r-- 1 root root  30288 Aug 17  2018 libvchiq_arm.so
-rw-r--r-- 1 root root 140520 Nov 12 09:26 libvchostif.a
-rw-r--r-- 1 root root  28462 Nov 12 09:26 libvcilcs.a
-rw-r--r-- 1 root root 173248 Aug 17  2018 libvcos.so
-rw-r--r-- 1 root root  23136 Aug 17  2018 libvcsm.so
drwxr-xr-x 2 root root   4096 Nov 13 05:14 pkgconfig
drwxr-xr-x 3 root root   4096 Nov 13 05:09 plugins
boss@ubuntu:~/Downloads$ 
```

kclyu, 2019-02-22:
It seems that there is a problem when making OS image to SD card. I do not know what procedure you are using, but please use SD card burning program [Etcher](https://www.balena.io/etcher/) to make SD again.

leofds, 2019-02-22:
I did it with Etcher. This is doing again and I will see the file libvcfiled_check.a is present.

kclyu, 2019-02-22:
After burning the OS to SD, use the following command to check whether the same result is displayed. If it is different, it is better to try SD card replacement.
```
pi@raspberrypi:/opt/vc $ du -sk
42776	.
pi@raspberrypi:/opt/vc $ ls -lR | wc -l
659
```

leofds, 2019-02-22:
I recorded the image and executed the command:
```
pi@raspberrypi:/opt/vc $ find . -name \*vcfiled_check\*
./include/interface/vmcs_host/linux/vcfiled/vcfiled_check.h
./lib/libvcfiled_check.a
```
But after the apt update and full-upgrade commands the file libvcfiled_check.a no longer exists.

```
pi@raspberrypi:/opt/vc $ find . -name \*vcfiled_check\*
./include/interface/vmcs_host/linux/vcfiled/vcfiled_check.h
```

leofds, 2019-02-22:
I recorded a new SD with a new download of the current image:

```
pi@raspberrypi:/opt/vc $ du -sk
40968	.
pi@raspberrypi:/opt/vc $ ls -lR | wc -l
626
```

kclyu, 2019-02-22:
I also did apt full-upgrade so the libvcfiled_check.a file disappeared. I actually did a full-upgrade within two weeks ago. Raspberry PI's /opt/vc has not changed in a several years,  but  file deletion seems to be a little less than a week.

I removed libvcfiled_check.a from mk/cross_mmal.mk, and there is no problem with build and running.  I will commit the change after checking the other one. First, modify the cross_mmal.mk file and build it yourself.

kclyu, 2019-02-22:
I forgot to reference this issue in commit. commit posted.

leofds, 2019-02-22:
Make is successful now \o/

But I get the following error when run the program:

```
pi@raspberrypi:~/Workspace/rws $ ./webrtc-streamer --verbose
config file :etc/webrtc_streamer.conf
Using config file base path:CWD
Failed to find the log directory /opt/rws/log
Failed to get log directory : log
```
The scp command copies the log folder, but it does not exist

```
boss@ubuntu:~/Workspace/rpi-webrtc-streamer$ ls -l
total 149784
drwxr-xr-x 3 boss boss      4096 Feb 22 11:05 etc
drwxr-xr-x 3 boss boss      4096 Feb 22 12:01 lib
drwxr-xr-x 2 boss boss      4096 Feb 22 11:05 LICENSE
-rw-r--r-- 1 boss boss       251 Feb 22 11:05 LICENSE.md
drwxr-xr-x 2 boss boss      4096 Feb 22 11:05 misc
drwxr-xr-x 2 boss boss      4096 Feb 22 11:05 mk
-rw-r--r-- 1 boss boss      2507 Feb 22 11:05 README_audio.md
-rw-r--r-- 1 boss boss      7387 Feb 22 11:05 README_building.md
-rw-r--r-- 1 boss boss      3923 Feb 22 11:05 README.md
-rw-r--r-- 1 boss boss      5799 Feb 22 11:05 README_motion.md
-rw-r--r-- 1 boss boss     15682 Feb 22 11:05 README_rws_setup.md
-rw-r--r-- 1 boss boss     10013 Feb 22 11:05 README_TelegramBot.md
drwxr-xr-x 2 boss boss      4096 Feb 22 11:05 scripts
drwxr-xr-x 4 boss boss      4096 Feb 22 12:03 src
drwxr-xr-x 2 boss boss      4096 Feb 22 11:05 tools
drwxr-xr-x 5 boss boss      4096 Feb 22 11:05 web-root
-rwxr-xr-x 1 boss boss 153283694 Feb 22 12:03 webrtc-streamer
```

leofds, 2019-02-22:
I created the /opt/rws/log directory and it worked.

In the test I did with the .deb file it looks like the equo cancellation worked very well, now it is not working.

closed 2019-02-25 by leofds