kclyu/rpi-webrtc-streamer #65: Android App No Video on Different Network

mintu19, 2019-02-26:
I am using the RWS deb file from the Readme. I am also using rws frontent which i have modified to work with firestore ([Link](https://github.com/mintu19/rpi-webrtc-streamer-frontend)). I have hosted the same on firebase and that is working perfectly.

Then i have created android app connected via firestore. When device and mobile are on same network, everything works but when they are on different networks, messages are transferred, media is added but no video on android app. Is it related to turn server(rws have turn support?)? 

Detailed question: [on stackoverflow](https://stackoverflow.com/questions/54893790/android-webrtc-doesnt-work-on-different-network-no-video)

kclyu, 2019-02-27:
>  I am under the impression that if web client is able to display video on different network (mobile hotspot), android client on same internet used by web client should be able to display video as well. Is it wrong? Why won't android display video (onAddStream is called)

As mentioned above, if you see a video in a web client and you can not see a video in android, I do not think  it as a turn related issue.
RWS uses H.264 video. Android App has an option to force H.264 codec to be used, so if you can not do it, please check if Video Negotiation SDP is set to H.264.

mintu19, 2019-02-27:
When same network 

Remote SDP
```
D/remoteSDP: v=0
    o=- 1355588823157143804 2 IN IP4 127.0.0.1
    s=-
    t=0 0
    a=group:BUNDLE audio video
    a=msid-semantic: WMS stream_id
    m=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 102 0 8 105 13 110 113 126
    c=IN IP4 0.0.0.0
    a=rtcp:9 IN IP4 0.0.0.0
    a=ice-ufrag:pDIh
    a=ice-pwd:c7rZvSUNyDgIZ3brGh4iIat/
    a=ice-options:trickle
    a=fingerprint:sha-256 5D:46:62:7B:DE:30:DE:50:80:09:FD:46:4A:39:88:55:CE:1F:AE:4D:F7:1C:78:68:CF:BD:33:D6:F4:AC:6D:44
    a=setup:actpass
    a=mid:audio
    a=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level
    a=sendrecv
    a=rtcp-mux
    a=rtpmap:111 opus/48000/2
    a=rtcp-fb:111 transport-cc
    a=fmtp:111 minptime=10;useinbandfec=1
    a=rtpmap:103 ISAC/16000
    a=rtpmap:9 G722/8000
    a=rtpmap:102 ILBC/8000
    a=rtpmap:0 PCMU/8000
    a=rtpmap:8 PCMA/8000
    a=rtpmap:105 CN/16000
    a=rtpmap:13 CN/8000
    a=rtpmap:110 telephone-event/48000
    a=rtpmap:113 telephone-event/16000
    a=rtpmap:126 telephone-event/8000
    a=ssrc:1739619451 cname:J/XcygPRhIPmQdxG
    a=ssrc:1739619451 msid:stream_id audio_label
    a=ssrc:1739619451 mslabel:stream_id
    a=ssrc:1739619451 label:audio_label
    m=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 127
    c=IN IP4 0.0.0.0
    a=rtcp:9 IN IP4 0.0.0.0
    a=ice-ufrag:pDIh
    a=ice-pwd:c7rZvSUNyDgIZ3brGh4iIat/
    a=ice-options:trickle
    a=fingerprint:sha-256 5D:46:62:7B:DE:30:DE:50:80:09:FD:46:4A:39:88:55:CE:1F:AE:4D:F7:1C:78:68:CF:BD:33:D6:F4:AC:6D:44
    a=setup:actpass
    a=mid:video
    a=extmap:2 urn:ietf:params:rtp-hdrext:toffset
    a=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
    a=extmap:4 urn:3gpp:video-orientation
    a=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
    a=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay
    a=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type
    a=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/video-timing
    a=sendonly
    a=rtcp-mux
    a=rtcp-rsize
    a=rtpmap:96 H264/90000
    a=rtcp-fb:96 goog-remb
    a=rtcp-fb:96 transport-cc
    a=rtcp-fb:96 ccm fir
    a=rtcp-fb:96 nack
    a=rtcp-fb:96 nack pli
    a=fmtp:96 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f
    a=rtpmap:97 rtx/90000
    a=fmtp:97 apt=96
    a=rtpmap:98 H264/90000
    a=rtcp-fb:98 goog-remb
    a=rtcp-fb:98 transport-cc
[... 232 lines cut ...]
    a=extmap:2 urn:ietf:params:rtp-hdrext:toffset
    a=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
    a=extmap:4 urn:3gpp:video-orientation
    a=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
    a=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay
    a=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type
    a=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/video-timing
    a=recvonly
    a=rtcp-mux
    a=rtcp-rsize
    a=rtpmap:98 H264/90000
    a=rtcp-fb:98 goog-remb
    a=rtcp-fb:98 transport-cc
    a=rtcp-fb:98 ccm fir
    a=rtcp-fb:98 nack
    a=rtcp-fb:98 nack pli
    a=fmtp:98 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f
    a=rtpmap:99 rtx/90000
    a=fmtp:99 apt=98
    a=rtpmap:100 red/90000
    a=rtpmap:101 rtx/90000
    a=fmtp:101 apt=100
    a=rtpmap:127 ulpfec/90000

```

mintu19, 2019-02-27:
So i have discovered that After videotrack.addsink(view) is called,

In same network i get log of surface connected and voice record starting and all the stuff logs and stream is received and decoded. Also in between log is this
`I/org.webrtc.Logging: AndroidVideoDecoder: Decoder format changed: {crop-right=1023, color-format=2141391878, slice-height=768, mime=video/raw, hdr-static-info=java.nio.HeapByteBuffer[pos=0 lim=25 cap=25], stride=1024, color-standard=1, color-transfer=3, crop-bottom=767, crop-left=0, width=1024, color-range=2, crop-top=0, height=768}`

Whereas on different network, there is no such log. And all these things are not initiated.

Going by sdp, H264 is the codec asked for so what maybe breaking playback?

kclyu, 2019-02-27:
There is no problem with sdp for codec negotiation on the sdp you sent.
Have you checked ICE candidate signaling messages and how the connection state changes?

mintu19, 2019-02-27:
Log from proxy server to rws logging messages
[android.log](https://github.com/kclyu/rpi-webrtc-streamer/files/2910248/android.log)

kclyu, 2019-02-27:
In a slight glance, there is an ice candidate message for Device -> Server, but there is no ice candidate message for Server -> Device candidate. It seems that the Android device does not properly deliver the ice candidate message.

Device -> Server : {"to":"server","deviceid":"AL001","roomid":"765302883","clientid":"09237923","message":"{\n   \"cmd\" : \"send\",\n   \"msg\" : \"{\\n   \\\"candidate\\\" : \\\"candidate:1710908802 1 udp 2122262783 2405:201:2801:9f86:d4ba:5ad0:b34:91e5 44396 typ host generation 0 ufrag VXa0 network-id 4 network-cost 50\\\",\\n   \\\"id\\\" : \\\"audio\\\",\\n   \\\"label\\\" : 0,\\n   \\\"type\\\" : \\\"candidate\\\"\\n}\\n\"\n}\n"}

 Server -> Device : "{\"cmd\":\"send\",\"msg\":\"{\\\"type\\\":\\\"candidate\\\",\\\"label\\\":0,\\\"id\\\":\\\"audio\\\",\\\"candidate\\\":\\\"\\\"}\",\"error\":\"\"}"

mintu19, 2019-02-27:
But i am getting similar messages on same network as well but both audio and video works perfectly.
I will recheck and let you know different log in both cases.

mintu19, 2019-02-27:
For same network android
[android_local.log](https://github.com/kclyu/rpi-webrtc-streamer/files/2910471/android_local.log)

For different network
[android_different.log](https://github.com/kclyu/rpi-webrtc-streamer/files/2910473/android_different.log)

From web application
[web.log](https://github.com/kclyu/rpi-webrtc-streamer/files/2910474/web.log)

There is difference between logs from web and android. However i tried to follow same code on android as on web application. (Code in above linked stackoverflow question)

However both local android and remote android code seems to be similar in way with some differences 
[Differences](https://www.diffchecker.com/bxxTPjM8)

kclyu, 2019-02-27:
Among candidate messages, srflx is nat hole.
In the local network, you made rtp connection with  candidate messages from one peer. But. Peers using the public network under nat must share the ice candidates between both of peers.

mintu19, 2019-02-27:
Thanks

It was indeed issue with ice-candidates. 

I was using
` message.put("candidate", iceCandidate.serverUrl)`

Instead was required to use 
`message.put("candidate", iceCandidate.sdp)`

Marked for close

closed 2019-02-27 by mintu19