kclyu/rpi-webrtc-streamer #66: Stream shows only in the local network

uCibar, 2019-03-04:
Hello Lyu,

I'm using `RWS-0.73.2` deb package on Raspberry Pi Zero W Stretch-Lite and Its works great. But when I connect my laptop or Rpi to different networks(my phone mobile internet for example), ICE connection state failed. I'm not using a TURN Server. I'm not sure if there is a problem with TURN or something else.

Here is the details:

I made a simple proxy works on RPi. Proxy simply works as follows:

web-app send signaling details to signaling-server, server sends to proxy, proxy sends to rws. or visa versa.
`(my-webrtc-web-app) <--> (my-signaling-server) <--> (proxy-on-rpi) <--> (rws-on-rpi)`

and its work fine if `web-app` and `RPi` in same network. `web-app` just a copy-paste of `web-root/native-peerconnection`. I ony change websocket url to `my-signaling-server`. long story short, in this setup signaling works fine.

I check signaling logs in browser and RWS successfully gather and send `srflx` type ice candidates and `web-app` successfully gather and send `srflx` type ice candidates too.

I run `stun_prober` on my laptop(connected to my home wifi) and it's `Success Precent` is 100. I connect my laptop to mobile internet and run `stun_prober` again and `Success Precent` is 100. So when web-app(laptop) connected to `home-internet` and RPi connected to `mobile-internet` (or visa versa) I don't need a TURN server? or I'm so wrong and I need a TURN server.

As I said at the beginning  I'm not sure if there is a problem with TURN or something else. But if its TURN related issue, can I add TURN server to `RWS-0.73.2`? Because in `README_rws_setup_md` there is a `not implemented yet` warning for turn_server configuration.

Really appreciate your work and your help.

kclyu, 2019-03-04:
On the network you are testing, you can try connecting to https://appr.tc/ on a different network to see if you really need a turn server. appr.tc supports both stun and turn server, so you can use wireshark to check if RTP  is connected directly or turn server after connecting video call.

You can specify the ip address of turn server provided by appr.tc in the wireshark capture filter and the ip address of the other end.

Here's how to check [multiple ip addresses with wireshark](https://osqa-ask.wireshark.org/questions/4538/capturing-packets-on-multiple-ip-address-filter).

uCibar, 2019-03-04:
I've tried https://appr.tc/ and yes I need a TURN server if one of the peers is connected to 4G/LTE. If both of the peers connected to NAT servers, there is a no problem.

Quick question before closing issue: Can I use TURN Server config in RWS-0.73.2 Deb package or  Do i need cross-compile the latest version of RWS? cross-compiling makes me scared :)

Thanks for answering

kclyu, 2019-03-05:
To test turn server, you need to use the latest rws, but currently only the binaries of Raspberry PI V2 or higher will work.
( I would like to create rws for RaspW (armv6l), but I can not get it because I can not check the problem with the latest kernel or library side. )

If you have Raspberry PI V2 or higher hardware, I will upload the compiled binaries.

kentloog, 2019-03-20:
@uCibar - did you get it working? I also can't see the video stream outside the network, in the device list it shows status "Available" though.

uCibar, 2019-03-20:
There is a weird situation:

I changed my mobile service provider one week ago and with my old service provider, I need a TURN server if I want connect rpi-webrtc-streamer with 4G/LTE. But I can't successfully manage TURN server and rpi-webrtc-streamer. Maybe my TURN server configuration was wrong, maybe rpi-webrtc-streamer is broken, I don't know. But without rpi-webrtc-streamer, Eg. browser-to-browser webrtc connection, my TURN server works. By the way; I tried [coturn](https://github.com/coturn/coturn) on Azure and [numb TURN Server](https://numb.viagenie.ca).

With new mobile service provider, I don't need a TURN server (I think) because ICE candidate connect with `srflx` on 4G/LTE. that's strange to me. And here is the other weird thing; I have a [Sixfab 4G/LTE  Shield](https://sixfab.com/product/raspberry-pi-3g-4glte-base-shield-v2/) for raspberry pi and if Rpi connect to internet(4G/LTE) with this shield, rpi-webrtc-streamer works great. its connect with `srflx` and connection and live stream so smooth. But if I open a mobile hotspot on my phone and Rpi connects to hotspot, its didn't work. ICE connection failed. Rpi streamer and browser peer not send `srflx` and `relay` candidates, they sent only `host` candidates. To try with relay; I changed `ice_transports_type` to `relay` in webrtc_streamer.conf but no, didn't worked. rpi-webrtc-streamer sent `relay` candidates but `relay addr` is `0.0.0.0` but its must TURN server IP (its didn't work both of 4G/LTE shield and mobile hotspot).

Long story short: I can connect rpi-webrtc-streamer on `srflx` with my new mobile service provider 4G/LTE. But if I change `ice_transports_type` to `relay` or if rpi connects a mobile hotspot, its didn't work. I don't want to talk in advance because there are too many different parameters in my experiments(TURN config, 4G/LTE shield config, hostspot config etc.) and my knowledge on webrtc is poor but I think rpi-webrtc-streamer can't work with `relay` connections.

@kentloog - When I opened this issue, my situation related with TURN Server and 4G/LTE. I can successfully connect to rpi-webrtc-streamer with different NAT networks(home wifi to home wifi in example) so rpi-webrtc-stream works great with NAT networks(its can successfully connect with `srflx` candidates). If you can't connect with different networks, debug your network(s). forget rpi-webrtc-streamer and try with a browser-to-browser webrtc example(Eg. [appr.tc](https://appr.tc)), if your browser peers connects with `srflx`, your network(s) not need a TURN server. So rpi-webrtc-streamer must works. if not work, there is a different issue there. But if your browser-to-browser example connects with `relay`, you need a TURN server and welcome to club. I can't successfully manage `relay` with rpi-webrtc-streamer. Note: you can debug with console log messages or `chrome://webrtc-internals`, [check this article](https://testrtc.com/webrtc-api-trace/)

kclyu, 2019-03-21:
Could you please try again with the webrtc binary attached below? The attached file fixes the signaling problem that occurred after applying the unitifed plan.
[webrtc-streamer.gz](https://github.com/kclyu/rpi-webrtc-streamer/files/2990796/webrtc-streamer.gz)
```
gzip -d webrtc-streamer.gz
chmod 755 webrtc-streamer
sudo systemctl stop rws
sudo cp webrtc-streamer /opt/rws
#  Edit your etc/webrtc_streamer.conf before staring RWS
#  You can run webrtc-streamer within console without using systemctl
cd /opt/rws
./webrtc-streamer --verbose
#  
sudo systemctl start rws
```

In fact, the file I attached earlier was a binary that was run with turn server in my test environment. And the newly attached file is a version that fixes the problem of a signaling issue.
There is a limit to test of the network environment because each user uses different network environment.  In my environment, it works well, so it needs to be tested in a network environment where it does not work.
Thank you in advance for your testing.

kentloog, 2019-03-21:
@uCibar do you use this configuration showed on the picture
[image]

kclyu, 2019-03-21:
I use reverse proxy and client certs. However, the configuration above does not seem to be related to the turn server, so I wonder why you ask that question.

kentloog, 2019-03-21:
> I use reverse proxy and client certs. However, the configuration above does not seem to be related to the turn server, so I wonder why you ask that question.

I am using the standalone configuration, the first one on the list, maybe that's why i cant see the stream outside wifi that Raspberry is in?

kclyu, 2019-03-21:
uCibar has already mentioned in its first issue that he is using developed signaling server by himself.  And as you read through the issues, uCibar is right about the turn server issue. It does not seem to be related to signaling message exchange.

uCibar, 2019-03-21:
@kentloog - I'm using RWS standalone and custom signaling logic. I wondered if you using RWS standalone, how your signaling works if raspberry pi out of your network? Because on RWS standalone, browser side peer Eg. `web-root/nativepeerconnection` directly connects RWS websocket server and signaling happens. But outside of network, it can't connect RWS websocket server obviously. If your issue signaling related, please open a new issue, I will try to help. If your issue TURN Server related(Eg. your ICE connection state: failed) We are trying to solve in this issue.

@kclyu - I will try the last rws binary you mentioned, but my raspberry pi was broken :D and I ordered new one. So test may be delayed 4-5 days.

uCibar, 2019-03-25:
@kclyu - I tried the last rws-binary you mentioned with [numb turn server](https://numb.viagenie.ca) and its worked great with `relay` candidates. thanks for helping.

Quick question(bug report maybe) before closing the issue: When I run the last rws-binary you mentioned, its give me a `Segmentation fault` error. (Eg. `sudo ./webrtc-streamer` or systemctl runs the rws) but if I run with `--verbose` there is a no error. Here is the full log:

```
$ sudo ./webrtc-streamer
config file :etc/webrtc_streamer.conf\
Using config file base path:CWD\
Base Log Path : log  
Segmentation fault 
```
I don't know how RWS works but I think when I start RWS(the last binary) its try to write log files and gives  `Segmentation fault` but when runs with `--verbose` its not write to log files and works without error. I deleted everything in the log folder (`sudo rm -rf log/*`) but its give me this error:
```
config file :etc/webrtc_streamer.conf
Using config file base path:CWD
Base Log Path : log
Failed to make directory iterator: log/08
Failed to rotate log files at path: log
Failed to init file message logger
```
I deleted only `rws_log_0` but its give me a  `Segmentation fault` (same as above)

kclyu, 2019-03-26:
Please try the attached file.
[webrtc-streamer.gz](https://github.com/kclyu/rpi-webrtc-streamer/files/3007086/webrtc-streamer.gz)

To use rws again in systemctl, use the following command.
```
gzip webrtc-streamer.gz
chmod 755 webrtc-streamer
sudo cp webrtc-streamer /opt/rws
sudo systemctl start rws 
```
Thank you for helping with the test.

uCibar, 2019-03-26:
thanks for helping again, it works perfect. I'm closing the issue.

closed 2019-03-26 by uCibar