kclyu/rpi-webrtc-streamer #67: packet drop after replacing MMAL encoder with custom H264 bitstream

BloodyF, 2019-03-09:
Hi Lyu,
I tried feed H264 bitstream from Hardware encoder of H3 Allwinner. So I reimplemented DrainProcess() in raspi_encoder_impl.cc. Firstly, webrtc cannot parse my H264 bitstream (frame by frame, and about 30-130 KB/frame). I added header to every frame and it can stream to browser now. But the quality is very bad because webrtc send too slow and cause dropping packet as log below:
`(stunport.cc:284): Port[b3168c30:audio:1:0:local:Net[wlan0:2405:4800:58c7:x:x:x:x:x/64:Unknown:id=4]]: UDP send of 1202 bytes failed with error 11
(stunport.cc:284): Port[b3168c30:audio:1:0:local:Net[wlan0:2405:4800:58c7:x:x:x:x:x/64:Unknown:id=4]]: UDP send of 190 bytes failed with error 11
(stunport.cc:284): Port[b3168c30:audio:1:0:local:Net[wlan0:2405:4800:58c7:x:x:x:x:x/64:Unknown:id=4]]: UDP send of 1202 bytes failed with error 11
(stunport.cc:284): Port[b3168c30:audio:1:0:local:Net[wlan0:2405:4800:58c7:x:x:x:x:x/64:Unknown:id=4]]: UDP send of 1202 bytes failed with error 11
(stunport.cc:284): Port[b3168c30:audio:1:0:local:Net[wlan0:2405:4800:58c7:x:x:x:x:x/64:Unknown:id=4]]: UDP send of 1202 bytes failed with error 11
(stunport.cc:284): Port[b3168c30:audio:1:0:local:Net[wlan0:2405:4800:58c7:x:x:x:x:x/64:Unknown:id=4]]: UDP send of 1203 bytes failed with error 11
get 0, 25648
(rtp_sender.cc:748): Failed resending RTP packet 36796, Discard rest of packets.
(rtp_sender.cc:748): Failed resending RTP packet 36968, Discard rest of packets.`

Do I need to re-implement any part of source code to be able to send smootly?
Thank you

kclyu, 2019-03-09:
I will answer if  I have experienced the same error, but for the moment I do not have the same experience so I can not say what is problem. 

First, you need to check if the frame is normal and save it as a file and check whether the video player plays normally.  If the saved file is normal, you may think there is a problem in the receiving part of the frame. Just check the part that receiving  the frame from your device and sending the encoded frame to WebRTC native stack.

BloodyF, 2019-03-10:
Thank you, I will try it and let you know soon if there is good news

BloodyF, 2019-03-15:
Hi @kclyu 
The problem seems to be my network card. When I reduce resolution, it's able to stream smoothly.
BTW, can you update source code to be compatible with latest webrtc native updates? I see many changes, especially how to create a fake video track
Thanks

kclyu, 2019-03-15:
master is already compiled with the latest native stack. Please try git pull and compile again.

closed 2019-04-05 by kclyu