kclyu/rpi-webrtc-streamer #69: Stream shows only in the same network using RWS-frontend

kentloog, 2019-03-25:
Hi again,

I am using RWS and RWS-frontend but i can't see the video stream in the frontend app when for example my smartphone is not in the same WiFi as my Rpi. Can you guide me to make it work? I have implemented RWS standalone so far, do i need to make nginx reverse proxy or what?

uCibar, 2019-03-25:
In the world of WebRTC, peers must send a variety of information to each other to establish a webrtc connection. this is called signaling. in a custom webrtc application you can do the signaling mechanism the way you want (you can use websockets or http request/response or google firebase or carrier pigeon. just be able to successfully exchange the information with peers). RWS already has a websocket server for signaling. So, when you use RWS-frontend; RWS-frontend connects the websocket server and signaling occurs (browser peer and RWS peer exchanges information with websocket connection). So, if RWS and browser peer(RWS-frontend) in the different networks, browser can't establish a websocket connection with RWS. Because they are in different networks and RWS has not a public IP.

So if you don't establish a signaling mechanism for different networks, you can't connect RWS. Here is some approaches to solve RWS signaling(for different networks):

1. Give public IP to your Raspberry Pi. In this way, your browser peers can find the RPi and connects the RWS websocket server. (Not recommended - Security issues)
2.  Use nginx reverse proxy. I have a very poor knowledge about nginx and reverse proxies but there is a documentation about it.
3. Use a Cloud provider/VPS etc. (My way): I have a websocket server on heroku and my browser peer connects the my websocket(signaling) server and sends WebRTC information to it. On the raspberry pi, I have a simple NodeJS app(I called it `rws-proxy`) and this app connects the my webrtc server. Same time its connects the RWS websocket server too. the app receives related information from my websocket server and sends RWS websocket server. Or receives related information from RWS websocket server and send my websocket server. So this app(30-40 lines code by the way.  'app' is a big word) just bridge beetween RWS and public internet. Here is a visual explanation:

**RWS connection on same network:** `(rws-frontend) <-- info --> (rws-websocket-server)`

**My Way(for different networks):** `(rws-frontend) <-- info --> (my-webrtc-server-on-public-internet) <-- info --> (my-nodejs-app-on-rpi) <-- info --> (rws-websocket-server-on-rpi)`

You don't have to use websocket server by the way, you can use firebase in example.

After establishing a signaling mechanism, if you still can't connect the RWS, please share your browser peer logs and RWS logs. If  you can write your issue more detailed, I will help you with more detail.

Sorry for bad English and language mistakes...

kentloog, 2019-03-26:
Thanks @uCibar for help. I decided to use nginx reverse proxy.
Question to @kclyu 
I have set up this nginx conf. I also set up route forwarding with my router on my Raspberry ip for ports 80.
`server {
        listen 80;
        server_name mydomain.xyz;
        root /home/pi/Workspace/rpi-webrtc-streamer/web-root;
        index index.html index.htm;

        location / {
                allow 10.0.0.0/24;
                allow 192.168.0.0/24;
                allow 127.0.0.1;
                deny all;
                try_files $uri $uri/ =404;
        }

        location /rws/ws {
                proxy_pass http://localhost:8889;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
        }
}
`
When i go to mydomain.xyz i get err 403. When i got to mydomain.xyz/rws/ws i get err 415
The config is exactly the same as you have made in the setup readme.

What should i change in my config. And after changing config what should i do next.

kclyu, 2019-03-26:
The above example is literally an example. Depending on the network you are using, you may need to modify the content accordingly. Instead of modifying the example, it would be rather convenient to modify the network to fit the local network. the use case such as reverse proxy must develop authentication procedures like user login.

If it is complicated to use, it is recommended to check whether the router supports VPN  and use it through VPN. VPN is the best way to use it because you do not need additional configuration or further development.

closed 2019-03-29 by kentloog

meduar, 2022-05-31:
Any example to develop the RWS WebSocket server?