kclyu/rpi-webrtc-streamer #70: Usnig np2 interface

tovil, 2019-03-27:
Hi,
When I'm changing some values and click apply at the configuration web page I receive:
```
(app_ws_client.cc:669): SendResponse
(app_ws_client.cc:690): Media Config JSON response : {
   "cmd" : "response",
   "data" : "{\n   \"video_awb_mode\" : \"auto\",\n   \"video_brightness\" : 50,\n   \"video_contrast\" : 0,\n   \"video_ev\" : 1,\n   \"video_exposure_mode\" : \"auto\",\n   \"video_hflip\" : false,\n   \"video_rotation\" : 90,\n   \"video_saturation\" : 0,\n   \"video_sharpness\" : 0,\n   \"video_vflip\" : false\n}\n",
   "error" : "",
   "result" : "SUCCESS",
   "type" : "config"
}

(websocket_server_callback.cc:147): CallbackLibWebsocketsLWS_CALLBACK_RECEIVE
(app_ws_client.cc:213): OnMessage(24)
(app_ws_client.cc:231): JSON Parsing Success: {"cmd":"request","type":"config","deviceid":"00000000843b9d34","data":"apply"}
(streamer_observer.cc:119): IsStreamSessionActive
(mmal_wrapper.cc:640): Start reinitialize the MMAL encode wrapper.1920x1080@30, 17000000kbps
(mmal_wrapper.cc:900): capture stopped.
(mmal_wrapper.cc:190): **** MMAL Frame size error (buffer size: 196608), real frame size : 0
Segmentation fault
```

What I'n doing wrong?
Thanks.

kclyu, 2019-03-28:
np2 is the next version of native-peerconnection being tested. Of course you can use it. The video setting interface has been tested successfully after development. However, there is currently a problem with the vidoe setting interface. If the issue is fixed, I'll post it here.

kclyu, 2019-03-29:
Closing issue. If there is a problem, please open it again.

closed 2019-03-29 by kclyu