kclyu/rpi-webrtc-streamer #71: Signaling server in the cloud

tovil, 2019-03-28:
Hi,
I'm trying to move the signaling server to the cloud so webrtc-streamer and the web client will connect to the same endpoint in the cloud (e.g. example.com/ws).

I've tried to change **rws_ws_url** inside webrtc_streamer.conf and when I click on connect at the web, I get:
```
(websocket_server_callback.cc:167): Callback Handler is not found in URI: /rws/ws
(websocket_server_callback.cc:130): Callback Handler is not found in URI: /rws/ws
```
I assume that webrtc-streamer connects to its ws server internally?
Can you point me to the relevant peace of code that I can change in order achieve my goal? 

Thanks.

kclyu, 2019-03-28:
uCibar is kind enough to explain the outline. If you need more, it may be helpful to googling WebRTC signaling related materials.

https://github.com/kclyu/rpi-webrtc-streamer/issues/69#issuecomment-476348111

tovil, 2019-03-28:
Thanks, his bridge idea is nice.
I'll try it.

closed 2019-03-28 by tovil