kclyu/rpi-webrtc-streamer #73: rpi_rootfs kills my raspberry

tinybrainrobotics, 2019-03-29:
Hey,

I was trying to build a docker image to do the building.

But when it gets to rootfs my raspberry becomes completely unresponsive.
until the connection breaks.

```
packet_write_wait: Connection to 192.168.8.100 port 22: Broken pipe
rsync: connection unexpectedly closed (983169 bytes received so far) [receiver]
rsync error: error in rsync protocol data stream (code 12) at io.c(235) [receiver=3.1.2]
rsync: connection unexpectedly closed (983162 bytes received so far) [generator]
rsync error: unexplained error (code 255) at io.c(235) [generator=3.1.2]
```

tinybrainrobotics, 2019-03-30:
Nevermind, it worked when I changed the connection.

Would a PR with this dockerfile be useful?

```
FROM ubuntu:18.04
ENV DEBIAN_FRONTEND=noninteractive

RUN apt update
RUN apt install -y git \
   wget \
   xz-utils \
   python \
   openssh-client \
   sshpass \
   pkg-config \
   rsync

RUN mkdir -p /home/tools
WORKDIR /home/tools
RUN git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
ENV PATH="${PATH}:/home/tools/depot_tools"

RUN mkdir -p /home/Workspace
WORKDIR /home/Workspace
RUN git clone https://github.com/kclyu/rpi_rootfs

RUN mkdir /home/Workspace/rpi_rootfs/tools
WORKDIR /home/Workspace/rpi_rootfs/tools

#RUN wget -O arm-linux-gnueabihf.tar.xz https://www.dropbox.com/s/zom2m8vwkcryn1e/gcc-linaro-6.4.1-2017.01-x86_64_arm-linux-gnueabihf.tar.xz%3Fdl%3D0?dl=0  && \
ADD gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf.tar.xz .
RUN    ln -sf gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf  arm-linux-gnueabihf && \
    cd /opt &&  ln -sf /home/Workspace/rpi_rootfs
ENV PATH="${PATH}:/opt/rpi_rootfs/tools/arm-linux-gnueabihf/bin"

RUN mkdir -p /home/Workspace/webrtc
WORKDIR /home/Workspace/webrtc
RUN fetch --nohooks webrtc && \
    gclient sync && \
    cd /home/Workspace/webrtc/src && \
    git config branch.autosetupmerge always && \
    git config branch.autosetuprebase always && \
    git checkout master

# Installing ssh client to connect to raspberry pi
RUN RUN echo -e "\n\n\n\n" | ssh-keygen -t rsa -N '' -f /root/.ssh/id_rsa && \
    sshpass -p raspberry ssh-copy-id -oStrictHostKeyChecking=no pi@raspberrypi.local

#hack to have lsb_release used by ./install-build-deps.sh
ADD ./lsb_release /usr/bin/
RUN cd /home/Workspace/webrtc/src/build && \
    sed 's/sudo//g' > install.sh && \
    chmod +x install.sh && \
    ./install.sh

RUN cd /home/Workspace/rpi_rootfs && \
    ./rpi_rootfs.py pi@raspberrypi.local ./

WORKDIR /home/Workspace
RUN git clone https://github.com/kclyu/rpi-webrtc-streamer
WORKDIR /home/Workspace/webrtc/src
RUN mkdir arm_build
RUN cp /home/Workspace/rpi-webrtc-streamer/misc/webrtc_arm_build_args.gn arm_build/args.gn
RUN gn gen arm_build && ninja -C arm_build

WORKDIR /home/Workspace/rpi-webrtc-streamer/src
RUN sh../mk/config_libwebsockets.sh
RUN make
```

kclyu, 2019-03-30:
I have been thinking about building dockers many times, but I have not made them up yet. If it works well, it will be helpful to those who compile and use it.

I think the following things need to be checked first. 

1. raspberry pi hardware is divided into armv6 and armv7. armv6 and armv7 are separate from args.gn file, so you need to be able to build with cpu type at build time.

2. Before creating root fs using rpi_rootfs in the docker file, you need to install the necessary packages first in raspberry pi, but this procedure is missing.
[3.1. Installing Raspberry PI Prerequisite](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md#31-installing-raspberry-pi-prerequisite)

3. WebRTC native code package is mostly successful in ninja build. Sometimes, however, build fails. It seems that there is no error case processing for build failure.  

4. Likewise, rws may fail to build using the latest WebRTC native code package, as the WebRTC native code packages commit a few times a day or change APIs or other native code libraries. 
In this case, the WebRTC native code package must be moved to a specific commit position. Here's how. [4.4. Moving to specific commit position of WebRTC native code package](https://github.com/kclyu/rpi-webrtc-streamer/blob/master/README_building.md#44-moving-to-specific-commit-position-of-webrtc-native-code-package)

5. Custom compiled GCC can be downloaded from the dropbox as shown below.
```
 wget -O gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf.tar.xz https://www.dropbox.com/s/qf6c68eia8478zk/gcc-linaro-6.4.1-2018.10-x86_64_arm-linux-gnueabihf.tar.xz?dl=1
```
I do not know if this is possible, but I wonder if a docker file can be created as shown below.

1. Is it possible to isolate the build procedure so that each procedure can be executed?
(If the user does not specify a specific procedure, build the whole procedure.)
    - build rpi root fs
    - fetch webrtc and build
    - build rws

2. Move the webrtc native code to a specific commit position and build
 
3. Installing the built rws binary into a specific raspberry pi

closed 2020-10-28 by kclyu