kclyu/rpi-webrtc-streamer #75: Does rpi-webrtc-streamer supports data channels?

tartavull, 2019-03-31:
Does rpi-webrtc-streamer supports data channels?

kclyu, 2019-04-01:
Datachannel is not used in RWS, so Datachannel is not implemented. I am wondering what the purpose of Datachannel in RWS is, even if it supports Datachannel.

kclyu, 2019-04-05:
Please reopen the issue again if you have an issue.

closed 2019-04-05 by kclyu

tartavull, 2019-04-17:
Hey @kclyu,
   if you have a camera which can be oriented using a motor, you could use a datachannel to set the position.

kclyu, 2019-04-18:
The browser datachannel uses javascript, so it can be useful for many purposes. However, since rws creates a PeerConnection with cpp, the DataChannel must also be created in cpp, and the ability to use datachannel must also be developed together with cpp. In a nutshell, it is not possible to use it freely like javascript of browser.

I know what you are going to do, but it seems to be a bad idea to include functionality to the app you are developing  in your rws. It seems to be the only way to modify the source directly if necessary.

tartavull, 2019-04-18:
That makes sense, thank you @kclyu !

miguelwon, 2020-04-30:
Hi,
Any change that this could be implemented? Like @tartavull wrote, a data channel would be great for controlling remotely the camera and other servos. I think it would be enough for the data channel output to be sent to stdout and then parse it with some python script.

kclyu, 2020-05-02:
So far, the idea of supporting DataChannel has not changed.

I don't know if it's a good idea to use log message to pass specific data to another process.
If I implement the necessary functions above, I will consider the implementation method with the following design.

In order to exchange the current signaling message, the WebSocket connection is maintained during streaming. I will not explain why WebSocket keeps connecting because it is not related to this issue. I  will create a new JSON message type that transmits a message to a specific UDP/TCP port in the signaling message and link it with other processes.

1. open UDP/TCP port which is bind to localhost only.
2. If it is a JSON message that delivers the message to another process, the message is sent to a specific socket.
3. If there is a JSON event/response message in the socket for message transmission, send this message to the client via WebSocket.

So far, it is a method for interworking with other processes by modifying the signaling websocket of RWS. However, this method is not recommended because it is a method of another message protocol on the signaling websocket. 

Another method is to place nginx between client and rws,  WebSocket for RWS signaling uses nginx Reverse Proxy, and websocket or http is used for interworking with other hardware control processes. 

The implementation method seems to be sufficient for using any additional module  or  rws source modifications.

miguelwon, 2020-05-02:
Ok, thanks.  Small feedback on my recent experience: I have tried webrtc with Gstreamer and sending commands (jsons strings) through the same WebSocket used for signaling.  It didn't work very well for the scenario where I need to send commands fast and constantly. It was hanging quite frequently. With a DataChannel I do not see that hanging problem.