kclyu/rpi-webrtc-streamer #76: ANSWER with the candidates ice

leofds, 2019-04-03:
Hi
I noticed something that could be a bug in RWS ...
When the RWS receives the ANSWER message immediately followed by the candidates ICE, it seems to me that the candidates are ignored by the program. Because no sent STUN ping to these candidates is generated.

as follows...

```
(direct_socket.cc:155): OnRead
(direct_socket.cc:170): Message IN from client: "{"sdp":"v=0\r\no=- 322066285453496108 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0 1\r\na=msid-semantic: WMS ARDAMS\r\nm=audio 9 UDP\/TLS\/RTP\/SAVPF 111 103 9 102 0 8 105 13 110 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:rIhv\r\na=ice-pwd:d8kophvOuXt0uIYFzr4RtkJe\r\na=ice-options:trickle renomination\r\na=fingerprint:sha-256 FD:6B:D7:76:C0:E1:51:90:E6:E4:6C:23:F3:35:9F:98:94:71:49:80:84:53:24:4F:2A:04:E0:50:A6:CC:38:6A\r\na=setup:active\r\na=mid:0\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:2 http:\/\/www.ietf.org\/id\/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:3 urn:ietf:params:rtp-hdrext:sdes:mid\r\na=extmap:4 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id\r\na=extmap:5 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id\r\na=sendrecv\r\na=msid:ARDAMS ARDAMSa0\r\na=rtcp-mux\r\na=rtpmap:111 opus\/48000\/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC\/16000\r\na=rtpmap:9 G722\/8000\r\na=rtpmap:102 ILBC\/8000\r\na=rtpmap:0 PCMU\/8000\r\na=rtpmap:8 PCMA\/8000\r\na=rtpmap:105 CN\/16000\r\na=rtpmap:13 CN\/8000\r\na=rtpmap:110 telephone-event\/48000\r\na=rtpmap:113 telephone-event\/16000\r\na=rtpmap:126 telephone-event\/8000\r\na=ssrc:1319991438 cname:TgMSWnVtRZYWYg7G\r\nm=video 9 UDP\/TLS\/RTP\/SAVPF 98 99 100 101 127\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:rIhv\r\na=ice-pwd:d8kophvOuXt0uIYFzr4RtkJe\r\na=ice-options:trickle renomination\r\na=fingerprint:sha-256 FD:6B:D7:76:C0:E1:51:90:E6:E4:6C:23:F3:35:9F:98:94:71:49:80:84:53:24:4F:2A:04:E0:50:A6:CC:38:6A\r\na=setup:active\r\na=mid:1\r\na=extmap:14 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:13 http:\/\/www.webrtc.org\/experiments\/rtp-hdrext\/abs-send-time\r\na=extmap:12 urn:3gpp:video-orientation\r\na=extmap:2 http:\/\/www.ietf.org\/id\/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:11 http:\/\/www.webrtc.org\/experiments\/rtp-hdrext\/playout-delay\r\na=extmap:6 http:\/\/www.webrtc.org\/experiments\/rtp-hdrext\/video-content-type\r\na=extmap:7 http:\/\/www.webrtc.org\/experiments\/rtp-hdrext\/video-timing\r\na=extmap:8 http:\/\/tools.ietf.org\/html\/draft-ietf-avtext-framemarking-07\r\na=extmap:9 http:\/\/www.webrtc.org\/experiments\/rtp-hdrext\/color-space\r\na=extmap:3 urn:ietf:params:rtp-hdrext:sdes:mid\r\na=extmap:4 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id\r\na=extmap:5 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id\r\na=sendrecv\r\na=msid:ARDAMS ARDAMSv0\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:98 H264\/90000\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=fmtp:98 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:99 rtx\/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:100 red\/90000\r\na=rtpmap:101 rtx\/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:127 ulpfec\/90000\r\na=ssrc-group:FID 1815093281 274512821\r\na=ssrc:1815093281 cname:TgMSWnVtRZYWYg7G\r\na=ssrc:274512821 cname:TgMSWnVtRZYWYg7G\r\n","type":"answer"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:1242744575 1 udp 2122262783 2804:18:5f:4966:c105:6420:f50d:e26c 46881 typ host generation 0 ufrag rIhv network-id 4 network-cost 900"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:3256763322 1 udp 2122194687 100.119.116.160 40306 typ host generation 0 ufrag rIhv network-id 3 network-cost 900"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:559267639 1 udp 2122136831 ::1 39380 typ host generation 0 ufrag rIhv network-id 2"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:1510613869 1 udp 2122063615 127.0.0.1 50809 typ host generation 0 ufrag rIhv network-id 1"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:842163049 1 udp 1685987071 177.79.70.85 17658 typ srflx raddr 100.119.116.160 rport 40306 generation 0 ufrag rIhv network-id 3 network-cost 900"}
"
(streamer_observer.cc:101): MessageFromPeer
```

In my tests this happens withfrequency in the mobile network 4g.

kclyu, 2019-04-04:
DirectSocket is intended to be used on an local network, not on an external network, so I did not care much about handling signaling messages in Direct Socket. DirectSocket of  Android  AppRTCMobile is supposed to be used only in environments where the original design is like Local Network.

Anyway, it is a problem to handle only one message when several messages come in.

kclyu, 2019-04-04:
If you are doing a project using DirectSocket, I have one thing to tell you. As you know, all signaling messages are sent over TCP, so users' IP information and other information can be easily seen. Because DirectSocket is particularly vulnerable to security, you should pay particular attention to where you can troubleshoot security issues. Using external networking for personal use may be a problem (if someone tries to view a cam installed in their home), it can be a problem especially if you have multiple users to service . If you want to use DirectSocket, please consider using VPN or other security related matters.

leofds, 2019-04-04:
I agree with your considerations. 

But DirectSocket can be modified by implementing security mechanisms in the App and RWS.

At the moment I am only conducting tests with the APPRTCMobile and the native web page of the project. As you know I've implemented a server and a proxy in RPI to allow the exchange of SPD messages on different networks.

As I am still new to WebRTC, I am trying to understand and resolve connection problems between different networks.

I'm having interesting results but this modification in code is required for App compatibility.

leofds, 2019-04-04:
I have seen that you have already modified the DirectSocket code, the same modification is needed in `streamer.cc` in `OnMessageFromPeer`.

```
(streamer.cc:420):  Received candidate :{"type":"candidate","label":0,"id":"0","candidate":"candidate:842163049 1 udp 1685987071 177.79.72.162 3423 typ srflx raddr 100.107.43.21 rport 57563 generation 0 ufrag JrAY network-id 3 network-cost 900"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:3966932273 1 udp 25043199 181.215.183.94 49168 typ relay raddr 177.79.72.162 rport 3419 generation 0 ufrag JrAY network-id 3 network-cost 900"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:2733730241 1 udp 41821183 181.215.183.94 61787 typ relay raddr 177.79.72.162 rport 3423 generation 0 ufrag JrAY network-id 3 network-cost 900"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:3966932273 1 udp 25043455 181.215.183.94 54614 typ relay raddr 177.79.68.65 rport 53825 generation 0 ufrag JrAY network-id 3 network-cost 900"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:4075018308 1 udp 8265471 181.215.183.94 60524 typ relay raddr 177.79.72.162 rport 3417 generation 0 ufrag JrAY network-id 3 network-cost 900"}
{"type":"candidate","label":0,"id":"0","candidate":"candidate:4075018308 1 udp 8265727 181.215.183.94 54133 typ relay raddr 177.79.68.65 rport 53852 generation 0 ufrag JrAY network-id 3 network-cost 900"}
```

kclyu, 2019-04-04:
Do you use your own proxy? Send a candidate message from 'cmd: send' to each candidate instead of sending the candidate to the array. If not, please capture the log with 'streamer --verbose' and send it as a file.

leofds, 2019-04-04:
I use my proxy, but I do not interfere with the messages, it's practically a TCP/IP bridge between RWS and the App.

[log file](https://drive.google.com/open?id=1p6k9GiMgpfLIvPryJZrDqswR41hBWfqt)

kclyu, 2019-04-04:
The proxy you are using seems to have truncated '\n'. Since DirectSocket is a protocol built on top of TCP to simplify the protocol, there must be '\n' as a delimiter between all messages.

And if you attach a file, you can just attach it to the comment with drag & drop.

leofds, 2019-04-04:
I'm sorry, I evaluated the log in the wrong way, I compiled the new version and it's ok.

[multiple DirectSocket messages.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/3044848/multiple.DirectSocket.messages.txt)

closed 2019-04-04 by leofds