kclyu/rpi-webrtc-streamer #78: Resolution drop and fps in mobile browser

leofds, 2019-04-17:
Hi,
I'm testing native-peerconnection in the browser Google Chrome on the local network.
On the laptop everything works fine, but in the mobile browser after a while begin to drop resolutions and frames.

In this video the problem occurs after 1 minute.
[video](https://youtu.be/aGQ0xFnOreY)
[log](https://github.com/kclyu/rpi-webrtc-streamer/files/3087579/drops.txt)

mintu19, 2019-04-17:
Last I have ran rpi-webrtc-frontend for more than 3-4 min without any drop in resolution or fps. Maybe some internet or config error.

leofds, 2019-04-17:
Updated video link 
[video](https://drive.google.com/file/d/1USz27SvIqs9rBDhOTObK9mmjrXMx1UsL/view?usp=sharing)

media_config.conf
```
max_bitrate=3500000
resolution_4_3_enable=true
video_rotation=0
video_hflip=false
video_vflip=false
use_dynamic_video_resolution=false
use_dynamic_video_fps=false
fixed_video_resolution=640x480
fixed_video_fps=15
video_resolution_list_4_3=320x240,400x300,512x384,640x480,1024x768,1152x864,1296x972,1640x1232
video_resolution_list_16_9=384x216,512x288,640x360,768x432,896x504,1024x576,1152x648,1280x720,1408x864,1920x1080
audio_processing_enable=true
audio_echo_cancellation=true
audio_gain_control=true
audio_high_passfilter=true
audio_noise_suppression=true
video_sharpness=0
video_contrast=0
video_brightness=50
video_saturation=0
video_ev=0
video_exposure_mode=auto
video_flicker_mode=auto
video_awb_mode=auto
video_drc_mode=high
video_enable_annotate_text=false
video_annotate_text_size_ratio=3
video_annotate_text=%Y-%m-%d.%X
```

kclyu, 2019-04-17:
WebRTC Native Code Package adjusts the fps and bandwidth according to the network bandwidth. The latest WebRTC Native Code Package adds the part that adjusts the fps to a very low level when the expected bandwidth is low. This part seems to be modified according to the WebRTC spec. It seems that it is not able to control fps properly depending on the bandwidth. There seems to be a situation that does not consider network bandwidth at present. I have been monitoring this issue for a while now, but I do not think it has changed so far. When this issue is resolved, I will post it here.

kclyu, 2019-04-17:
Nobody knows how the bandwidth of the mobile network will change. It may vary depending on the situation of the mobile network or the wired network.

leofds, 2019-04-17:
In this case, I am connected to the same local network as RPI. Smartphone connected by Wifi and RPI by cable.
On the mobile network (4g) I understand that this behavior would be expected.

kclyu, 2019-04-17:
You using the latest version of the WebRTC native code package, and a month ago you used a slightly older native code package. Please, think about the steps you've tested.

p.s. If possible, please do not use issue post as chatting message. As I said before, it could be a mail bomb to other people monitoring.

leofds, 2019-04-18:
Analyzing the log better I could see the bitrate changing.
I'm looking at this information in the log:
`(bitrate_allocator.cc:115): Current BWE 4017270`
I realized that values above 1200k the video is excellent at 640x480 (max 15fps).

I tested with some different smartphones on the local network and on the 4g mobile network.
The result was kind of intriguing to me. In the 4g mobile network in almost every test the bitrate stayed above 1200k, but in the local network the average is 400k. I expected otherwise. (I had the opposite in the past weeks!)

I ran the test with some versions of RWS from the oldest to the most recent, and exchanging the STUN server of google with others already tested. But the results were the same.

I'm still analyzing my steps looking for what I did differently...

Is there access to any external information that may influence this? Some information in the SDP (offer or answer)?

Is there a setting or code that can be modified to try new results?
In the RWS or WebRTC Native Code Package ?

kclyu, 2019-04-18:
Bandwidth estimation (BWE) is located inside the WebRTC native code package. It would be a good idea to wait until bandwidth-related estimates change a bit more reliably.

closed 2020-10-28 by kclyu