kclyu/rpi-webrtc-streamer #79: Audio playback not working

mintu19, 2019-04-17:
I have installed rws latest deb and am using rpi-webrtc-streamer-frontend.
I am not able to record audio from browser and play it on usb sound card.
sound card is working and can play audio with aplay or mpg123.

Note that previously both recording and playback was working but now only recording is working. Playback is not working. 

```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 002: ID 1b3f:2008 Generalplus Technology Inc.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
pi@raspberrypi:~ $ cat .asoundrc
defaults.pcm.card 0;
defaults.ctl.card 0;
```

I have also tried with mobile app i have built. But same. I have not changed any config just in case. It was previously working (on other network). 

Also if i use AppRTC (local), audio works. But not with frontend or with custom mobile app (via internet).
Is there any issue when router comes in between? Or something else? 

My frontend code is [https://github.com/mintu19/rpi-webrtc-streamer-frontend](here)

kclyu, 2019-04-17:
In chrome://webrtc-internals /, please check whether the packet is transmitted or received correctly in the stat graph of audio related part. If you can not send audio from the stat graph, please try to replace it by receiving the latest binary file below.
[webrtc-streamer.gz](https://github.com/kclyu/rpi-webrtc-streamer/files/3089900/webrtc-streamer.gz)

To use rws again in systemctl, use the following command.
```
gzip -d webrtc-streamer.gz
chmod 755 webrtc-streamer
sudo systemctl stop rws
sudo cp webrtc-streamer /opt/rws
sudo systemctl start rws 
```

mintu19, 2019-04-17:
I will try the new binary.
Meanwhile i was not able to understand graphs. Attaching few screenshots.

[webrtc.zip](https://github.com/kclyu/rpi-webrtc-streamer/files/3091121/webrtc.zip)

mintu19, 2019-04-17:
I tried new binary (after installing libavahi, libgtk ??).

But now video only works for 2 sec and then stops. (Need to change anything else? Like config or at frontend?)

mintu19, 2019-04-18:
I changed config file with enable audio and other ice urls, it seems to not hang after 2 sec now. But 2 things

1.) Main - Audio is working with local (Apprtc) but does not work when used via internet (frontend on browser).

2.) Secondary - Resolution of video is lower than needed. I got 50 Mbps internet. I am getting resolution of 320 x 240.

kclyu, 2019-04-18:
I did not mention it before, but the reason I uploaded the binary was because I suspect that the latest chrome or browser is compatible with the previous deb binary. Please ignore the new binary and use the old binaries first.  (The latest binaries have issues with fps and bandwidth. I recommend that you do not use the latest binaries unless they are for testing purposes.)

Please, check that the audio icon is marked as muted on video. I did not consider it because it was an old issue, but the browser changed the audio muted state to default a long time ago.

mintu19, 2019-04-18:
Yes icon was muted. I have changed it to full volume. On both new and old binaries. But same issue. No audio while on internet. But works in local in both old and new.

Another thing i have noticed is that frontend is not asking for microphone permission anymore (which it used to ask 1 month back). Is it problem with browser? But audio doesn't work in my custom mobile app also. (it was working before)

I am trying to test it with other browsers. Lets see.

mintu19, 2019-04-18:
Some observations:

1.) Pollifills dialog lib is missing (added to index.html in frontend)

2.) In other brosers (especially mozilla), stream stop after few seconds

```
Apr 18 11:30:52 raspberrypi webrtc-streamer[1068]: [044:318] [1085] (audio_device_alsa_linux.cc:1539): capture snd_pcm_avail_update error: Broken pipe
Apr 18 11:30:52 raspberrypi webrtc-streamer[1068]: [044:406] [1085] (audio_device_alsa_linux.cc:1539): capture snd_pcm_avail_update error: Broken pipe
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: [044:488] [1085] (audio_device_alsa_linux.cc:1539): capture snd_pcm_avail_update error: Broken pipe
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: [044:577] [1085] (audio_device_alsa_linux.cc:1539): capture snd_pcm_avail_update error: Broken pipe
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: [044:659] [1085] (audio_device_alsa_linux.cc:1539): capture snd_pcm_avail_update error: Broken pipe
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: [044:747] [1085] (audio_device_alsa_linux.cc:1539): capture snd_pcm_avail_update error: Broken pipe
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: #
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: # Fatal error in ../modules/audio_processing/aec/aec_core.cc, line 205
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: # last system error: 11
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: # Check failed: 0 < AvaliableSpace() (0 vs. 0)
Apr 18 11:30:53 raspberrypi webrtc-streamer[1068]: #
Apr 18 11:30:53 raspberrypi systemd[1]: rws.service: Main process exited, code=killed, status=6/ABRT
Apr 18 11:30:53 raspberrypi systemd[1]: rws.service: Unit entered failed state.
Apr 18 11:30:53 raspberrypi systemd[1]: rws.service: Failed with result 'signal'.
```

3.) Chrome do not exhibit any such error in regard to stop playing

4.) Audio is still not working. Have you checked the graph scrshots? Because i don't know which one i have to look out for and how to check if it is true.

5.) Frontend is not asking for microphone permission in any browser.

6.) Test webrtc works.(https://webrtc.github.io/samples). Microphone works.

7) Native Peer Connection not working. Websocket disconnects very fast. No video is loaded. Same output on all browsers.

```
56.130:  Created RTCPeerConnnection with config: {"iceServers":[{"urls":"stun:stun.l.google.com:19302"}]}
common.js:6 
56.131:  WebSocket URL : ws://192.168.0.106:8889/rws/ws
common.js:6 
56.218:  Websocket connnected: ws://192.168.0.106:8889/rws/ws
common.js:6 
56.219:  C --> WSS: {"cmd":"register","roomid":"347400128","clientid":"63314797"}
common.js:6 
56.593:  Websocket Disconnected
```

kclyu, 2019-04-18:
Please specify the specific version of adater.js part of index.html as below and test it.
From
```
<script src="https://webrtc.github.io/adapter/adapter-latest.js"></script>
```
To
```
<script src="https://webrtc.github.io/adapter/adapter-7.1.0.js"></script>
```

p.s. I'm working on something else, so it's hard for me to test it myself.

mintu19, 2019-04-18:
This did not work.
browser is not even asking for microphone permission. Is problem in frontend where we have not used navigator.getusermedia? Something like

```
navigator.mediaDevices
    .getUserMedia({
      audio: true,
      video: false
    })
    .then(gotStream)
    .catch(e => {
      alert(`getUserMedia() error: ${e.name}`);
    });
```

I am going to try to find where to put that and let you know about that.

But this 2 way audio was working before. So don't know what went wrong in a month.

kclyu, 2019-04-18:
If you have already allowed permission before, you will not be asked anymore. If you look at the far left of the chrome URL input, you can see what is allowed in your site information. And when you open a microphone or video in a browser, you will see a red circle in the tab you are using that shows the recording in progress. Please check.

mintu19, 2019-04-18:
Yes of course I have checked that (and using new browser just renewed permissions). That is what I am saying. No permission asked. No red circle.

mintu19, 2019-04-18:
I have found the solution. It was indeed because it was not asking for microphone permission. Added code to ask permission and add microphone by navigator.getusermedia().

If you or someone need updated code, check [https://github.com/mintu19/rpi-webrtc-streamer-frontend/commit/fd942e184b70f6c70e28c0a145b17625fd7753e1](this commit)

kclyu, 2019-04-19:
You do not have to use the proxy to use audio, you can use https and use the static web frontend with rws port. But If I create a web frontend or other audio related part through rws port and upload it as an example, rws user  will use it without care about the security side, Of course, there are some advanced users who can do enough security and user account management, but I think most users will not care. 

For this reason I have not developed a web frontend yet, and I have developed a firebase frontend as a way to solve such problems. however the firebase frontend is available only to experienced developers,  it was too difficult to most users to use them.

The current idea is to develop a web frontend that can use self signed client certificate with docker, or a web frontend and backend that will not be perfect, but can solve some security problems.

If you think that audio quality related problems are serious, it would be better to create new issue and talk about them.  the WebRTC native code used in the deb package is a lot old, so audio issues should be addressed if there is a lot of problems after the new deb package.  (The new deb package will be updated if the current bandwidth estimation becomes stable.)

If you are interested in firebase front-end and you are willing to update it, I will give you permission to update the frontend repository.

closed 2019-04-20 by kclyu

mintu19, 2019-04-21:
I am using rws on Raspi which is connected to an IPv6 network that do not support port forwarding (or in IPv6 term, direct access to port via IP) somehow and I want to access my device over Internet.
I am using frontend via firebase (with firestore) proxy based on your frontend and it is working perfectly for me (I think security issues are also less via it right?). So I do not have any issue regarding using firebase proxy solution (or just one that firebase only have US based servers which makes signalling take time in other countries but no complaints there) .

I am currently testing voice on other network, if there is issue in other network as well, i will create a new issue. Other than that, everything is working fine. 

I just added the code for firebase frontend to ask microphone permission and just wanted to let you know if you wanted to update it in your repo as well. 

Also Looking forward to new deb. :)