kclyu/rpi-webrtc-streamer #81: Failed to set remote video description send parameters

leofds, 2019-04-24:
I am testing on a smartphone from a friend and the video does not appear. In the browser console log enter the following error after receiving the `offer`:

```
setRemoteDescription: InvalidAccessError: Failed to execute 'setRemoteDescription' on 'RTCPeerConnection': Failed to set remote offer sdp: Failed to set remote video description send parameters.
```

and

```
setLocalDescription: OperationError: Failed to execute 'setLocalDescription' on 'RTCPeerConnection': Session error code: ERROR_CONTENT. Session error description: Failed to set remote video description send parameters..
```

[chrome console.txt](https://github.com/kclyu/rpi-webrtc-streamer/files/3113873/chrome.console.txt)

The device that is not working is a Quantum MUV Pro.

Testing on AppRTCMobile I got the same result:

```
E/webrtc_video_engine.cc: (line 611): No video codecs supported.
E/peer_connection.cc: (line 5560): Failed to set remote video description send parameters. (INVALID_PARAMETER)
E/peer_connection.cc: (line 2521): Failed to set remote offer sdp: Failed to set remote video description send parameters.
```

offer:

```
    v=0
    o=- 236982666531954439 2 IN IP4 127.0.0.1
    s=-
    t=0 0
    a=group:BUNDLE 0 1
    a=msid-semantic: WMS stream_id
    m=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 102 0 8 105 13 110 113 126
    c=IN IP4 0.0.0.0
    a=rtcp:9 IN IP4 0.0.0.0
    a=ice-ufrag:X/z7
    a=ice-pwd:VHzfGDCHjv2bwVUw2tC1EpO/
    a=ice-options:trickle
    a=fingerprint:sha-256 AF:BE:92:A3:D2:B9:BB:2B:75:65:7A:84:D1:D2:39:0D:47:88:97:41:6B:B6:CB:46:49:91:33:17:03:73:DA:37
    a=setup:actpass
    a=mid:0
    a=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level
    a=extmap:2 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
    a=extmap:3 urn:ietf:params:rtp-hdrext:sdes:mid
    a=extmap:4 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id
    a=extmap:5 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id
    a=sendrecv
    a=msid:stream_id audio_label
    a=rtcp-mux
    a=rtpmap:111 opus/48000/2
    a=rtcp-fb:111 transport-cc
    a=fmtp:111 minptime=10;useinbandfec=1
    a=rtpmap:103 ISAC/16000
    a=rtpmap:9 G722/8000
    a=rtpmap:102 ILBC/8000
    a=rtpmap:0 PCMU/8000
    a=rtpmap:8 PCMA/8000
    a=rtpmap:105 CN/16000
    a=rtpmap:13 CN/8000
    a=rtpmap:110 telephone-event/48000
    a=rtpmap:113 telephone-event/16000
    a=rtpmap:126 telephone-event/8000
    a=ssrc:3551698015 cname:1OsgV2Ww9MHpOzY/
    a=ssrc:3551698015 msid:stream_id audio_label
    a=ssrc:3551698015 mslabel:stream_id
    a=ssrc:3551698015 label:audio_label
    m=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 127
    c=IN IP4 0.0.0.0
    a=rtcp:9 IN IP4 0.0.0.0
    a=ice-ufrag:X/z7
    a=ice-pwd:VHzfGDCHjv2bwVUw2tC1EpO/
    a=ice-options:trickle
    a=fingerprint:sha-256 AF:BE:92:A3:D2:B9:BB:2B:75:65:7A:84:D1:D2:39:0D:47:88:97:41:6B:B6:CB:46:49:91:33:17:03:73:DA:37
    a=setup:actpass
    a=mid:1
    a=extmap:14 urn:ietf:params:rtp-hdrext:toffset
    a=extmap:13 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
    a=extmap:12 urn:3gpp:video-orientation
    a=extmap:2 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
    a=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay
    a=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type
    a=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing
    a=extmap:8 http://tools.ietf.org/html/draft-ietf-avtext-framemarking-07
    a=extmap:9 http://www.webrtc.org/experiments/rtp-hdrext/color-space
    a=extmap:3 urn:ietf:params:rtp-hdrext:sdes:mid
    a=extmap:4 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id
    a=extmap:5 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id
    a=sendrecv
    a=msid:stream_id video_label
    a=rtcp-mux
    a=rtcp-rsize
    a=rtpmap:96 H264/90000
    a=rtcp-fb:96 goog-remb
    a=rtcp-fb:96 transport-cc
    a=rtcp-fb:96 ccm fir
    a=rtcp-fb:96 nack
    a=rtcp-fb:96 nack pli
    a=fmtp:96 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f
    a=rtpmap:97 rtx/90000
    a=fmtp:97 apt=96
    a=rtpmap:98 H264/90000
    a=rtcp-fb:98 goog-remb
    a=rtcp-fb:98 transport-cc
    a=rtcp-fb:98 ccm fir
    a=rtcp-fb:98 nack
    a=rtcp-fb:98 nack pli
    a=fmtp:98 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f
    a=rtpmap:99 rtx/90000
    a=fmtp:99 apt=98
    a=rtpmap:100 red/90000
    a=rtpmap:101 rtx/90000
    a=fmtp:101 apt=100
    a=rtpmap:127 ulpfec/90000
    a=ssrc-group:FID 3509771163 3906979821
    a=ssrc:3509771163 cname:1OsgV2Ww9MHpOzY/
    a=ssrc:3509771163 msid:stream_id video_label
    a=ssrc:3509771163 mslabel:stream_id
    a=ssrc:3509771163 label:video_label
    a=ssrc:3906979821 cname:1OsgV2Ww9MHpOzY/
    a=ssrc:3906979821 msid:stream_id video_label
    a=ssrc:3906979821 mslabel:stream_id
    a=ssrc:3906979821 label:video_label
```

It seems to me to be something related to profile-level-id.

The codec used by RWS is the H264 level 3.1 right?

I have seen in the commit history that there is something about level 4 and 4.1, is it possible to use them?

May be the video codec is not supported by the Mediatek MT6753 processor ! What do you think ?

kclyu, 2019-04-25:
The default profile-level-id 3.1 is required for compatibility. if the device  do not get 3.1 support from OS, it seems that mobile device does not recognize h.264 related codec properly.

closed 2019-05-02 by kclyu

leofds, 2019-05-03:
Hi Lyu

I do not know if you received the emails I sent you, to find a solution to this question. We can combine a value for your services. I believe it is possible to increase compatibility because I tested another project with WebRTC in RPI that worked on the Smartphone in question.

I found another incompatible Smartphone. Samsung Galaxy J2 Prime (SM-G532MT). Both have the same GPU, ARM Mali-T720.